cetic/6lbr #67: Mesh network not working properly

Devesh24, 2015-01-13:
Hello Guys,
                   I am working on a project where i am using 6lbr for routing. I am using cc2538 EM , here i am facing strange problem with mesh network, problem is that as long as a mote directly connected to slip-radio the mote getting ping, but when it got connected to slip-radio  via another mote the mote stopped pinging.
I have made following changes,
In /platform/cc2538/
UIP_CONF_IPV6 =1
UIP_CONF_IPV6_RPL =1 ,
PANID = 0XABCD
CHANNEL = 26
Any lead ?

Thanks

Devesh24, 2015-01-14:
Any lead ?

Devesh24, 2015-01-19:
Hello Guys 
Any tips or lead ?
@g-oikonomou 
@malvira

malvira, 2015-04-10:
Sorry no clue. Haven't really kept up with the latest or greatest in these matters.

closed 2016-03-01 by laurentderu