cetic/6lbr #68: Connecting 6lbr to Internet..

Krishna421TSP, 2015-01-16:
Hi..Guys..
I have got 6lbr working with my CC2538dk.I am able to open the web server and all the node configuration i could do.
Now i wanted to connect my 6lbr to internet.By using Freenet gogo client i am getting the below ipv6 address....
IPv6 connectivity
IPv6 	Supported 	
Address 	2001:5c0:1000:a::d67 	
Type 	Native IPv6 	
SLAAC 	No 	
ICMP 	Reachable 	
Hostname 	None 	
ISP 	gogo6 Inc. [Canada] 	

But i don't know how to access 6lbr from external network..
Any suggestion can be greatly appreciated..

latapata, 2015-03-06:
hello @Krishna421TSP 
I think you also have to change the IPV6 of 6lbr and the WSN nodes by configuring nvm.dat file to the IPV6 that you have obtained from the gogo client and then your WSN system is accessible.

latapata, 2015-03-12:
@Krishna421TSP I am also trying to accomplish the same. But I am facing another problem that my Internet service provider does not support IPV6. And hence the COAP server cannot be accessed any how. Do you have some Idea how can i acheive this. I know that 6lbr does not support IPV4 but if there is available NAT64 from some other sources also it would be better.

Krishna421TSP, 2015-03-12:
I am also having the same problem my ISP doesn't support ipv6 so i got stucked here.
I tried using gogo client from freenet.com however they provide only local ipv6 addresses for free.

latapata, 2015-03-12:
@Krishna421TSP we are i deadlock. But what I am planning to try is using contiki mist which has already a IPV64 translation mechanism. But unfortunately they do not have coap implemented and we may have to work on implementing coap from udp looking at RFC7252 IETF document. Do you have any better Idea ?

sdawans, 2015-03-12:
Have you considered an IPv6 tunnel? I know this is used by some 6LBR users in similar situation as you

http://en.wikipedia.org/wiki/List_of_IPv6_tunnel_brokers

latapata, 2015-03-12:
Hey sdawans, 
Thanks for the answer. I have already created an account in sixxs.net (tunnel broker) and obtained a tunnel id, username and password. The problem my ISP i.e Duetsche Telekom does not support IPV6 addressing.  I am using a LTE Mobile network with Fritz box wifi router. Can you please point out the user page where they have reported this problem.

sdawans, 2015-03-12:
You may find useful information in the 6lbr-dev Mailing List thread entitled `IPV6 Tunnel via 6lbr to WSN`, I've linked the archives below:

http://lists.cetic.be/pipermail/6lbr-dev/2013-October/thread.html#101

latapata, 2015-03-20:
I cornered out the problem. I am using Frtiz box LTE to Wlan router and that happens to be a problem.
1. I obtained IPV6 from sixxs.net successfully
2. The fritzbox does not support static routing and thats the reason I cannot Route the IPV6 packets to 6lbr border router

closed 2016-03-01 by laurentderu