cetic/6lbr #69: insserv rejected the script header

g-oikonomou, 2015-02-13:
Hi guys. On BeagleBone Black

```
make all_native
make plugins
make tools
```

So far so good. Then

```
$ sudo make install
TARGET not defined, using target 'native'
install -d /etc/6lbr
cp -r package/etc/6lbr /etc
install -d /etc/init.d
install package/etc/init.d/* /etc/init.d
install -d /usr/bin
install package/usr/bin/* /usr/bin
install -d /usr/lib/6lbr
cp -r package/usr/lib/6lbr /usr/lib
install -d /usr/lib/6lbr/bin
install bin/* /usr/lib/6lbr/bin
install  obj_native/dummy.so /usr/lib/6lbr/plugins
install tools/nvm_tool /usr/lib/6lbr/bin
update-rc.d 6lbr defaults
update-rc.d: using dependency based boot sequencing
insserv: warning: script 'S18led_aging' missing LSB tags and overrides
insserv: warning: script 'led_aging.sh' missing LSB tags and overrides
insserv: There is a loop between service rmnologin and mountnfs-bootclean if started
insserv:  loop involving service mountnfs-bootclean at depth 9
insserv:  loop involving service mountnfs at depth 8
insserv: There is a loop between service rmnologin and mountall if started
insserv:  loop involving service mountall at depth 6
insserv:  loop involving service checkroot-bootclean at depth 5
insserv:  loop involving service networking at depth 8
insserv:  loop involving service checkfs at depth 6
insserv: There is a loop between service rmnologin and urandom if started
insserv:  loop involving service urandom at depth 8
insserv:  loop involving service mountall-bootclean at depth 8
insserv: There is a loop between service rmnologin and mountdevsubfs if started
insserv:  loop involving service mountdevsubfs at depth 2
insserv:  loop involving service udev at depth 1
insserv: There is a loop at service rmnologin if started
insserv:  loop involving service hwclock at depth 4
insserv: There is a loop at service led_aging if started
insserv:  loop involving service kbd at depth 9
insserv: There is a loop between service rmnologin and mountall-bootclean if started
insserv: Starting led_aging.sh depends on rmnologin and therefore on system facility `$all' which can not be true!
[ Repeats many times ]
insserv: Max recursions depth 99 reached
insserv:  loop involving service lightdm at depth 1
insserv: There is a loop between service rmnologin and mountnfs if started
insserv: Starting led_aging.sh depends on rmnologin and therefore on system facility `$all' which can not be true!
[ Repeats many times ]
insserv: exiting now without changing boot order!
update-rc.d: error: insserv rejected the script header
make: *** [install] Error 1

```

```
$ uname -a
Linux beaglebone 3.8.13-bone47 #1 SMP Fri Apr 11 01:36:09 UTC 2014 armv7l GNU/Linux
```

```
$ lsb_release -a
No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux 7.4 (wheezy)
Release:	7.4
Codename:	wheezy
```

```
$ git show
commit 7ebf3f03dee7b869aab61e92e812742866168634
Author: Laurent Deru <laurent.deru@cetic.be>
Date:   Thu Sep 11 09:54:52 2014 +0200

    Bump 6LBR version to 1.3.2
[...]
```

g-oikonomou, 2015-02-13:
Which I now realise has nothing to do with 6lbr...

https://groups.google.com/forum/#!topic/beagleboard/LPjCn4LEY2I

closed 2015-02-13 by g-oikonomou

laurentderu, 2015-02-13:
We had the same issue a few days ago, I'll add this link to the FAQ, thanks.