cetic/6lbr #70: cc2538 RPL router not responding

latapata, 2015-02-24:
Hello All,

I am new to this forum. I have followed all the guide to install 6LBR router to my cc2538dk. I am getting proper messages in the UART as below. 

INFO: 6LBR: Tentative local IPv6 address fe80::212:4b00:413:3b0b
INFO: 6LBR: Tentative global IPv6 address (WSN) aaaa::212:4b00:413:3b0b
INFO: 6LBR: Tentative global IPv6 address (ETH) bbbb::100
INFO: 6LBR: RA Daemon enabled
INFO: NVM: Flashing 6LBR NVM
INFO: 6LBR: Configured as DODAG Root
INFO: 6LBR: Starting as RPL ROUTER
INFO: 6LBR: CETIC 6LBR Started

But what should i do to access this RPL from the ubuntu indside instant contiki.

laurentderu, 2015-02-26:
Please see https://github.com/cetic/6lbr/wiki/6LBR-Connection on how to setup your connection with the Border Router. Note that you don't need Instant Contiki to communicate.

latapata, 2015-02-28:
Hello
Thanks a lot for the help. I am now able to see the webpage. I had connected the CS of CC2538 board wrongly. And now it is working properly. Meanwhile I was using VMWARE for instant contiki which create some problems. So i switched it off and disabled all other network connections accept local network and connected the CC2538DK to lan router. After that in chrome in windows i entered the http://[bbbb::100] and now I can see something.

closed 2015-02-28 by latapata