cetic/6lbr #72: trouble accessing sensor from network

lcharpenwi6labs, 2015-03-03:
Hi
First thanks for the great job you are doing on 6lbr!

I work with a RPI and NDP-ROUTER mode. I don't want to use RPL in my application.
I can correctly connect to the web page and I see my devices.
My network address is in fec0:: and my sensor network is in aaaa::
When I try to ping one of the aaaa device from my PC (connected to a ethernet switch and then to the RPI) , I have no response.
I run TCPdump and I activated the logs on the RPI to see what happens. I see that the RPI receives the PING ICMPV6 request but it does nothing with it just like sensors were not its business at all...
I have to add this on my PC to achieve to reach the RPI:
sudo ip -6 addr add aaaa::/64 dev eth0
I don't know if it's mandatory....
Here is my configuration

This file contains a default configuration for Raspberry PI platform using
#a Telos SLIP Radio
#The full list of parameters and their meaning can be found in 6lbr.conf.example

#MODE=ROUTER
#MODE=SMART-BRIDGE
#MODE=RPL-RELAY
#MODE=FULL-TRANSPARENT-BRIDGE
MODE=NDP-ROUTER
#MODE=6LR
#MODE=RPL-ROOT

RAW_ETH=0
BRIDGE=1
DEV_BRIDGE=br0
DEV_TAP=tap0
DEV_ETH=eth0
RAW_ETH_FCS=0

DEV_RADIO=/dev/ttyUSB0
BAUDRATE=115200

LOG_6LBR=/var/log
LOG_6LBR_OUT=$LOG_6LBR/6lbr.log
LOG_6LBR_ERR=$LOG_6LBR/6lbr.err
LOG_LEVEL=5
LOG_SERVICES=ffffffff

Reading nvm file '/etc/6lbr/nvm.dat'
Channel : 26

WSN network prefix : aaaa::
WSN network prefix length : 64
WSN IP address : aaaa::100
WSN accept RA : True
WSN IP address autoconf : True

Eth network prefix : fec0::
Eth network prefix length : 64
Eth IP address : fec0::100
Eth default router : fe80::1
Eth IP address autoconf : False

Local address rewrite : True
Smart Multi BR : False

RA daemon : True
RA router lifetime : 1800
RA maximum interval : 600
RA minimum interval : 200
RA minimum delay : 3
RA PIO enabled : True
RA prefix valid lifetime : 86400
RA prefix preferred lifetime : 14400
RA RIO enabled : True
RA RIO lifetime : 1800

RPL instance ID : 30
RPL Preference : 0
RPL version ID :  : 16
RPL DIO interval doubling : 8
RPL DIO minimum interval : 12
RPL DIO redundancy : 10
RPL default lifetime : 30
RPL minimum rank increment : 256
RPL lifetime unit : 256

Webserver configuration page disabled : False
Writing nvm file '/etc/6lbr/nvm.dat'

Any clue why it is not working?

Thanks for your support

BR

Ludovic

laurentderu, 2015-03-03:
Hello Ludovic,

When you do 
  sudo ip -6 addr add aaaa::/64 dev eth0

you are actually assigning the actual aaaa::0 address on your ethernet interface, and that means your PC assumes the aaaa::/64 subnet is local and not routed by the BR. So when you try to access, for example, the node aaaa::1, it will try to access it directly iso forwarding the request to the RPi and 6LBR.

You can see the correct commands to configure your PC here : https://github.com/cetic/6lbr/wiki/6LBR-Connection

lcharpenwi6labs, 2015-03-03:
Thanks Laurent for this prompt reply.
I will try it ASAP

lcharpenwi6labs, 2015-03-04:
That was exactly what I didn't understand. It works fine!

Thanks for your support

closed 2016-03-01 by laurentderu