cetic/6lbr #73: gpio button event in coap server of cc2538dk not observable

latapata, 2015-03-03:
Hello All,

I have a  6lbr-coap-server on cc2538dk connected to 6lbr-router. I have copper plugin in firefox which I am using to extract information from the coap server. When I observe the GPIO btn 0 in the copper plugin nothing appears. My copper plugin retransmits the message. 

Another problem is when i Press the button all the lights in the harware blink and the board resets. 

If anybody has the screen shot of copper plugin or something please share it across.

latapata, 2015-03-04:
The harware reboot has become abig problem for me

Initially coap-server prints 

Contiki 3.x
TI SmartRF06 + cc2538EM
 Net: sicslowpan
 MAC: CSMA
 RDC: nullrdc
 Channel: 26
 PAN-ID: abcd
Rime configured with address 00:12:4b:00:04:13:36:cb

---> then i press the select button  

Contiki 3.x
TI SmartRF06 + cc2538EM
 Net: sicslowpan
 MAC: CSMA
 RDC: nullrdc
 Channel: 26
 PAN-ID: abcd
Rime configured with address 00:12:4b:00:04:13:36:cb

I have traced the program and have found that this is because of 
REST_FORMAT_ONE_INT("count", button_sensor.value(0))) 
in the button-resource.c and i am not sure about the reason.

Please let me know If i am doing some common mistake.

latapata, 2015-03-04:
@laurentderu This is the new problem which I am moving from thread #64

laurentderu, 2015-03-05:
We don't have a CC2538DK here with buttons, so I can not reproduce your problem. You you try with er-rest-example and see if there is a crash there too ?

latapata, 2015-03-05:
Hello 

Yes I am doing the same. I have cc2538 development board. There is some problem with the coap-server code. I replaced it by er-example-server.c and the LED toggle is working. But the button press observe resource says 'NoFreeTraBuffer' which I am now trying to debug. 

Meanwhile I found out that there is an issue with REST_FORMAT_ONE_INT. which makes board reboot when it is called.

laurentderu, 2015-03-18:
Could you tell me what was the issue with REST_FORMAT_ONE_INT ? I will include the fix into the upcoming 1.3.3.

latapata, 2015-03-18:
Hello @laurentderu 

Thanks for the assistance. Yes when I try My CC2538dk board with er-example-server.c there is no problem of the reboot. But the button press does not work. But other functions like LED toggle and others are working fine. When I look and trace the program i assume a condition that there is reboot because of the REST_FORMAT_ONE_INT. Debugging the program with this I did not have much Idea. But will try and post it later.

closed 2016-03-01 by laurentderu