cetic/6lbr #74: coap client in 6lbr

latapata, 2015-03-13:
Hello All,

I am in a need of er-coap-client on cc2538dk. I have a nodejs coap server running on my PC and I need my cc2538dk client to interact with this over 6lbr network. The cc2538dk just sends a button press information through POST mehtod of COAP. Any help is greatly appreciated.

laurentderu, 2015-03-18:
There is a functional coap client in the develop branch of 6LBR, it supports POST, PUT as well ass asynchronous push and dynamic binding.

latapata, 2015-03-18:
Hello @laurentderu 

Thanks a million for the reply. Where can I find and How can I use it ?

laurentderu, 2015-03-18:
For the time being, it's available in 6lbr-demo/apps/coap. You can have a look at coap-push.c and binding-table.c The next version should be more clear and easier to use.

latapata, 2015-03-18:
I do not see any coap-push.c in 6lbr/examples/6lbr-demo/apps/coap as well as the coap-push.c

sdawans, 2015-03-18:
it's on the ``develop`` branch

https://github.com/cetic/6lbr/tree/develop/examples/6lbr-demo/apps/coap

latapata, 2015-03-18:
@sdawans Thanks a lot !!

shantanoo-desai, 2015-06-23:
Hello I have some similar work. A bunch of CoAP servers keep sending some information like eui,temp and Up time counts to a linux pc which is connected to 6lbr router (in router mode). I need to show the information as of now just on the terminal window or a text file. What could be an optimum solution for make the PC like a SINK which can just keep collecting values?

Thanks.
Shan

the work looks something like this..

[image]

closed 2016-03-01 by laurentderu