cetic/6lbr #75: Unable to access the webserver

darkopetrovic, 2015-03-13:
I followed the 'Hello 6LBR' example with a RPI connected to a CC2538DK as slip-radio using the maint_1.3.2 branch. I'm still unable to access the webserver on `http://[bbbb::100]` from another machine having the address `bbbb::101` (or with wget directly on the RPI).

Here is my configuration and logs:

```text
#MODE=ROUTER
#MODE=SMART-BRIDGE
#MODE=RPL-RELAY
#MODE=FULL-TRANSPARENT-BRIDGE
MODE=NDP-ROUTER
#MODE=6LR
#MODE=RPL-ROOT

RAW_ETH=0
BRIDGE=1
CREATE_BRIDGE=6LBR
ETH_JOIN_BRIDGE=0
DEV_BRIDGE=br0
DEV_TAP=tap0
DEV_ETH=eth0
RAW_ETH_FCS=0

DEV_RADIO=/dev/ttyUSB1
BAUDRATE=115200

LOG_LEVEL=3 #INFO and above only

```

```text
pi@raspberrypi / $ cat /var/log/6lbr.log
********************************************************************************
Fri Mar 13 14:35:35 CET 2015 : Starting 6LBR
/usr/lib/6lbr/bin/cetic_6lbr_ndp_router -c /etc/6lbr/nvm.dat -s /dev/ttyUSB1 -t tap0 -R -B 115200 -U /usr/lib/6lbr/6lbr-ifup -D /usr/lib/6lbr/6lbr-ifdown -w /usr/lib/6lbr/www -L 3  -W /var/log/6lbr.timestamp -P 60 -C /var/log/6lbr.ip  -m /usr/lib/6lbr/plugins  
Contiki-6lbr-1.3.2 started with IPV6
Rime started with address 1.2.3.4.5.6.7.8
MAC CSMA RDC br-rdc NETWORK sicslowpan
Log level : 30
Log services : ffffffff
2015-02-13 14:35:35.218713: INFO: ETH: 6LBR watchdog started (interval: 60)
2015-02-13 14:35:35.220111: INFO: 6LBR: Starting 6LBR version 1.3.2 (Contiki-6lbr-1.3.2)
2015-02-13 14:35:35.220166: INFO: NVM: Opening nvm file '/etc/6lbr/nvm.dat'
2015-02-13 14:35:35.220384: INFO: NVM: NVM Magic : 2009
2015-02-13 14:35:35.220419: INFO: NVM: NVM Version : 1
2015-02-13 14:35:35.220442: INFO: SCMD: Started br-cmd process
2015-02-13 14:35:35.220477: INFO: NODECFG: No node_config.conf file specified
2015-02-13 14:35:35.220511: INFO: ETH: RAW/TAP init
2015-02-13 14:35:35.236557: INFO: SLIP: SLIP started on /dev/ttyUSB1
2015-02-13 14:35:35.257029: INFO: TAP: opened device /dev/tap0
2015-02-13 14:35:35.257437: INFO: TAP: Running 6lbr-ifup script '/usr/lib/6lbr/6lbr-ifup'
6lbr-ifup: Create bridge br0
br0       Link encap:Ethernet  HWaddr 02:0a:0b:0c:0d:0e  
          inet6 addr: fe80::acd0:41ff:fe8d:21c5/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:1 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:0 (0.0 B)  TX bytes:110 (110.0 B)

tap0      Link encap:Ethernet  HWaddr 02:0a:0b:0c:0d:0e  
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:500 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

2015-02-13 14:35:36.108709: INFO: BR-RDC: Reset SLIP Radio
2015-02-13 14:35:36.108932: INFO: BR-RDC: Fetching MAC address
2015-02-13 14:35:36.109554: WARN: PF: No input function set
2015-02-13 14:35:36.110180: WARN: PF: No input function set
2015-02-13 14:35:36.110483: WARN: PF: No input function set
2015-02-13 14:35:36.174247: INFO: BR-RDC: Got MAC: 00:12:4b:00:04:0e:fa:db
2015-02-13 14:35:36.491155: WARN: PF: No input function set
2015-02-13 14:35:36.511146: WARN: PF: No input function set
2015-02-13 14:35:37.108224: INFO: ETH: Eth MAC address : 02:00:04:0e:fa:db
2015-02-13 14:35:37.109149: INFO: 6LBR: Tentative local IPv6 address fe80::212:4b00:40e:fadb
2015-02-13 14:35:37.109727: INFO: 6LBR: Tentative global IPv6 address (WSN) aaaa::212:4b00:40e:fadb
2015-02-13 14:35:37.110315: INFO: 6LBR: Tentative global IPv6 address (ETH) bbbb::100
2015-02-13 14:35:37.110621: INFO: 6LBR: RA Daemon enabled
2015-02-13 14:35:37.110765: INFO: 6LBR: Starting as NDP ROUTER
2015-02-13 14:35:37.121258: INFO: UDPS: UDP server started
2015-02-13 14:35:37.121891: INFO: PLUGIN: Loading /usr/lib/6lbr/plugins/dummy.so
2015-02-13 14:35:37.122785: INFO: PLUGIN: Initialising dummy
2015-02-13 14:35:37.123087: INFO: DUMMY: Dummy init
2015-02-13 14:35:37.123290: INFO: 6LBR: CETIC 6LBR Started
2015-02-13 14:35:37.123572: INFO: SLIP: cc2538_cmd: setting channel: 26
```

```text
pi@raspberrypi /usr/lib/6lbr/bin $ ./nvm_tool --print /etc/6lbr/nvm.dat 
Reading nvm file '/etc/6lbr/nvm.dat'
Channel : 26

WSN network prefix : aaaa::
WSN network prefix length : 64
WSN IP address : aaaa::100
WSN accept RA : True
WSN IP address autoconf : True

Eth network prefix : bbbb::
Eth network prefix length : 64
Eth IP address : bbbb::100
Eth default router : ::
Eth IP address autoconf : False

Local address rewrite : True
Smart Multi BR : False

RA daemon : True
RA router lifetime : 0
RA maximum interval : 600
RA minimum interval : 200
RA minimum delay : 3
RA PIO enabled : True
RA prefix valid lifetime : 86400
RA prefix preferred lifetime : 14400
RA RIO enabled : True
RA RIO lifetime : 1800

RPL instance ID : 30
RPL Preference : 0
RPL version ID :  : 253
RPL DIO interval doubling : 8
RPL DIO minimum interval : 12
RPL DIO redundancy : 10
RPL default lifetime : 30
RPL minimum rank increment : 256
RPL lifetime unit : 256

Webserver configuration page disabled : False
```

Why the `eth0` interface doesn't get an IPv6 address? I tried to add it manually but I'm still unable to access the webserver. Note that I can ping the address `bbbb::100` from the other machine. 

Should we join the `eth0` to the `br0` interface with `ETH_JOIN_BRIDGE=1` in order to make it work? If I activate this option I loose SSH my connection. Could we bypass that if the join required? 

What is the real meaning of `br0` and `tap0` interface? 

Many thanks in advance for your help!

sdawans, 2015-03-13:
I believe this issue is that you are using `CREATE_BRIDGE=6LBR`, where 6lbr creates br0 at 6lbr boot. When eth0 is then added to the bridge, its configuration gets cleared. You should create the bridge manually rather than letting 6LBR do it, with 

    CREATE_BRIDGE=0

And setting an appropriate configuration in /etc/network/interfaces to create the bridge when you start the Linux host. Here are examples of that:

* https://github.com/cetic/6lbr/blob/master/examples/6lbr/package/etc/6lbr/interfaces.dhcp.example
* https://github.com/cetic/6lbr/blob/master/examples/6lbr/package/etc/6lbr/interfaces.static.example

For your last question, br0 is a bridge to connect the physical eth0 to 6lbr (tap0).

That's about all I know

darkopetrovic, 2015-03-13:
Thanks for your answer!

It works now. But I don't know why. So, I've setup the bridge at startup with:

```text
auto lo

iface lo inet loopback
iface eth0 inet static
address 0.0.0.0

auto br0
iface br0 inet static
    pre-up brctl addbr br0
    pre-up brctl addif br0 eth0
    pre-up ip addr flush dev eth0
    post-down ip link set eth0 down
    post-down ip link set br0 down
    post-down brctl delif br0 eth0
    post-down brctl delbr br0
    address 153.109.5.42
    netmask 255.255.255.0 
    network 153.109.5.0
    broadcast 153.109.5.255
    gateway 153.109.5.1
    dns-nameservers 153.109.10.11

allow-hotplug wlan0
iface wlan0 inet manual
wpa-roam /etc/wpa_supplicant/wpa_supplicant.conf
iface default inet dhcp
```

I still don't get well how this bridge is working since I've to configure the static address on `br0` and not on `eth0`. With this config I can access normally the RPI with SSH but I cannot access if I set the static address on `eth0`. 

Secondly, I had to configure manually the `bbbb::100` address on either `br0` or `eth0` to access the webserver from another machine. Why is it not set automatically? What is the correct practice if I've to set it manually?

sdawans, 2015-03-13:
When you have an interface attached to the bridge, the bridge sort of takes control and you have to apply all configuration to the bridge, not the interface. I think that's how bridge-utils is designed, but I'm not much of an expert. Maybe @laurentderu might be able to confirm I'm not spitting too much BS.

For your second question about the `bbbb:: prefix` - normally this info would come from an RA packet from an IPv6 router. If you don't have one you could also run radvd on the Linux host to emulate the Router Advertisement. Again, it's been a while for me so this needs to be confirmed ;)

Hope it helps

darkopetrovic, 2015-03-17:
Thanks for your answer. 

Is the `radvd` tool equally required for the WSN side? Since I'm not using RPL and I configured 6LBR in NDP-ROUTER mode, should it send RA messages on the WSN? Its seems that it doesn't. My nodes doesn't receive any RA message after requesting it with RS messages. Do you know what means `RA daemon : True` configuration in the `nvm.dat` file? More generaly I'm asking if we can use the 6LBR without RPL? The documentation about the NDP-ROUTER is lacking.

laurentderu, 2015-03-18:
Hi,

As Sébastien correctly said, the bridge is used to connect the 6LBR interface, tap0, to the Ethernet interface, ether. However, when you attach an interface to a bridge it loses all its configuration, so when the Ethernet interface joins the bridge, it loses its addresses and the external connectivity, that's why it's better to manually create the bridge and use it as the main interface of the RPi.

About the NDP-Router mode, it should send RA to the WSN side by default (that's what the RA daemon : True means), but you also must set the RA Router Lifetime to a non zero value, or the nodes will not use it as default router.

When created, the NDP Router was meant to be only a test mode, that's why the documentation is lacking for this mode. Now it is part of the supported mode (and tested in our non-regression tests), so I should really update the doc :)

darkopetrovic, 2015-03-18:
Thanks for your feedback! By looking at the code I was finally able to understand that. Now as you may have seen I'm on the issue #76 which explain why the sensor doesn't receive actually this RA message :)

closed 2016-03-01 by laurentderu