cetic/6lbr #76: RDC problem with C2538

bregell, 2015-03-18:
I am trying to setup a network of OpenMote nodes running a Erbium server so I can access their sensor data and everyting works fine as long as I use the **nullrdc** RDC mode on both 6lbr in router mode and the nodes.

However when I try to setup the same network with **contikimac** on both 6lbr and the nodes the mesh network wont configure. In most cases I can reach the first node in the DODAG but nodes after that is unreachable. 
The routing table can look like this:
```
Routes
[del] aaaa::212:4b00:60d:9b57/128 via fe80::212:4b00:60d:9ac3 7668 s
[del] aaaa::212:4b00:60d:9b59/128 via fe80::212:4b00:60d:9ac3 7659 s
[del] aaaa::212:4b00:60d:9abb/128 via fe80::212:4b00:60d:9abb 7600 s
[del] aaaa::212:4b00:60d:9ac3/128 via fe80::212:4b00:60d:9ac3 7586 s
```

Any takes on how to solve this problem?

darkopetrovic, 2015-03-18:
Such a coincidence, I was going to ask almost the same question. I was testing 6lbr and noticed that it use nullrdc, which is logic. My previous configuration was to use the standard `rpl-border-router` provided in the contiki example, and this later is using contikimac when we use it with the CC2538dk platform at least. I experience now the same problem as you where my sensor doesn't receive any packet from 6lbr due to the use of nullrdc. But very rarely, it can receive the packet! 

The explanation is simple: due to the radio duty cycle the sensor miss the packet because the radio is turned off most of the time. The border-router need in this case to send a large packet in order that the sensor can see it. Contikimac is doing that but it seems that nullrdc doesn't take care of that. This is why I guess in my case the combination of contikimac on sensor and nullrdc on border-router fails. I was wondering how you guys are dealing with that?

It is strange however that in your case with contikimac on both devices this doesn't work. The border-router should in any case use contikimac since it need to remains constantly on to receive a packet at any time. Furthermore, with contikimac, when a packet is broadcasted, it'll be transmitted my times in order that every sensor can see it (looks at the STROBE_TIME parameter). In the other case, for a direct link transmission, only the packet length matter I guess. 

I would equally much appreciate if we could get more insight on this.

bregell, 2015-03-18:
Well from what I have figured out by reading RDCs should not be mixed, for example **nullrdc** does not strobe packages when sending and thus nodes with **conticimac** might miss those packages.

But it should work when all receivers are using the same RDC but in my case this does not seem to work.

I can sniff all the packages and when using nullrdc almost 100% of the packages are OK, but when using contikimac I get a lot of bad FCS or malformed packages! 
For sniffing i use an CC2531 USB Dongle, ccsniffpiper and Wireshark.

laurentderu, 2015-03-18:
6LBR is using NullRDC by default as it is the easiest and less error prone RDC to start with :) We should make it clearer on the Wiki.

It seems there were bugs in the CC2538 RF driver (see https://github.com/contiki-os/contiki/pull/946) that break ContikiMAC, could you try merging these fixes and see if it solves your problems ?

bregell, 2015-03-18:
I have pulled that fix into my own contiki fork but now when you mention it i have not done the same thing for the 6lbr repo. I will try this and see what happens. Have anyone on the cetic team or any other developer got contikimac to work properly with CC2538 and contikimac in multiphop networks?

laurentderu, 2015-03-18:
We never tried ContikiMAC on cc2538 on our testbed unfortunately. On the other hand, it seems that Drowsie works quite well on CC2538 and 6LBR.

darkopetrovic, 2015-03-18:
I just tried the mentioned fix on the default rpl-border-router configured with nullrdc (instead of contikimac). Although this seems to be a important fix we need to implement for those using the cc2538 chip, this actually doesn't have resolved the current problem. I really suspect that the problem is rather on the device running contikimac which is unable to receive a message send from a nullrdc driver => no strobe in case of broadcast, no shortest packet length test (SHORTEST_PACKET_SIZE parameter in contikimac.c).

bregell, 2015-03-18:
I will try this Drowsie thing. Cant be harder than adding it as an app in contiki or even put it into the mac folder!

bregell, 2015-03-19:
I modified and put the drowsie driver in the mac folder of contiki and put it as the RDC driver for both 6lbr and my erbium project and now things work like a charm!

darkopetrovic, 2015-03-23:
When we use 6lbr on a Raspberry PI (connected with a slip-radio hardware), by default the used RDC is `border_router_rdc_driver`, right? How are we supposed to use the LPL mechanism on a battery powered nodes (no matter what is the platform) if this RDC are not strobing the packet neither it send a long wake-up tone? :) 

We are forced to port 6lbr directly on the platform and use a different RDC driver that the native one?

laurentderu, 2015-03-25:
When using 6LBR on a RPi (or any Linux platform) as you said the actual RDC is located inside the slip-radio, 'border_router_rdc_driver' is only a wrapper used to communicate with the slip-radio.

bregell, 2015-03-30:
So I have now tried both ContikiMAC, Drowsie and CX-MAC.
* ContikiMAC
  * Does not seem to work at all
* Drowsie 
  * Works good in when LPM >= 0
  * When LPM < 0, 1 hop 50%, 2 hops 30%, 3 hops 10% 
* CX-MAC
  * Works good in when LPM >= 0
  * When LPM < 0, 1 hop 80%, 2 hops 60%, 3 hops 50% 
I have tried to understand why the LPM level alters how good the connection is but the only thing I can think of is that when the device drops into Sleep Mode the SRAM data is lost. Is there some way of solving that problem?

darkopetrovic, 2015-03-30:
In the CC2538, when LPM >= 2, only the second half of the SRAM is retained. We've to tell the linker where the non-retained SRAM is located with the following code in the linker command file (cc2538.ld): 

```
NRSRAM (RWX) : ORIGIN = 0x20000000, LENGTH = 0x00004000

.nrdata (NOLOAD) :
{
    _nrdata = .;
    *(.nrdata*)
    _enrdata = .;
} > NRSRAM
```

Have a look on the generated .ld script, this should be automatic when you set `LPM_CONF_MAX_PM=2`. 

Meanwhile, how did you achieve a LPM < 0 ? :)

bregell, 2015-03-30:
So I'm no expert on linking but I modified the SRAM part of cc2538.lds to this:
    
    #define SRAM_START      0x20004000
    #define SRAM_LEN        0x00004000

Also removed line 66-68 and 109-116 to make all data go to the part of SRAM that is kept while sleeping. All though when loading the .elf onto the chip it still puts all the data like this:
```
Downloading 4096 bytes @ address 0x00200000
Downloading 4096 bytes @ address 0x00201000
Downloading 4096 bytes @ address 0x00202000
Downloading 4096 bytes @ address 0x00203000
Downloading 4096 bytes @ address 0x00204000
Downloading 4096 bytes @ address 0x00205000
Downloading 4096 bytes @ address 0x00206000
Downloading 4096 bytes @ address 0x00207000
Downloading 4096 bytes @ address 0x00208000
Downloading 4096 bytes @ address 0x00209000
Downloading 4096 bytes @ address 0x0020A000
Downloading 4096 bytes @ address 0x0020B000
Downloading 4096 bytes @ address 0x0020C000
Downloading 4096 bytes @ address 0x0020D000
Downloading 4096 bytes @ address 0x0020E000
Downloading 2188 bytes @ address 0x0020F000
Downloading 1898 bytes @ address 0x0020F88C
Downloading 8 bytes @ address 0x0020FFF8
Downloading 44 bytes @ address 0x0027FFD4
Writing register (PC = 0x00200000)
Read 4 bytes @ address 0x00200000 (Data = 0x20004FC0)
```
For me this means that it still puts data at the wrong memory address!

LPM < 0 
I set the LPM level to 2 and set the SILENT_MODE to 1.

darkopetrovic, 2015-03-30:
This is effectively strange. You have to check the generated .map file to see where the data are effectively located in the memory. Normally you should have the .bss section like below in your linker script. This is enough I guess to tell the linker where to put the uninitialized data in the SRAM. But I'm not an expert neither. 

```
.bss :
{
    _bss = .;
    *(.bss*)
    *(COMMON)
    _ebss = .;
} > SRAM
```

With SRAM defined like you said to:

`SRAM (RWX) : ORIGIN = 0x20004000, LENGTH = 0x00004000`

I don't know further.

laurentderu, 2015-04-01:
The ELF loader loads at the right address, 0x00200000 is the start of the Flash zone, the SRAM is at 0x20000000 The data are copied from the Flash to the SRAM by the bootstrap code at startup, maybe the bug is there.

vgangal101, 2016-02-24:
Do any of you have instructions on how to setup erbium on the raspberry pi? Especially as a CoAP server ?

Radmand, 2016-02-24:
You can use libcoap

vgangal101, 2016-02-25:
is there a java implementation

shantanoo-desai, 2016-02-26:
You can find the details for CoAP implementations on the [coap.technology](https://coap.technology) website

closed 2016-03-01 by laurentderu