cetic/6lbr #77: CC2538 + OpenMote out of memory

bregell, 2015-03-24:
When I'm running 6lbr in router mode on my OpenMote + OpenBase config I get this error:
```
ERROR: NODE: Not enough memory to create node
``` 
What does it mean and where is it coming from?

laurentderu, 2015-03-25:
That means that the node-info module has reached the maximum number of entries it can manage. This number is set to UIP_DS6_ROUTE_NB, the maximum number of routes, which should be 100 by default.

If you have more that 100 nodes, then you have to increase the value in 6lbr-conf-openmote.h.

If not, there is a bug, either UIP_DS6_ROUTE_NB is too small, or you have nodes changing their IP address resulting in too many routes.

bregell, 2015-03-25:
Well I only have 4 nodes on the 802.15.4 net with static IPv6 addresses.
In the webserver I have 4 routes and ~10 neighbors so I don't know why the error is produced.

laurentderu, 2015-03-25:
Looking at the code, it seems that we do dot distinguish between no more place and node already created, so I think this is just a spurious error message.

laurentderu, 2015-03-25:
I pushed a fix on the develop branch, could you try it ?

bregell, 2015-03-30:
I will try it asap! Will keep you posted.

bregell, 2015-03-30:
Haven't seen the error during today's testing so I guess the problem is solved! Tkx M8!

closed 2015-04-06 by bregell