cetic/6lbr #79: ICMPv6 Source link-layer option address error

arurke, 2015-05-13:
I am having some problems with 6lbr and the ICMPv6 Source link-layer option in Neighbor Solicitations originating from the 6lbr GW. The problem exists after startup, and then seems to fix itself after some minutes. To reproduce, I reboot the 6lbr, wait a minute or two, and then power-on one contiki node (CC2650 sensortag bleeding-edge contiki).

On the air I see the following:

1. Neighbor solicit:
6lbr (ip: fe80::212:4b00:2f1:d676) -> node (ip: fe80::212:4b00:687:ab80)
Source link-layer option: Length 2, Address 00:12:4b:00:02:f1:d6:76

2. Misc. data from node to 6lbr with correct L2 destination addr.: 00:12:4b:00:02:f1:d6:67

3. Neighbor solicit:
6lbr (ip: fe80::212:4b00:2f1:d676) -> node (ip: **fe80::1:ff87:ab80**)
Source link-layer option: **Length 1, Address 02:00:02:f1:d6:67**

4. Misc. data from node to 6lbr with incorrect L2 destination addr.: ***02:00:02:f1:d6:766e:69***
From this point the node will use this L2 address (2 last bytes changes, indicating it is some overflow), and no data will be received. Many identical solicit as in # 3 is transmitted.

5. Neighbor solicit:
6lbr (ip: fe80::212:4b00:2f1:d676) -> node (ip: fe80::212:4b00:687:ab80)
Source link-layer option: Length 2, Address 00:12:4b:00:02:f1:d6:76

At this point the node will use correct L2 address again, and traffic will resume. This whole procedure last a few minutes. 

I'm using 6lbr on a Beaglebone black, reproduced on both 1.3.2 and 1.3.3. I have blindly followed the configuration steps outlined here: http://processors.wiki.ti.com/index.php/Contiki-6LOWPAN-BBB (Smart bridge mode combined with Jool and Bind for NAT64/DNS64)
Also tested with aaaa:: instead of 2001:etc.

Captures, configuration, logs etc. can be provided if needed.

Any hints appreciated,
Regards,
Andreas

laurentderu, 2015-05-13:
This is a strange problem (thank you for reporting it), it seems that, for a few moment, the border router uses it's Ethernet MAC address instead of the 802.15.4 MAC address. 02:00:02:f1:d6:67 is the generated Ethernet MAC address, created from the 802.15.4 (The two middle bytes are dropped and the administrative flag is set to 1).
Normally, the border router stack is inactive until the system is properly configured, but there might be a race condition somewhere. The border router believe that the NA is for the Ethernet segment (and so include the Ethernet MAC address) but then send it to the 802.15.4 interface.
Could you send me the 6lbr.log file with debug level set to 10, so I'll be able to see what the stack is doing.

Thanks a lot,
Laurent.

arurke, 2015-05-14:
Please find the log here: https://gist.github.com/arurke/98485a94dc42cb004863

Procedure was: Clean log, reboot BBB, wait approx 1 min, power-on node.

From capture I see the 6lbr transmit neighbor solicitation (NS) with wrong source-L2. But for some reason I was not able to reproduce the complete issue where the Contiki-node transmit to wrong L2 address, and 6lbr later transmit NS with correct L2. Maybe this is some race on the contiki - who populates the neighbor L2 address first. Either way I think the NS is what you were looking for.

Note my prefix is aaaa:: not 2001:etc as in the guide.

Hope this helps, please let me know if you need more.

laurentderu, 2015-05-21:
I believe I have found the issue, when 6LBR sends a broadcast message without knowing which subnet is the target, it sends it towards both interface, radio and Ethernet. But it seems that sometimes the packet is still pending in the Radio interface queue when it is sent to the Ethernet interface. So, when the Ethernet interface update the MAC address it corrupts the packet sent on the Radio.

Once I'm able to reproduce the problem, I will issue a fix.

laurentderu, 2015-05-22:
I pushed the modification onto the maintenance branch https://github.com/cetic/6lbr/tree/dev-1.3.x

arurke, 2015-08-24:
I finally found time to re-test this, and I am unable to reproduce it on 1.3.x - good work.

laurentderu, 2015-08-25:
Thanks for the feedback !

closed 2015-08-25 by laurentderu