cetic/6lbr #82: How to use 6lbr as rpl root with COOJA slip radio?

tarunkshk, 2015-05-28:
Hello,

I am new to contiki. I am using 6LBR & Simulated WSN example but i am confused about, where to give following commands
sysctl -w net.ipv6.conf.br0.accept_ra=1 
sysctl -w net.ipv6.conf.br0.accept_ra_rt_info_max_plen=64
or manually add the route :
route -A inet6 add aaaa::/64 gw bbbb::100

sdawans, 2015-05-28:
Assuming you are following [this example](https://github.com/cetic/6lbr/wiki/COOJA-Interface), you are interfacing a 6LBR with a simulated WSN. Both 6LBR and COOJA run on the same Linux host. Those commands should be typed in a console on the Linux PC. The first two allow 6LBR's Router Advertisements to not get dropped by Linux. The last command is just cheating and adding the route manually if you can't get the stack to process the RAs for whatever reason.

tarunkshk, 2015-05-29:
Thanks done

closed 2015-05-29 by tarunkshk