cetic/6lbr #85: Ping a node directly from the border-router (Raspberry PI)

darkopetrovic, 2015-06-16:
I'm using 6LBR in NDP-ROUTER mode on a Raspberry PI with a CC2538DK slip-radio. The configuration file is as follow:

```
#MODE=ROUTER
#MODE=SMART-BRIDGE
#MODE=RPL-RELAY
#MODE=FULL-TRANSPARENT-BRIDGE
MODE=NDP-ROUTER
#MODE=6LR
#MODE=RPL-ROOT

RAW_ETH=0
BRIDGE=1
CREATE_BRIDGE=0
DEV_BRIDGE=br0
DEV_TAP=tap0
DEV_ETH=eth0
RAW_ETH_FCS=0

DEV_RADIO=/dev/ttyUSB1
BAUDRATE=115200

LOG_LEVEL=3 #INFO and above only
```

Interfaces on the Raspberry PI:

```
br0       Link encap:Ethernet  HWaddr 02:0a:0b:0c:0d:0e
          inet addr:153.109.5.42  Bcast:153.109.5.255  Mask:255.255.255.0
          inet6 addr: fe80::ba27:ebff:fe48:fcd6/64 Scope:Link
          inet6 addr: bbbb::100/64 Scope:Global
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:26852917 errors:0 dropped:509352 overruns:0 frame:0
          TX packets:96322 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:3130718188 (2.9 GiB)  TX bytes:11075788 (10.5 MiB)

eth0      Link encap:Ethernet  HWaddr b8:27:eb:48:fc:d6
          inet6 addr: 2001::42/64 Scope:Global
          inet6 addr: fe80::ba27:ebff:fe48:fcd6/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:30059802 errors:0 dropped:1 overruns:0 frame:0
          TX packets:185431 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:3379799107 (3.1 GiB)  TX bytes:23734977 (22.6 MiB)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:16436  Metric:1
          RX packets:537 errors:0 dropped:0 overruns:0 frame:0
          TX packets:537 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:74825 (73.0 KiB)  TX bytes:74825 (73.0 KiB)

tap0      Link encap:Ethernet  HWaddr 02:0a:0b:0c:0d:0e
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:19180 errors:0 dropped:0 overruns:0 frame:0
          TX packets:2312745 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:500
          RX bytes:2442735 (2.3 MiB)  TX bytes:317693619 (302.9 MiB)
```

I can ping a node from a windows computer on the same network but not directly from the Raspberry PI. I tried to add the the route with the command bellow but without success.

    sudo route -A inet6 add aaaa::/64 dev br0

**I still have the following error:**

```
pi@raspberrypi ~ $ ping6 aaaa::212:4b00:433:eee6
PING aaaa::212:4b00:433:eee6(aaaa::212:4b00:433:eee6) 56 data bytes
From 2001::42 icmp_seq=1 Destination unreachable: Address unreachable
From 2001::42 icmp_seq=2 Destination unreachable: Address unreachable
From 2001::42 icmp_seq=3 Destination unreachable: Address unreachable
```

Here some other configuration stuff:

```
pi@raspberrypi ~ $ route -n6
Kernel IPv6 routing table
Destination                    Next Hop                   Flag Met Ref Use If
2001::/64                      ::                         U    256 0     0 eth0
aaaa::/64                      ::                         U    1   0     0 br0
bbbb::/64                      ::                         U    256 0     1 br0
...
```

```
pi@raspberrypi ~ $ brctl show
bridge name     bridge id               STP enabled     interfaces
br0             8000.020a0b0c0d0e       no              eth0
                                                        tap0
```

```
pi@raspberrypi ~ $ sudo ebtables -L
Bridge table: filter

Bridge chain: INPUT, entries: 0, policy: ACCEPT

Bridge chain: FORWARD, entries: 0, policy: ACCEPT

Bridge chain: OUTPUT, entries: 0, policy: ACCEPT

```

```
pi@raspberrypi ~ $ sudo iptables -L
Chain INPUT (policy ACCEPT)
target     prot opt source               destination

Chain FORWARD (policy ACCEPT)
target     prot opt source               destination

Chain OUTPUT (policy ACCEPT)
target     prot opt source               destinatio
```

```
pi@raspberrypi ~ $ sysctl net.ipv6.conf.default.forwarding
net.ipv6.conf.default.forwarding = 1
```

Does anyone have any ideas how can I go further?

laurentderu, 2015-06-16:
Hello,

I thing the Linux stack does not get the next hop for aaa::/64 right. You should try instead the following command :

  sudo route -A inet6 add aaaa::/64 gw <6lbr bbbb address>

Also, you probably have an address conflict between br0 and 6lbr if you have the default configuration, as they are using the same address bbbb::100

darkopetrovic, 2015-06-16:
I don't know why I thought the bridge must take the address configured for the border-router. 

Both of your suggestions resolved my problem. 

Thanks a lot!

closed 2016-03-01 by laurentderu

ashutalewar, 2016-07-20:
I am facing similar issue.  I didn't get the solution properly. Can you explain a little bit more.  
In my case the routes are like these:

```
pi@bananapim2plus:~$ sudo route -n6
Kernel IPv6 routing table
Destination                    Next Hop                   Flag Met Ref Use If
::1/128                        ::                                    U    256 0     0 lo
aaaa::/64                      fe80::7ff:ff68:ce5d        UGAe 1024 0     0 br0
bbbb::/64                      ::                                  UAe  256 0     4 wlan0
bbbb::/64                      ::                                  UAe  256 0     1 br0
fe80::/64                      ::                                   !n   256 0     0 lo
fe80::/64                      ::                                   U    256 0     0 wlan0
fe80::/64                      ::                                   U    256 0     0 br0
..
..

```

darkopetrovic, 2016-07-25:
No your routes should look like this:

```
pi@raspberrypi ~ $ sudo route -n6
Kernel IPv6 routing table
Destination                    Next Hop                   Flag Met Ref Use If
aaaa::/64                      bbbb::100                  UG   1   0     0 br0
bbbb::/64                      ::                         U    256 0     1 br0
```

First add an ipv6 address to the bridge: 

    sudo ip addr add bbbb::101/64 dev br0

Then add the route to `aaaa::/64` via the 6lbr:

    sudo route -A inet6 add aaaa::/64 gw bbbb::100

ashutalewar, 2016-07-27:
I am using normal ROUTER mode. The routes that I mentioned in the previous comment was automatically added by 6lbr service. 

I also tried adding the routes suggested by you but still the ping6 to nodes is not responding.

jchari, 2017-09-07:
Hi,
i am able to ping fd00::fd00::b00:f6ff:71c6:8e39 wsn address. but i tried to ping node's in wsn ping becoming not reachable/ waiting for replay.

how to make route the nodes in my host machine.?