cetic/6lbr #86: Unabe to use Wrapsix with 6lbr on raspberrypi

deepikay, 2015-06-17:
Hi laurentderu,

I am trying to upload z1 data to ubidots.com by following this link :

https://github.com/marcozennaro/IPv6-WSN-book/blob/master/5.asc

I installed 6lbr on raspberrypi, connected it with a slip-radio loaded z1 mote and,
To use NAT64 with Wrapsix then I installed wrapsix on raspberry, modifed wrapper.c with necessary configuration:

when I start wrapsix as daemon I get :

[1] 3039
[Info] WrapSix 0.2.0 is starting
pi@raspberrypi ~/wrapsix-0.2.0/src $ [Error] Unable to create listening socket

sudo service 6lbr restart

Logs:

*******************************************************************************
Tue Jun 16 09:52:34 UTC 2015 : Starting 6LBR
/usr/lib/6lbr/bin/cetic_6lbr_router -c /etc/6lbr/nvm.dat -s /dev/ttyUSB0 -t eth0 -r -B 115200 -U /usr/lib/6lbr/6lbr-ifup -D /usr/lib/6lbr/6lbr-ifdown -w /usr/lib/6lbr/www -L 3  -W /var/log/6lbr.timestamp -P 60 -C /var/log/6lbr.ip  -m /usr/lib/6lbr/plugins
Contiki-6lbr-1.3.2-34-g3715b49 started with IPV6, RPL
Rime started with address 1.2.3.4.5.6.7.8
MAC CSMA RDC br-rdc NETWORK sicslowpan
Log level : 30
Log services : ffffffff
2015-05-16 9:52:34.465054: INFO: ETH: 6LBR watchdog started (interval: 60)
2015-05-16 9:52:34.466461: INFO: 6LBR: Starting 6LBR version 1.3.3 (Contiki-6lbr-1.3.2-34-g3715b49)
2015-05-16 9:52:34.466502: INFO: NVM: Opening nvm file '/etc/6lbr/nvm.dat'
2015-05-16 9:52:34.475495: INFO: NVM: NVM Magic : 2009
2015-05-16 9:52:34.475550: INFO: NVM: NVM Version : 1
2015-05-16 9:52:34.475574: INFO: SCMD: Started br-cmd process
2015-05-16 9:52:34.475598: INFO: NODECFG: No node_config.conf file specified
2015-05-16 9:52:34.475620: INFO: ETH: RAW/TAP init
2015-05-16 9:52:34.478330: INFO: SLIP: SLIP started on /dev/ttyUSB0
2015-05-16 9:52:34.500564: INFO: TAP: opened device /dev/eth0
2015-05-16 9:52:34.501371: INFO: TAP: Running 6lbr-ifup script '/usr/lib/6lbr/6lbr-ifup'
eth0      Link encap:Ethernet  HWaddr b8:27:eb:33:e7:28
          inet addr:192.168.10.5  Bcast:192.168.10.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:3531 errors:0 dropped:0 overruns:0 frame:0
          TX packets:1726 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:627824 (613.1 KiB)  TX bytes:266945 (260.6 KiB)

2015-05-16 9:52:34.870443: INFO: TAP: Eth MAC address : b8:27:eb:33:e7:28
2015-05-16 9:52:34.870684: INFO: BR-RDC: Reset SLIP Radio
2015-05-16 9:52:34.870810: INFO: BR-RDC: Fetching MAC address
2015-05-16 9:52:34.904707: ERROR: SLIP: Unknown command
2015-05-16 9:52:34.935126: INFO: BR-RDC: Got MAC: c1:0c:00:00:00:00:13:e8
2015-05-16 9:52:35.870855: INFO: 6LBR: Tentative local IPv6 address fe80::c30c:0:0:13e8
2015-05-16 9:52:35.871634: INFO: 6LBR: Tentative global IPv6 address (WSN) aaaa::c30c:0:0:13e8
2015-05-16 9:52:35.872290: INFO: 6LBR: Tentative global IPv6 address (ETH) bbbb::100
2015-05-16 9:52:35.872644: INFO: 6LBR: RA Daemon enabled
2015-05-16 9:52:35.873114: INFO: NVM: Opening nvm file '/etc/6lbr/nvm.dat'
2015-05-16 9:52:35.874336: INFO: 6LBR: Configured as DODAG Root
2015-05-16 9:52:35.874611: INFO: 6LBR: Starting as RPL ROUTER
2015-05-16 9:52:35.877235: INFO: UDPS: UDP server started
2015-05-16 9:52:35.878332: INFO: 6LBR: CETIC 6LBR Started
2015-05-16 9:52:35.878704: ERROR: SLIP: Unknown command

In my browser I able to see border router page : [bbbb::100]

After this when I run ubidots_demo.c on another z1 mote I get following logs: 

Starting 'Ubidots demo process'
Ubidots client: STATE_ERROR_NO_NET
Ubidots client: STATE_ERROR_NO_NET
Ubidots client: STATE_ERROR_NO_NET
Ubidots client: STATE_ERROR_NO_NET
Ubidots client: STATE_ERROR_NO_NET
Ubidots client: STATE_ERROR_NO_NET
Ubidots client: STATE_ERROR_NO_NET
Ubidots client: STATE_ERROR_NO_NET
Ubidots client: STATE_ERROR_NO_NET
Ubidots client: STATE_ERROR_NO_NET
Ubidots client: STATE_ERROR_NO_NET
Ubidots client: STATE_STARTING
Ubidots client: Checking 64:ff9b::3217:7c44
Ubidots client: 'Host: [64:ff9b::3217:7c44]' (remaining 44)
Ubidots client: STATE_TCP_CONNECT (1)
Ubidots client: Connect 64:ff9b::3217:7c44 port 80
event_callback: disconnected
Ubidots client: STATE_DISCONNECTED
Ubidots client: STATE_TCP_CONNECT (2)
Ubidots client: Connect 64:ff9b::3217:7c44 port 80
event_callback: disconnected
Ubidots client: STATE_DISCONNECTED
Ubidots client: STATE_TCP_CONNECT (3)
Ubidots client: Connect 64:ff9b::3217:7c44 port 80
event_callback: disconnected
Ubidots client: STATE_DISCONNECTED
Ubidots client: STATE_TCP_CONNECT (4)
Ubidots client: Connect 64:ff9b::3217:7c44 port 80

Any suggestions would be helpful 

Thanks!!

laurentderu, 2015-06-17:
I never used wrapsix, but I guess the problem is coming from : 

```
 [1] 3039
[Info] WrapSix 0.2.0 is starting
pi@raspberrypi ~/wrapsix-0.2.0/src $ [Error] Unable to create listening socket
```

Without a listening socket, you can not have NAT64 and connectivity.

Also, what could be missing is the configuration of the default router on 6LBR, without that, it can not route the 64:ff9b:: traffic towards WrapSix

deepikay, 2015-06-17:
Raspberry pi is connected to a router by ethernet cable and my laptop also connected to router through ethernet

My 6lbr.conf is like this :

MODE=ROUTER
RAW_ETH=1
BRIDGE=0
DEV_BRIDGE=br0
DEV_TAP=tap0
DEV_ETH=eth0
RAW_ETH_FCS=0
DEV_RADIO=/dev/ttyUSB0
BAUDRATE=115200
LOG_LEVEL=3 #INFO and above only

My wrapper.c looks like :
#define INTERFACE       "eth0"
#define PREFIX          "64:ff9b::"
#define IPV4_ADDR       " x.x.x.x" //unused ip in my network
#define HOST_IPV6_ADDR  "bbbb::101"
#define HOST_IPV4_ADDR  "x.x.x.x"// ip address of raspberry pi as mentioned in ifcocnfig

Where do I need to add default router?

laurentderu, 2015-06-19:
You can add the default route in the configuration page of 6LBR (http://[bbbb::100]/config.html), but first check if WrapSix is correctly started

weijian19391, 2015-07-03:
@deepikay in case your issue isn't solve, u can do the following to start wrapsix:
```` sudo wrapsix````

@laurentderu what is the default route that we should set? and what do u mean by routing traffic towards wrapsix?

laurentderu, 2015-07-03:
@weijian19391 I never used wraps, so it's only a guess, I think you have to tell 6lbr where to send the data for wrapsix and so define a route towards its endpoint. But maybe wrapsix simply sniff the traffic and you don't need that route :)

closed 2016-03-01 by laurentderu