cetic/6lbr #88: IPv6 mote communication to internet without border router

deepikay, 2015-06-25:
Hi All,

I am a new to IoT area and wants to know how can I let my z1 mote communicate with my smartphone or PC over Wi-Fi. I don't want any border-router in between.

Most of the examples I see, have a border-router connected over USB to PC mediating communication between other sensor nodes and outside world. Because in this case purpose of IoT is not met, its like simple wireless networks, many nodes communicating to gateway.

Can't it connect by its link local IPv6 address to a nearby smartphone like other devices having IPv4 connects?

Any suggestions would be appreciated and please let me know where my understanding is not correct

Regards,
Deepika

shantanoo-desai, 2015-06-25:
Correct me if I wrong but WiFi works on 802.11 IEEE standards while as sensor nodes are standardized under 802.15.4 IEEE. Though they all share the same frequency band of 2.4 GHz I presume just getting data directly to a phone or PC isn't why the whole field of IoT or Wireless Sensor Networks was developed. 
A border router is like a translator which understands the IEEE 802.15.4 and the 6lbr will take care of the Ethernet and WSN which are different standards as Ethernet is 802.3 IEEE.
Though I think @laurentderu  may have a better understanding.

laurentderu, 2015-06-26:
@shantanoo-desai you're quite right, WSN also require routing and network management, hence the need of a Border Router (or Network Coordinator or whatever...) Only the most basic network could run without and in any case you would need a specific radio device and network stack in your phone or pc.

deepikay, 2015-06-26:
@shantanoo-desai  @laurentderu ....thanks for correcting me.

So, I want to know that in a case where I need simple data transfer/communication between constrained devices and my smartphone, I would need 2 sensor devices having IPv6 stack, one of them shall be a border router and other does the sensing.

Can this border router communicate over wi-fi with the phone using its link local IPv6 address, does it also need the NAT64 capability?

In case of Contiki OS we have RPL border router programs, by running it on z1 mote I am able to see its routing table in the browser, it is connected through USB to PC and a TUNSLIP6 script makes connection over serial port. How can see connect it through wi-fi ?

Thanks

laurentderu, 2015-06-29:
If you have a6LBR running on Linux, it can be use a WiFi interface instead of the Ethernet interface. See the FAQs, https://github.com/cetic/6lbr/wiki/FAQ#the-examples-use-ethernet-as-the-ipv6-interface-is-it-possible-to-use-wi-fi-instead

deepikay, 2015-06-29:
@laurentderu ....thanks for sharing the link, yes we can replace ethernet connected by wi-fi but 
I want to know can we make z1 mote running border router  communicate to raspberry pi having 6lbr over wi-fi instead of serial port?

shantanoo-desai, 2015-06-29:
@deepikay  I dont think I understand well, but you want to run a Z1 border router and send the data to a 6LBR raspberry pi? 
so according to my understanding first of all you need to have a device that can understand the 802.15.4 language on the Z1 border router that can translate it to 802.11 language and then this data gets sent to a raspberry pi which needs to understand the 802.11 language? 
I am not really sure that is such good logic or am I taking this in a wrong sense?

deepikay, 2015-07-01:
@shantanoo-desai .....sorry but I am bit confused, actually I want to test a scenario where a sensor is mounted on a moving vehicle where it sense something like temperature and its data get transferred to nearby smartphone. So that it can be further processed and uploaded to the server.

Does phone also support 802.15.4 (6LOWPAN)? if it supports then can't it directly talk to the sensor node and then act as  gateway so that I don't need raspberry pi in between.

laurentderu, 2015-07-01:
If phones support 802.15.4 and 6LoWPAN and CoAP and ... then your scenario would work, but afaik there is not phone supporting 802.15.4 nor 6LoWPAN, so you must have a BR somewhere (or select another protocol, like Bluetooth or BLE...)

closed 2016-03-01 by laurentderu