cetic/6lbr #89: 6lbr not start correctly

sigmahertz, 2015-06-26:
Hello everyone,

I am now in a starting phase of a 6lowpan project. 

I did exactly as the instruction from 
http://processors.wiki.ti.com/index.php/Cc26xx_sw_examples

The compile was done and I can see the CC2531 dongle information as well.

I have one problem when I start the 6lbr service by using the following 6lbr.conf
MODE=ROUTER
#MODE=SMART-BRIDGE
#MODE=RPL-RELAY
#MODE=FULL-TRANSPARENT-BRIDGE
#MODE=NDP-ROUTER
#MODE=6LR
#MODE=RPL-ROOT

RAW_ETH=1
BRIDGE=0
DEV_BRIDGE=br0
DEV_TAP=tap0
DEV_ETH=eth0
#RAW_ETH_FCS=1
RAW_ETH_FCS=0

LIB_6LBR=/usr/lib/6lbr
IFUP=/usr/lib/6lbr/6lbr-ifup
IFDOWN=/usr/lib/6lbr/6lbr-ifdown

D_RADIO=/dev/ttyACM0
BAUDRATE=115200

#Enable this line if radio device is not detected
#SCAN_DEVICE=/usr/lib/6lbr/bb-scan-usb.sh

LOG_LEVEL=3 #INFO and above only

When I start 6lbr by sudo service 6lbr start, the message is 
[ ok ] Starting 6lbr (via systemctl): 6lbr.service.
But I can not see nvm.dat in /etc/6lbr
The I check the 6lbr.log which is
Waiting for
Waiting for
Waiting for
Waiting for
Waiting for
Waiting for
...

and 6lbr.err is empty.

Then I run cetic_6lbr_router, the ourput is 
Contiki-6lbr-1.3.2-34-g3715b49 started with IPV6, RPL
Rime started with address 1.2.3.4.5.6.7.8
MAC CSMA RDC br-rdc NETWORK sicslowpan
Log level : 30
Log services : ffffffff
2015-05-26 14:25:02.675214: INFO: ETH: 6LBR watchdog started (interval: 60)
2015-05-26 14:25:02.678456: ERROR: ETH: Can not reset watchdog : Permission denied
2015-05-26 14:25:02.679997: INFO: 6LBR: Starting 6LBR version 1.3.3 (Contiki-6lbr-1.3.2-34-g3715b49)
2015-05-26 14:25:02.681210: INFO: NVM: Opening nvm file 'nvm.dat'
2015-05-26 14:25:02.682267: ERROR: NVM: Could not read nvm file
2015-05-26 14:25:02.683399: INFO: NVM: NVM Magic : ffff
2015-05-26 14:25:02.684293: INFO: NVM: NVM Version : ffff
2015-05-26 14:25:02.684947: ERROR: NVM: Invalid NVM magic number or unsupported NVM version, reseting it...
2015-05-26 14:25:02.685153: WARN: NVM: Migrate NVM version 0 towards 1
2015-05-26 14:25:02.685659: INFO: NVM: Opening nvm file 'nvm.dat'
2015-05-26 14:25:02.685909: ERROR: NVM: Could not write nvm file
2015-05-26 14:25:02.686548: INFO: SCMD: Started br-cmd process
2015-05-26 14:25:02.687043: INFO: NODECFG: No node_config.conf file specified
2015-05-26 14:25:02.687245: INFO: ETH: RAW/TAP init
2015-05-26 14:25:02.688887: FATAL: SLIP: can't open siodev : No such file or directory

I realize it can not reset the watchdog 
Anyone can point out the answer to this.

Thanks!
Yong

laurentderu, 2015-06-26:
It's DEV_RADIO, not D_RADIO :) A check should be added in the startup script.

sigmahertz, 2015-06-30:
Hello,

Yes a stupid mistake. 
I changed it but the problem is still there. 
in the 6lbr.log, it is 
Waiting for /dev/ttyACM0
Waiting for /dev/ttyACM0
......

in the 6lbr.err, it is only one line
Tue Jun 30 18:17:37 UTC 2015 : Starting 6LBR

After adding /usr/lib/6lbr/bin to the PATH, the nvm.dat is created with the following content after start
Reading nvm file 'nvm.dat'
Error reading nvm file: No such file or directory
vin@6lowpan:/var/log$ nvm_tool --print /etc/6lbr/nvm.dat
Reading nvm file '/etc/6lbr/nvm.dat'
Channel : 26

WSN network prefix : aaaa::
WSN network prefix length : 64
WSN IP address : aaaa::100
WSN accept RA : True
WSN IP address autoconf : True

Eth network prefix : bbbb::
Eth network prefix length : 64
Eth IP address : bbbb::100
Eth default router : ::
Eth IP address autoconf : False

Local address rewrite : True
Smart Multi BR : False

RA daemon : True
RA router lifetime : 0
RA maximum interval : 600
RA minimum interval : 200
RA minimum delay : 3
RA PIO enabled : True
RA prefix valid lifetime : 86400
RA prefix preferred lifetime : 14400
RA RIO enabled : True
RA RIO lifetime : 1800

RPL instance ID : 30
RPL Preference : 0
RPL version ID :  : 242
RPL DIO interval doubling : 8
RPL DIO minimum interval : 12
RPL DIO redundancy : 10
RPL default lifetime : 30
RPL minimum rank increment : 256
RPL lifetime unit : 256

Webserver configuration page disabled : False

When I run 6lbr, (just type 6lbr instead of sudo service 6lbr start)
it is still 
Tue Jun 30 18:23:43 UTC 2015 : Starting 6LBR
/usr/bin/../lib/6lbr/bin/cetic_6lbr_router -c /etc/6lbr/nvm.dat -s /dev/ttyACM0 -t eth0 -r -B 115200 -U /usr/bin/../lib/6lbr/6lbr-ifup -D /usr/bin/../lib/6lbr/6lbr-ifdown -w /usr/bin/../lib/6lbr/www -L 3  -W /var/log/6lbr.timestamp -P 60 -C /var/log/6lbr.ip  -m /usr/bin/../lib/6lbr/plugins
Contiki-6lbr-1.3.2-34-g3715b49 started with IPV6, RPL
Rime started with address 1.2.3.4.5.6.7.8
MAC CSMA RDC br-rdc NETWORK sicslowpan
Log level : 30
Log services : ffffffff
2015-05-30 18:23:43.843206: INFO: ETH: 6LBR watchdog started (interval: 60)
2015-05-30 18:23:43.847720: ERROR: ETH: Can not reset watchdog : Permission denied
2015-05-30 18:23:43.848859: INFO: 6LBR: Starting 6LBR version 1.3.3 (Contiki-6lbr-1.3.2-34-g3715b49)
2015-05-30 18:23:43.849348: INFO: NVM: Opening nvm file '/etc/6lbr/nvm.dat'
2015-05-30 18:23:43.850056: INFO: NVM: NVM Magic : 2009
2015-05-30 18:23:43.850741: INFO: NVM: NVM Version : 1
2015-05-30 18:23:43.852075: INFO: SCMD: Started br-cmd process
2015-05-30 18:23:43.852716: INFO: NODECFG: No node_config.conf file specified
2015-05-30 18:23:43.853161: INFO: ETH: RAW/TAP init
2015-05-30 18:23:43.853931: FATAL: SLIP: can't open siodev /dev/ttyACM0 : Permission denied
/usr/bin/6lbr: line 208:  3115 Terminated              application_watchdog
Tue Jun 30 18:23:43 UTC 2015 : /usr/bin/../lib/6lbr/bin/cetic_6lbr_router failed, exit code: 1

Any clue for this?
Thanks!
BR
Yong

laurentderu, 2015-06-30:
First of all, are you sure that your CC2531 dongle is detected by your linux host and the device /dev/ttypACM0 is the correct device ?
If you are using a BeagleBone Black and no device appears when you plug in your dongle, you have to enable the workaround to force detection :

<pre>
#Enable this line if radio device is not detected
SCAN_DEVICE=/usr/lib/6lbr/bb-scan-usb.sh
</pre>

(btw, You can not execute 6LBR as non-root, that's why your second test always fails)

sigmahertz, 2015-07-01:
Hello,

Thanks a lot!
It looks like the dongle is running,
when I start the service, the LED blinks several times and the once every
10 second. I got the 6lbr.log like this
********************************************************************************
Wed Jul 1 12:41:07 UTC 2015 : Starting 6LBR
/usr/lib/6lbr/bin/cetic_6lbr_router -c /etc/6lbr/nvm.dat -s /dev/ttyACM0 -t
eth0 -r -B 115200 -U /usr/lib/6lbr/6lbr-ifup -D /usr/lib/6lbr/6lbr-ifdown
-w /usr/lib/6lbr/www -L 3  -W /var/log/6lbr.timestamp -P 60 -C
/var/log/6lbr.ip  -m /u$
Contiki-6lbr-1.3.2-34-g3715b49 started with IPV6, RPL
Rime started with address 1.2.3.4.5.6.7.8
MAC CSMA RDC br-rdc NETWORK sicslowpan
Log level : 30
Log services : ffffffff
2015-06-01 12:41:07.298703: INFO: ETH: 6LBR watchdog started (interval: 60)
2015-06-01 12:41:07.299868: INFO: 6LBR: Starting 6LBR version 1.3.3
(Contiki-6lbr-1.3.2-34-g3715b49)
2015-06-01 12:41:07.299908: INFO: NVM: Opening nvm file '/etc/6lbr/nvm.dat'
2015-06-01 12:41:07.302413: INFO: NVM: NVM Magic : 2009
2015-06-01 12:41:07.302445: INFO: NVM: NVM Version : 1
2015-06-01 12:41:07.302461: INFO: SCMD: Started br-cmd process
2015-06-01 12:41:07.302474: INFO: NODECFG: No node_config.conf file
specified
2015-06-01 12:41:07.302489: INFO: ETH: RAW/TAP init
2015-06-01 12:41:07.304745: INFO: SLIP: SLIP started on /dev/ttyACM0
2015-06-01 12:41:07.379174: INFO: TAP: opened device /dev/eth0
2015-06-01 12:41:07.379525: INFO: TAP: Running 6lbr-ifup script
'/usr/lib/6lbr/6lbr-ifup'
eth0      Link encap:Ethernet  HWaddr 54:4a:16:bb:7e:5c
          inet addr:192.168.0.102  Bcast:192.168.0.255  Mask:255.255.255.0
          inet6 addr: fe80::564a:16ff:febb:7e5c/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:326 errors:0 dropped:0 overruns:0 frame:0
          TX packets:101 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:39076 (39.0 KB)  TX bytes:11916 (11.9 KB)
          Interrupt:56

2015-06-01 12:41:07.560562: INFO: TAP: Eth MAC address : 54:4a:16:bb:7e:5c
2015-06-01 12:41:07.560738: INFO: BR-RDC: Reset SLIP Radio
2015-06-01 12:41:07.560810: INFO: BR-RDC: Fetching MAC address
2015-06-01 12:41:07.593836: INFO: SLIP: cc2531: Reboot requested
2015-06-01 12:41:07.595204: INFO: SLIP: cc2531: Reverting to h/w MAC but
not rebooting
2015-06-01 12:41:07.624569: INFO: BR-RDC: Got MAC: 00:12:4b:00:02:f1:d8:04
2015-06-01 12:41:08.560644: INFO: 6LBR: Tentative local IPv6 address
fe80::212:4b00:2f1:d804
2015-06-01 12:41:08.561066: INFO: 6LBR: Tentative global IPv6 address (WSN)
aaaa::212:4b00:2f1:d804
2015-06-01 12:41:08.561481: INFO: 6LBR: Tentative global IPv6 address (ETH)
bbbb::100
2015-06-01 12:41:08.561652: INFO: 6LBR: RA Daemon enabled
2015-06-01 12:41:08.561844: INFO: NVM: Opening nvm file '/etc/6lbr/nvm.dat'
2015-06-01 12:41:08.563115: INFO: 6LBR: Configured as DODAG Root
2015-06-01 12:41:08.563337: INFO: 6LBR: Starting as RPL ROUTER
2015-06-01 12:41:08.564725: INFO: UDPS: UDP server started
2015-06-01 12:41:08.569453: INFO: 6LBR: CETIC 6LBR Started
2015-06-01 12:41:08.569736: INFO: SLIP: cc2531: Setting RF channel 0x19

I still have some questions about verifying the setup
now the beaglebone and my labtop are connected to a router. From nvm.dat,
the eht IP address is bbbb::100, i have to assign the pc within the same ip
segment. But when I configure the router, it says bbbb is not a ipv6 prefix.
now if I use ifconfig, the ip address on the beaglebone is
eth0      Link encap:Ethernet  HWaddr 54:4a:16:bb:7e:5c
          inet addr:192.168.0.102  Bcast:192.168.0.255  Mask:255.255.255.0
          inet6 addr: fe80::564a:16ff:febb:7e5c/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:1358 errors:0 dropped:0 overruns:0 frame:0
          TX packets:137 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:261746 (261.7 KB)  TX bytes:14870 (14.8 KB)
          Interrupt:56

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

usb0      Link encap:Ethernet  HWaddr 66:3d:72:ad:d2:d6
          inet addr:192.168.7.2  Bcast:192.168.7.3  Mask:255.255.255.252
          inet6 addr: fe80::643d:72ff:fead:d2d6/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:609 errors:0 dropped:0 overruns:0 frame:0
          TX packets:371 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:49650 (49.6 KB)  TX bytes:74103 (74.1 KB)

wlan0     Link encap:Ethernet  HWaddr 00:c1:41:29:0c:5b
          inet addr:172.17.9.119  Bcast:172.17.9.255  Mask:255.255.255.0
          inet6 addr: fe80::2c1:41ff:fe29:c5b/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:980 errors:0 dropped:0 overruns:0 frame:0
          TX packets:64 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:201146 (201.1 KB)  TX bytes:10766 (10.7 KB)

and the ip of the labtop is
Link-local IPv6 Address . . . . . : fe80::44ae:b1c9:f99b:e440%11

What can I do with this?

I also download the cc26xx-web-demo firmware to a TI CC2650 sensortag, how
can I verify the link between the dongle and the sensortag?

Thanks!

Best Regards！
-----------------------
Yong Wang
|+46 70 279 97 82|

closed 2016-03-01 by laurentderu