cetic/6lbr #90: Non-Compliant DODAG ID

g-oikonomou, 2015-06-26:
I just noticed that 6lbr advertises DODAG ID `fe80::<blah>`

In RFC6550 [6.3.1.](https://tools.ietf.org/html/rfc6550#section-6.3.1) (Format of the DIO Base Object)
```
DODAGID: 128-bit IPv6 address set by a DODAG root that uniquely
         identifies a DODAG.  The DODAGID MUST be a routable IPv6
         address belonging to the DODAG root.
```

I interpret "routable" as "not link-local". Thus, I would expect the DODAG ID to be somewhere in `<WSN network prefix>::/<WSN network prefix length>`?

laurentderu, 2015-06-29:
Indeed it's non compliant. In the Develop branch, this has been modified and you can select the DODAG ID (either the BR global address, the local address to be backward compatible or an arbitrary address).

The idea was to be able to support prefixes DODAG in order to speed up the routing tree building at startup or prefix change, but actually it's not really useful (and not compliant :))

laurentderu, 2015-06-29:
But I see the default configuration is still link-local address, I should fix that

laurentderu, 2016-03-31:
Integrated and validated in the upcoming 1.4.0 release

closed 2016-03-31 by laurentderu