cetic/6lbr #91: Merging the latest Contiki CC2538 files into 6LBR

troymccann, 2015-06-28:
I've been working on a project where I want to use CoAP over DTLS with Contiki so I've been developing on 6LBR instead of the vanilla Contiki release. I've just noticed that the CC2538 files are pretty out of date though, in particular to do with crypto, so I've been trying to move them into 6LBR.

I've directly copied the cpu/cc2538 and platform/cc2538dk folders across expecting that these should be pretty straight forward since everything seems to be pretty modular, but I'm having trouble compiling. 

I'm currently getting the error "undefined reference to crypto_isr" from within platform/cc2538/startup-gcc. Looking here [1] it does look like there's no explicitly included file for the function, so I presume there's some magic happening within the makefile structure somewhere that I'm not aware of, because there's no problems with these files when compiled within the vanilla contiki source. If anyone has any ideas, please share! Just when I thought I'd got a grasp on navigating the Contiki code, I hit a wall with this.

Any help will be greatly appreciated!

[1] https://github.com/contiki-os/contiki/search?utf8=%E2%9C%93&q=crypto_isr

troymccann, 2015-06-28:
I'm trying to instead move TinyDTLS to the latest Contiki build, and I've moved all the apps found in 6LBR as well, but now I'm getting a list of 'undefined reference' errors for TinyDTLS functions.

What are the differences between 6LBR and Contiki? I'm not sure what I'm missing.

laurentderu, 2015-06-29:
Hello,

You can use the develop branch and one of the develop snapshot, they are at most a few weeks behind Contiki and quite stable (and contains the latest fixes for CC2538). The delta between the last stable 6LBR (1.3.3) and latest Contiki is quite huge, so I would not go that way.

troymccann, 2015-06-30:
Oh, fantastic. Thanks for pointing that out!

closed 2015-08-05 by laurentderu