cetic/6lbr #92: unable to see TI sensortag 2.0 from webpage

weijian19391, 2015-07-01:
Hi 

I followed the example from http://processors.wiki.ti.com/index.php/Cc26xx_sw_examples, but instead of using BBB i used a Raspberry Pi as my edge router, configuration for 6lbr.config as follows: 
````

MODE=ROUTER
#MODE=SMART-BRIDGE
#MODE=RPL-RELAY
#MODE=FULL-TRANSPARENT-BRIDGE
#MODE=NDP-ROUTER
#MODE=6LR
#MODE=RPL-ROOT

RAW_ETH=1
BRIDGE=0
DEV_BRIDGE=br0
DEV_TAP=tap0
DEV_ETH=eth0
RAW_ETH_FCS=0

DEV_RADIO=/dev/ttyACM0
BAUDRATE=115200

LOG_LEVEL=1 #INFO and above only

````

I am able to see the web page showing me information about the boarder router, but under the sensor tab, the Sensortag are not being displayed. Checking on the log file : 
````
Wed Jul 1 09:39:31 UTC 2015 : Starting 6LBR
/usr/lib/6lbr/bin/cetic_6lbr_router -c /etc/6lbr/nvm.dat -s /dev/ttyACM0 -t eth0 -r -B 115200 -U /usr/lib/6lbr/6lbr-ifup -D /usr/lib/6lbr/6lbr-ifdown -w /usr/lib/6lbr/www -L 1  -W /var/log/6lbr.timestamp -P 60 -C /var/log/6lbr.ip  -m /usr/lib/6lbr/plugins  
Contiki-6lbr-1.3.3 started with IPV6, RPL
Rime started with address 1.2.3.4.5.6.7.8
MAC CSMA RDC br-rdc NETWORK sicslowpan
Log level : 10
Log services : ffffffff
eth0      Link encap:Ethernet  HWaddr b8:27:eb:47:62:57  
          inet addr:172.29.32.12  Bcast:172.29.35.255  Mask:255.255.252.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:2501624 errors:0 dropped:1598 overruns:0 frame:0
          TX packets:36261 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:350150429 (333.9 MiB)  TX bytes:5755660 (5.4 MiB)

2015-06-01 9:39:52.706820: ERROR: BR-RDC: br-rdc: send failed, slip ack timeout (6)
2015-06-01 9:39:55.4644: ERROR: BR-RDC: br-rdc: send failed, slip ack timeout (7)
2015-06-01 9:39:57.413032: ERROR: BR-RDC: br-rdc: send failed, slip ack timeout (8)
2015-06-01 9:39:59.670708: ERROR: BR-RDC: br-rdc: send failed, slip ack timeout (9)
2015-06-01 9:40:02.28807: ERROR: BR-RDC: br-rdc: send failed, slip ack timeout (10)
2015-06-01 9:40:04.154226: ERROR: BR-RDC: br-rdc: send failed, slip ack timeout (11)
2015-06-01 9:40:06.409359: ERROR: BR-RDC: br-rdc: send failed, slip ack timeout (12)
2015-06-01 9:40:08.747607: ERROR: BR-RDC: br-rdc: send failed, slip ack timeout (13)
2015-06-01 9:40:11.830729: ERROR: BR-RDC: br-rdc: send failed, slip ack timeout (14)
2015-06-01 9:40:14.749025: ERROR: BR-RDC: br-rdc: send failed, slip ack timeout (15)
2015-06-01 9:40:16.875006: ERROR: BR-RDC: br-rdc: send failed, slip ack timeout (0)
2015-06-01 9:40:19.215748: ERROR: BR-RDC: br-rdc: send failed, slip ack timeout (1)
2015-06-01 9:40:21.541121: ERROR: BR-RDC: br-rdc: send failed, slip ack timeout (2)
2015-06-01 9:40:24.43239: ERROR: BR-RDC: br-rdc: send failed, slip ack timeout (3)
2015-06-01 9:40:26.526165: ERROR: BR-RDC: br-rdc: send failed, slip ack timeout (4)
2015-06-01 9:40:28.651151: ERROR: BR-RDC: br-rdc: send failed, slip ack timeout (5)
2015-06-01 9:40:31.19045: ERROR: BR-RDC: br-rdc: send failed, slip ack timeout (6)
2015-06-01 9:40:33.368616: ERROR: BR-RDC: br-rdc: send failed, slip ack timeout (7)
2015-06-01 9:40:36.160376: ERROR: BR-RDC: br-rdc: send failed, slip ack timeout (8)
2015-06-01 9:40:38.742339: ERROR: BR-RDC: br-rdc: send failed, slip ack timeout (9)
2015-06-01 9:40:40.867357: ERROR: BR-RDC: br-rdc: send failed, slip ack timeout (10)
2015-06-01 9:40:43.140344: ERROR: BR-RDC: br-rdc: send failed, slip ack timeout (11)
2015-06-01 9:40:45.634505: ERROR: BR-RDC: br-rdc: send failed, slip ack timeout (12)
2015-06-01 9:40:48.83386: ERROR: BR-RDC: br-rdc: send failed, slip ack timeout (13)
2015-06-01 9:40:50.960031: ERROR: BR-RDC: br-rdc: send failed, slip ack timeout (14)
2015-06-01 9:41:35.813553: ERROR: BR-RDC: br-rdc: send failed, slip ack timeout (15)
2015-06-01 9:41:37.962901: ERROR: BR-RDC: br-rdc: send failed, slip ack timeout (0)
2015-06-01 9:41:40.474580: ERROR: BR-RDC: br-rdc: send failed, slip ack timeout (1)
2015-06-01 9:41:42.819200: ERROR: BR-RDC: br-rdc: send failed, slip ack timeout (2)
2015-06-01 9:41:45.500826: ERROR: BR-RDC: br-rdc: send failed, slip ack timeout (3)
2015-06-01 9:43:49.989961: ERROR: BR-RDC: br-rdc: send failed, slip ack timeout (4)
2015-06-01 9:43:52.158674: ERROR: BR-RDC: br-rdc: send failed, slip ack timeout (5)
2015-06-01 9:43:54.647044: ERROR: BR-RDC: br-rdc: send failed, slip ack timeout (6)
2015-06-01 9:43:57.400956: ERROR: BR-RDC: br-rdc: send failed, slip ack timeout (7)
2015-06-01 9:43:59.826454: ERROR: BR-RDC: br-rdc: send failed, slip ack timeout (8)
2015-06-01 9:44:23.812086: ERROR: BR-RDC: br-rdc: send failed, slip ack timeout (9)
2015-06-01 9:44:25.991675: ERROR: BR-RDC: br-rdc: send failed, slip ack timeout (10)
2015-06-01 9:44:28.223968: ERROR: BR-RDC: br-rdc: send failed, slip ack timeout (11)
2015-06-01 9:44:30.719097: ERROR: BR-RDC: br-rdc: send failed, slip ack timeout (12)
2015-06-01 9:44:33.597958: ERROR: BR-RDC: br-rdc: send failed, slip ack timeout (13)
````

I have flashed the Sensortag with the web-demo.bin (http://processors.wiki.ti.com/index.php/Contiki_setting_up_sw) using over the air downloader from my android phone, and CC2531 as my slip radio, flashed with  ````cc2531-slip-radio_contikimac.zip```` from   http://processors.wiki.ti.com/index.php/Cc26xx_sw_examples. 

May i know where i might have done wrong? Thank you

laurentderu, 2015-07-01:
According to the log, it seems 6LBR can not communicate properly with the slip-radio (hence all the errors). Please check if it is correctly programmed. If so, maybe you have issues with the cc2531 and the raspberry, see the raspberry page on the Wiki for some workarounds (https://github.com/cetic/6lbr/wiki/RaspberryPi-Software-Configuration)

Laurent.

weijian19391, 2015-07-02:
@laurentderu Thanks for your reply. Any idea where and how can i debug where went wrong? i manage to get to the web page served by 6lbr, hence i supposed the software is being set up correctly on my RPI. How can i check if the slip radio is working and that it is working with RPI?

weijian19391, 2015-07-02:
@laurentderu I got it to work! Turns out, i need the follow settings to be done for RPi even though i am not using Nooliberry radio
````
remove the following parameters from /boot/cmdline.txt
console=ttyAMA0,115200 kgdboc=ttyAMA0,115200
note that you do not need dwc_otg.speed=1 in that file for the Nooliberry, due to the lower baudrate
Also, in /etc/inittab the following line at the bottom of the file should be commented out, because the Nooliberry needs to use the ttyAMA0:
#Spawn a getty on Raspberry Pi serial line

#T0:23:respawn:/sbin/getty -L ttyAMA0 115200 vt100
````

Thank you very much for your help!

sdawans, 2015-07-02:
hello @weijian19391, on your RPi, are you using one of the official raspbian images? If so, do you know which release it is? This problem appeared on certain versions of Raspbian, it would help us to know which one you are using.

weijian19391, 2015-07-02:
@sdawans Yes i am running the official Raspbian image. Not sure if this is what you're looking for, but using ```` uname -a```` gave me the following: 
````
Linux raspberrypi 4.0.7+ #801 PREEMPT Tue Jun 30 18:15:24 BST 2015 armv6l GNU/Linux
````

On a side note, do you happen to know any NAT64 program that i can run in my RPi for it to translate between IPv6 and IPv4?

laurentderu, 2015-07-02:
For NAT64, you can either use the latest develop snapshot which integrate NAT64 or use Jool on your RPi.

weijian19391, 2015-07-03:
@laurentderu is there any elegant way of updating my current 6lbr to the develop branch, or do i have to remove the current 6lbr and git clone the develop snapshot to reinstall 6lbr again?

laurentderu, 2015-07-03:
if you are using git and the github repository of 6lbr, you can do a git fetch to have the lastest version and then do a git checkout develop-20150508 to select the latest snapshot

weijian19391, 2015-07-06:
@laurentderu  I see. Thanks! 

Is there any way for me to inspect the traffic flow of the 6lbr? I have implemented a mqtt client on my TI sensor tag, which is supposed to send data to port 1883, however when i tried to listen on port 1883 nothing is being received. Hence, i am now trying to see which part of the connection went wrong.

msnyder415, 2015-07-17:
I have run into the same problem, tho the 2 tweaks for RPI did not really improve things. 

I am using raspberry pi2, cc2531 with cc2531-slip-radio_contikimac.zip from http://processors.wiki.ti.com/index.php/Cc26xx_sw_examples.

raspian version : 
Linux raspberrypi 3.18.7-v7+ #755 SMP PREEMPT Thu Feb 12 17:20:48 GMT 2015 armv7l GNU/Linux

Here are my rpi tweaks :
/boot/cmdline.txt
dwc_otg.lpm_enable=0 console=tty1 root=/dev/mmcblk0p6 rootfstype=ext4 elevator=deadline rootwait

/etc/inittab
#Spawn a getty on Raspberry Pi serial line
#T0:23:respawn:/sbin/getty -L ttyAMA0 115200 vt100

I am using ttyACM0, so I am not quite sure this step is necessary

Here is my cfg :
# cat /etc/6lbr/6lbr.conf
MODE=ROUTER
#MODE=SMART-BRIDGE
#MODE=RPL-RELAY
#MODE=FULL-TRANSPARENT-BRIDGE
#MODE=NDP-ROUTER
#MODE=6LR
#MODE=RPL-ROOT

RAW_ETH=1
BRIDGE=0
DEV_BRIDGE=br0
DEV_TAP=tap0
DEV_ETH=wlan0
#RAW_ETH_FCS=1
RAW_ETH_FCS=0

DEV_RADIO=/dev/ttyACM0
BAUDRATE=115200

LOG_LEVEL=3
 

Here is my log : 
Waiting for /dev/ttyACM0
Waiting for /dev/ttyACM0
Waiting for /dev/ttyACM0
Waiting for /dev/ttyACM0
********************************************************************************
Fri Jul 17 22:48:12 UTC 2015 : Starting 6LBR
/usr/lib/6lbr/bin/cetic_6lbr_router -c /etc/6lbr/nvm.dat -s /dev/ttyACM0 -t wlan0 -r -B 115200 -U /usr/lib/6lbr/6lbr-ifup -D /usr/lib/6lbr/6lbr-ifdown -w /usr/lib/6lbr/www -L 3  -W /var/log/6lbr.timestamp -P 60 -C /var/log/6lbr.ip  -m /usr/lib/6lbr/plugins  
Contiki-6lbr-1.3.2-34-g3715b49 started with IPV6, RPL
Rime started with address 1.2.3.4.5.6.7.8
MAC CSMA RDC br-rdc NETWORK sicslowpan
Log level : 30
Log services : ffffffff
2015-06-17 22:48:12.513215: INFO: ETH: 6LBR watchdog started (interval: 60)
2015-06-17 22:48:12.514149: INFO: 6LBR: Starting 6LBR version 1.3.3 (Contiki-6lbr-1.3.2-34-g3715b49)
2015-06-17 22:48:12.514177: INFO: NVM: Opening nvm file '/etc/6lbr/nvm.dat'
2015-06-17 22:48:12.514266: INFO: NVM: NVM Magic : 2009
2015-06-17 22:48:12.514287: INFO: NVM: NVM Version : 1
2015-06-17 22:48:12.514303: INFO: SCMD: Started br-cmd process
2015-06-17 22:48:12.514318: INFO: NODECFG: No node_config.conf file specified
2015-06-17 22:48:12.514333: INFO: ETH: RAW/TAP init
2015-06-17 22:48:12.519091: INFO: SLIP: SLIP started on /dev/ttyACM0
2015-06-17 22:48:12.550802: INFO: TAP: opened device /dev/wlan0
2015-06-17 22:48:12.551161: INFO: TAP: Running 6lbr-ifup script '/usr/lib/6lbr/6lbr-ifup'
wlan0     Link encap:Ethernet  HWaddr 00:0f:13:40:28:33  
          inet addr:192.168.1.26  Bcast:192.168.1.255  Mask:255.255.255.0
          inet6 addr: fe80::20f:13ff:fe40:2833/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:9370 errors:0 dropped:0 overruns:0 frame:0
          TX packets:1492 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:1033283 (1009.0 KiB)  TX bytes:407827 (398.2 KiB)

2015-06-17 22:48:12.732721: INFO: TAP: Eth MAC address : 00:0f:13:40:28:33
2015-06-17 22:48:12.732874: INFO: BR-RDC: Reset SLIP Radio
2015-06-17 22:48:12.732941: INFO: BR-RDC: Fetching MAC address
2015-06-17 22:48:12.766697: INFO: SLIP: cc2531: Reboot requested
2015-06-17 22:48:12.768036: INFO: SLIP: cc2531: Reverting to h/w MAC but not rebooting
2015-06-17 22:48:12.796906: INFO: BR-RDC: Got MAC: 00:12:4b:00:02:f1:d6:09
2015-06-17 22:48:13.732482: INFO: 6LBR: Tentative local IPv6 address fe80::212:4b00:2f1:d609
2015-06-17 22:48:13.732902: INFO: 6LBR: Tentative global IPv6 address (WSN) aaaa::212:4b00:2f1:d609
2015-06-17 22:48:13.733275: INFO: 6LBR: Tentative global IPv6 address (ETH) bbbb::100
2015-06-17 22:48:13.733581: INFO: 6LBR: RA Daemon enabled
2015-06-17 22:48:13.733725: INFO: NVM: Opening nvm file '/etc/6lbr/nvm.dat'
2015-06-17 22:48:13.734586: INFO: 6LBR: Configured as DODAG Root
2015-06-17 22:48:13.734843: INFO: 6LBR: Starting as RPL ROUTER
2015-06-17 22:48:13.736290: INFO: UDPS: UDP server started
2015-06-17 22:48:13.736685: INFO: 6LBR: CETIC 6LBR Started
2015-06-17 22:48:13.736890: INFO: SLIP: cc2531: Setting RF channel 0x19
2015-06-17 22:48:16.232817: ERROR: BR-RDC: br-rdc: send failed, slip ack timeout (1)
2015-06-17 22:48:18.539070: ERROR: BR-RDC: br-rdc: send failed, slip ack timeout (2)
2015-06-17 22:48:21.155954: ERROR: BR-RDC: br-rdc: send failed, slip ack timeout (3)
2015-06-17 22:48:23.953336: ERROR: BR-RDC: br-rdc: send failed, slip ack timeout (4)

opendsvn, 2015-07-29:
We need to reduce the speed of the CC2531 USB Dongle: 
I added this parameter  "dwc_otg.speed=1" to the beginning of the /boot/cmdline.txt like this:

dwc_otg.speed=1  dwc_otg.lpm_enable=0 console=tty1 root=/dev/mmcblk0p3 rootwait

Then it works!

laurentderu, 2015-07-29:
@opendsvn thanks for the info, I will update the wiki to make more clear that one need this parameter for CC2531, Telos, ... slip-radios.

zubairhamed, 2015-07-29:
Is this property only for slip radios or also for sniffers?

Z

laurentderu, 2015-07-29:
It's for both, you need that parameter to have the USB device working on RPi.

rohitsivakumar, 2016-05-12:
I have a similar problem and my hardware runs raspbian (supported on BananaPI hardware). my cmdline.txt settings are: 
`dwc_otg.speed=1 dwc_otg.lpm_enable=0 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline rootwait`

I am still facing the same problem and the same error:
`ERROR: BR-RDC: br-rdc: send failed, slip ack timeout`

schankam, 2016-06-02:
Same problem here, was running with Raspbian and the soft was unstable, then switched to Ubuntu 16 LTS Mate, and I now got:

`ERROR: BR-RDC: br-rdc: send failed, slip ack timeout`

laurentderu, 2016-06-02:
What is the frequency of the error, do you have it for each packet sent or do you have one once in a while ?
Could you check if in /etc/inittab you don't have a console opened on your serial device ?

naivewong, 2017-06-02:
My raspberry pi's version is 4.4.50-v7+. I have the same problem. But I don't have the /etc/inittab file and my /boot/cmdline.txt is dwc_otg.lpm_enable=0 console=serial0, 115200 console=tty1 root=/dev/mmcblk0p7 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait splash plymouth.ignore-serial-consoles dwc_otg.speed=1
Can anyone tell me how to configure? Thank you very much!!!

closed 2017-06-15 by laurentderu

Usman78930, 2019-05-18:
I am also facing the same issue that i am able to access my web page that showing 6lbr information but under the sensor tab sensor values are not showing.  i am using raspbian stretch. Kindly help me fix this problem