cetic/6lbr #94: CC2538 + enc28j60: requirements for working

nolex90, 2015-07-02:
Hello,

I flashed 6lbr mainlatest to CC2538 with CETIC_6LBR_ROUTER. So now I can't reach the webpage.
I flashed with debugger, is it necessary that the NVM must contain legal information?
What is the network requirement to get work?
There is no any packet from this mac address.

Packets:
TI SmartRF06 + cc2538EM
 Net: sicslowpan
 MAC: CSMA          
 RDC: nullrdc         
 Channel: 25
 PAN-ID: abcd
Rime configured with address 00:12:4b:00:03:d0:a6:b7
PACKET: PF: bridge_output: Sending packet to Broadcast
PACKET: PF: eth_output: broadcast
PACKET: PF: eth_output: Sending packet to Ethernet
PACKET: ETH: write: 78            
PACKET: PF: bridge_output: Sending packet to Broadcast
PACKET: PF: wireless_output: sending packet
PACKET: PF: eth_output: broadcast          
PACKET: PF: eth_output: Sending packet to Ethernet
PACKET: ETH: write: 78                     
PACKET: PF: bridge_output: Sending packet to Broadcast
PACKET: PF: wireless_output: sending packet
PACKET: PF: bridge_output: Sending packet to Broadcast
PACKET: PF: wireless_output: sending packet
PACKET: PF: eth_output: broadcast  
PACKET: PF: eth_output: Sending packet to Ethernet
PACKET: ETH: write: 78              
PACKET: PF: bridge_output: Sending packet to Broadcast
PACKET: PF: wireless_output: sending packet
PACKET: PF: eth_output: broadcast
PACKET: PF: eth_output: Sending packet to Ethernet
PACKET: ETH: write: 78                                
PACKET: ETH: read: 14
PACKET: PF: eth_input: Dropping packet type=0x0800
PACKET: ETH: read: 14
PACKET: PF: eth_input: Dropping packet type=0x0800
PACKET: ETH: read: 14        
PACKET: PF: eth_input: Dropping packet type=0x0800
PACKET: ETH: read: 14     
PACKET: PF: eth_input: Dropping packet type=0x0800
PACKET: ETH: read: 14
  
Info + debug:
Rime configured with address 00:12:4b:00:03:d0:a6:b7
INFO: 6LBR: Starting 6LBR version 1.4.x (Contiki-develop-20150508-246-g49997f9)
INFO: NVM: Reading 6LBR NVM
INFO: NVM: NVM Magic : ffff
INFO: NVM: NVM Version : ffff
ERROR: NVM: Invalid NVM magic number or unsupported NVM version, reseting it...
WARN: NVM: Migrate NVM version 0 towards 1
WARN: NVM: Migrate NVM version 1 towards 2
INFO: NVM: Flashing 6LBR NVM
ERROR: NVM: verify NVM failed, retry
INFO: NVM: Flashing 6LBR NVM
INFO: ETH: ENC28J60 init
INFO: ETH: Eth MAC address : 06:00:03:d0:a6:b7
INFO: ENC: resetting chip
DEBUG: ENC: REV 6
INFO: ETH: ENC-28J60 Process started
INFO: LLSEC: Using 'nullsec' llsec driver
INFO: 6LBR: Security layer initialized
INFO: 6LBR: Tentative local IPv6 address fe80::212:4b00:3d0:a6b7
INFO: 6LBR: Tentative global IPv6 address (WSN) aaaa::212:4b00:3d0:a6b7
INFO: 6LBR: Tentative global IPv6 address (ETH) bbbb::100
DEBUG: RIO: Adding route information: length=64, flags=0, route lifetime=1800, d
est=aaaa::
INFO: 6LBR: RA Daemon enabled
INFO: 6LBR: Checking addresses duplication
INFO: NVM: Flashing 6LBR NVM
ERROR: NVM: verify NVM failed, retry
INFO: NVM: Flashing 6LBR NVM
INFO: 6LBR: Configured as DODAG Root fe80::212:4b00:3d0:a6b7
INFO: 6LBR: Starting as RPL ROUTER
DEBUG: HTTP: Adding page : index.html
DEBUG: HTTP: Adding page : sensors.html
DEBUG: HTTP: Adding page : sensor
DEBUG: HTTP: Adding page : sensors_tree.html
DEBUG: HTTP: Adding page : sensors_prr.html
DEBUG: HTTP: Adding page : sensors_ps.html
DEBUG: HTTP: Adding page : sensors_hc.html
DEBUG: HTTP: Adding command : reset-stats-all
DEBUG: HTTP: Adding command : reset-stats
DEBUG: HTTP: Adding command : rm-node
DEBUG: HTTP: Adding page : rpl.html
DEBUG: HTTP: Adding page : network.html
DEBUG: HTTP: Adding page : config.html
DEBUG: HTTP: Adding page : statistics.html
DEBUG: HTTP: Adding command : rpl-gr
DEBUG: HTTP: Adding command : rpl-reset
DEBUG: HTTP: Adding command : rpl-child
DEBUG: HTTP: Adding command : route-rm
DEBUG: HTTP: Adding command : nbr-rm     
DEBUG: HTTP: Adding command : config  
DEBUG: HTTP: Adding command : reset-config
DEBUG: HTTP: Adding page : admin.html          
DEBUG: HTTP: Adding command : restart                       
INFO: UDPS: UDP server started                                 
INFO: 6LBR: CETIC 6LBR Started

laurentderu, 2015-07-02:
Hello,

I see nothing wrong in your logs, your enc is receiving and at least try to send packets. Are you able to ping6 bbbb::100 ? If so, could you send me a pcap when you try to access the webserver ?

Laurent.

nolex90, 2015-07-04:
Here is the PCAP file:
https://drive.google.com/folderview?id=0B0T6b2efIoQ5YmlQRXo1d2NjTEU&usp=drive_web
[test.pcapng]
Test method:
Disable all connections and build a point-point connection to the 6lbr via ethernet.
Set bbbb::400 to the host address.
Start capturing.
Power up the system.

Any idea? I think the enc chip is bad...

nolex90, 2015-07-04:
Here is the PCAP file:
https://drive.google.com/folderview?id=0B0T6b2efIoQ5YmlQRXo1d2NjTEU&usp=drive_web
test.pcapng
Test method:
Disable all connections and build a point-point connection to the 6lbr via ethernet.
Set bbbb::400 to the host address.
Start capturing.
Power up the system.

Any idea? I think the enc chip is bad...

laurentderu, 2015-07-29:
I'm sorry this one slipped under the radar. Do you still have an issue or was the enc chip bad ?

closed 2016-03-01 by laurentderu