cetic/6lbr #95: Is Contiki rebooting?

troymccann, 2015-07-17:
I've been trying to get another process to run within the 6lbr-demo example, and that part seems to be working. The process is supposed to run forever, handing over to other processes when necessary but hopefully for no longer than around 50ms, so I'm using a PROCESS_PAUSE() call.

Unfortunately every so often the process stops running for around a second while the 3 LED sequence runs (CC2538 + SmartRF board) and the configuration parameters are again printed to the terminal, lines which are only at the beginning of some processes and outside while(1) loops, like "CoAP server started".

Would I be correct in saying that Contiki is in fact rebooting when this happens? It's the same with the plain 6LBR-demo example as well. What would be causing it to do this? When I comment out the CoAP server process the LED sequence only occurs the one time, but I can't find anything within coap-server.c which is causing it.

laurentderu, 2015-07-28:
It's hard to tell without more info or traces, but either the MCU crashes or the watchdog is triggered because a process is taking too much time.

Standard 6lbr-demo does not seems to crash on CC2548 based board (although not exactly identical to smartrf), but I don't see what could trigger a crash.

troymccann, 2015-08-05:
Thanks Laurent. I have a feeling it was something to do with my oscilloscope attached to one of the LED outputs, because without the oscilloscope attached it seems to run indefinitely as intended. 

Thanks for your help.

troymccann, 2015-08-05:
Scratch that. This is an example of what I get from the console output with the 6lbr-demo; although sometimes there's just no output at all. At the moment all I'm trying to do is access the web server using the CC2538DK+ENC28J60 combination...

```
Contiki 3.x
TI SmartRF06 + cc2538EM
 Net: sicslowpan
 MAC: CSMA
 RDC: nullrdc
 Channel: 26
 PAN-ID: abcd
Rime configured with address 00:12:4b:00:04:13:36:fc
Reading 6LBR NVM
NVM Magic : 2009
NVM Version : 0
NVM Size : 296
COAP-DTLS listening on port 5683
CoAP server started
Contiki 3.x
TI SmartRF06 + cc2538EM
 Net: sicslowpan
 MAC: CSMA
 RDC: nullrdc
 Channel: 26
 PAN-ID: abcd
Rime configured with address 00:12:4b:00:04:13:36:fc
Reading 6LBR NVM
NVM Magic : 2009
NVM Version : 0
NVM Size : 296
COAP-DTLS listening on port 5683
CoAP server started
Contiki 3.x
TI SmartRF06 + cc2538EM
 Net: sicslowpan
 MAC: CSMA
 RDC: nullrdc
 Channel: 26
 PAN-ID: abcd
Rime configured with address 00:12:4b:00:04:13:36:fc
Reading 6LBR NVM
NVM Magic : 2009
NVM Version : 0
NVM Size : 296
COAP-DTLS listening on port 5683
CoAP server started
Contiki 3.x
TI SmartRF06 + cc2538EM
 Net: sicslowpan
 MAC: CSMA
 RDC: nullrdc
 Channel: 26
 PAN-ID: abcd
Rime configured with address 00:12:4b:00:04:13:36:fc
Reading 6LBR NVM
NVM Magic : 2009
NVM Version : 0
NVM Size : 296
COAP-DTLS listening on port 5683
CoAP server started
Contiki 3.x
TI SmartRF06 + cc2538EM
 Net: sicslowpan
 MAC: CSMA
 RDC: nullrdc
 Channel: 26
 PAN-ID: abcd
Rime configured with address 00:12:4b:00:04:13:36:fc
Reading 6LBR NVM
NVM Magic : 2009
NVM Version : 0
NVM Size : 296
COAP-DTLS listening on port 5683
CoAP server started
Contiki 3.x
TI SmartRF06 + cc2538EM
 Net: sicslowpan
 MAC: CSMA
 RDC: nullrdc
 Channel: 26
 PAN-ID: abcd
Rime configured with address 00:12:4b:00:04:13:36:fc
Reading 6LBR NVM
NVM Magic : 2009
NVM Version : 0
NVM Size : 296
COAP-DTLS listening on port 5683
CoAP server started
```

laurentderu, 2015-08-05:
Hi,

It could have two root cause, either you have flashing problem, Uniflash does not handle well gcc's .elf files, or you must disable random_init() in apps/tinydtls/prng.h which could mess with the cc2538dk radio (random_init() is already called by the platform code).

troymccann, 2015-08-05:
I'm using Flash Programmer 2 on a Windows machine for flashing, so that should be ok I think. I removed the definition of prng_init(value) on line 65, and so far it seems to have stopped the rebooting. 

Thanks for your help Laurent!

laurentderu, 2015-08-05:
Thanks for the feedback, this problem is fixed in the latest version of Tinydtls, I'll push to include it in 6lbr soon.

troymccann, 2015-08-11:
Hi Laurent.

As discovered in my other thread (https://github.com/cetic/6lbr/issues/102) it looks like this might still be happening.

Here's a snippet of console output that shows 6LBR rebooting, without an explanation as to why:
```
INFO: NVM: Flashing 6LBR NVM
ERROR: NVM: verify NVM failed, retry
INFO: NVM: Flashing 6LBR NVM
INFO: 6LBR: Configured as DODAG Root fe80::212:4b00:413:36fc
INFO: 6LBR: Starting as RPL ROUTER
DEBUG: HTTP: Adding page : index.html
DEBUG: HTTP: Adding page : sensors.html
DEBUG: HTTP: Adding page : sensor
DEBUG: HTTP: Adding page : sensors_tree.html
DEBUG: HTTP: Adding page : sensors_prr.html
DEBUG: HTTP: Adding page : sensors_ps.html
DEBUG: HTTP: Adding page : sensors_hc.html
DEBUG: HTTP: Adding command : reset-stats-all
DEBUG: HTTP: Adding command : reset-stats
DEBUG: HTTP: Adding command : rm-node
DEBUG: HTTP: Adding page : rpl.html
DEBUG: HTTP: Adding page : network.html
DEBUG: HTTP: Adding page : config.html
DEBUG: HTTP: Adding page : statistics.html
DEBUG: HTTP: Adding command : rpl-gr
DEBUG: HTTP: Adding command : rpl-reset
DEBUG: HTTP: Adding command : rpl-child
DEBUG: HTTP: Adding command : route-rm
DEBUG: HTTP: Adding command : nbr-rm
DEBUG: HTTP: Adding command : config
DEBUG: HTTP: Adding command : reset-config
DEBUG: HTTP: Adding page : admin.html
DEBUG: HTTP: Adding command : restart
INFO: UDPS: UDP server started
INFO: 6LBR: CETIC 6LBR Started
PACKET: ENC: EPKTCNT 0x01
PACKET: ENC: rx: 60: 01:80:c2:00:00:00
PACKET: ENC: received_packets 31
PACKET: ETH: read: 14
DUMP: ETH: 
         0180c200 0000e8de 279aa2b4 0026
PACKET: PF: eth_input: Dropping packet type=0x0026
PACKET: ENC: EPKTCNT 0x01
PACKET: ENC: rx: 60: 01:80:c2:00:00:00
PACKET: ENC: received_packets 32
PACKET: ETH: read: 14
DUMP: ETH: 
         0180c200 0000e8de 279aa2b4 0026
PACKET: PF: eth_input: Dropping packet type=0x0026
PACKET: ENC: EPKTCNT 0x01
PACKET: ENC: rx: 60: ff:ff:ff:ff:ff:ff
PACKET: ENC: received_packets 33
PACKET: ETH: read: 14
DUMP: ETH: 
         0180c200 0000e8de 279aa2b4 0026
PACKET: PF: eth_input: Dropping packet type=0x0806
PACKET: PF: bridge_output: Sending packet to Broadcast
PACKET: PF: wireless_output: sending packet
PACKET: ENC: EPKTCNT 0x01
PACKET: ENC: rx: 60: ff:ff:ff:ff:ff:ff
PACKET: ENC: received_packets 34
PACKET: ETH: read: 14
DUMP: ETH: 
         ffffffff ffff6003 470aa5cd 0806
PACKET: PF: eth_input: Dropping packet type=0x0806
PACKET: ENC: EPKTCNT 0x01
PACKET: ENC: rx: 60: 01:80:c2:00:00:00
PACKET: ENC: received_packets 35
PACKET: ETH: read: 14
[... 100 lines cut ...]
PACKET: PF: eth_input: Dropping packet type=0x0800
PACKET: PF: bridge_output: Sending packet to Broadcast
PACKET: PF: wireless_output: sending packet
PACKET: PF: eth_output: broadcast
TRACE: 6LE: translateIP: ICMP Message detected
PACKET: PF: eth_output: Sending packet to Ethernet
PACKET: ETH: write: 78
DUMP: ETH: 
         3333ff13 36fc0600 041336fc 86dd6000 
         00000018 3aff0000 00000000 00000000 
         00000000 0000ff02 00000000 00000000 
         0001ff13 36fc8700 10cd0000 0000aaaa 
         00000000 00000212 4b000413 36fc
PACKET: ENC: tx: 78: 33:33:ff:13:36:fc
PACKET: ENC: sent_packets 4
PACKET: ENC: EPKTCNT 0x01
PACKET: ENC: rx: 60: 01:80:c2:00:00:00
PACKET: ENC: received_packets 3
PACKET: ETH: read: 14
DUMP: ETH: 
         3333ff13 36fc0600 041336fc 86dd
PACKET: PF: eth_input: Dropping packet type=0x0026
```

I'm now thinking that the rebooting might be why the 6LBR isn't getting an IPv6 address from the Billion router.

laurentderu, 2015-08-17:
Sorry, it has been a busy week :)

I can't help you much with the root cause of the reboot, all I can see is that the received packets are malformed, an Ethernet packet 14 bytes long is impossible as it's only the Ethernet header. So you have probably a problem there with the capture of the packet.

Btw, you are running in RPL Router mode, in this mode the Border Router does not try to get an address from a Router, instead it must be manually configured.

troymccann, 2015-08-19:
Not a problem, I know what that's like. :)

I think you might be onto something with the 14 byte length packets. I can see that the network is sending out much longer packets to the 6LBR, but it's only ever reading the first 14 bytes. Do you have any tips on where I might like to look to find out why? Hopefully resolving that will resolve the rebooting too.

For clarification, in router mode the 6LBR will attempt to check the network for any other devices with the IP bbbb::100 before setting it's eth interface with that address?  At what stage does it accept that no other device is using that address and will stop broadcasting NSs?

troymccann, 2015-08-19:
I modified enc28j60-drv.c to log the values for uip_len and UIP_LLH_LEN, and it's showing that uip_len is always 0 and UIP_LLH_LEN is 14 for incoming packets.

I'm still working out how all the functions are working together and what modifies uip_len, but I'm presuming it should not be zero.

troymccann, 2015-08-19:
I'm not sure what's changed, but I left the sniffer going for a while and came back to find that there were lots of RPL Control (DODAG Information Solicitation) packets, which I wasn't getting before. These are coming from the link-local address obviously. But I'm also noticing that there's no longer any mention of the global addresses at boot.

I've found that if I ping6 the tentative link-local address then the 6LBR reboots immediately.

laurentderu, 2015-08-19:
I found some time to investigate further, its seems to be an alignment problem in uip-nd6.c the lines 

<pre>
uip_ipaddr_copy(&UIP_IP_BUF->srcipaddr, &UIP_ND6_NS_BUF->tgtipaddr);
</pre>

are making the CC2538 reboot. If I replace uip_ipaddr_copy() by memcpy() everything is fine and I'm able to ping 6LBR in whatever mode.

Now I don't understand why it's happening now and why the line just above is working as srcipaddr and destipaddr are also not aligned... I guess it's a compiler optimisation going wrong :(

I'll force the alignment within uip_buf and hope it fix the issue.

Btw, in version develop-20150506 the problem does not occur at all.

troymccann, 2015-08-25:
Hi Laurent,

Thanks for looking into this for me. I've switched to the develop-20150506 release and everything seems to be working fine - I can finally load the web server, there are no unexpected reboots, and I can ping the other 802.15.4 nodes on the mesh from the LAN.

I'll keep an eye out for a new release with the alignment fix, otherwise I'll have to keep building on top of 20150506. When I get a chance I'll try to learn what's changed between the two releases.

Now I'm trying to work out how to access the nodes using CoAPS with either copper or californium to confirm CoAP over DTLS is working. Where do I add the pre-shared key to either app? Copper doesn't seem to support CoAPS and californium doesn't have a handshaker.java file as specified in the dtls coap server wiki.

laurentderu, 2015-09-22:
The alignment issue will be fixed in the next development snapshot, it could have solved sooner but there was another problem specific to the OpenMote platform (that will also be solved in the snapshot).

closed 2017-06-15 by laurentderu