cetic/6lbr #96: 6lbr clarification for a hopefully simple use case

msnyder415, 2015-07-21:
Hi 6lbr/Laurent. Pretty cool project you've got here.
I have read the info in this repository, and I could use some clarification.

I followed this tutorial - http://processors.wiki.ti.com/index.php/Cc26xx_sw_examples
Everything here works, except wrapsix is not cooperating. I am not so worried about that right now. 

I would like the BBB to have access to the ipv6 network, which requires using TAP if I read things correctly. My goal is to send coap commands from the BBB to the ipv6/6lowpan network. In this configuration, I do not even need to route any ipv6 traffic, or do ip4 NAT. 

This seems so simple, yet I am totally lost. Every time I try to use bridge mode, I lose the ability to SSH into the BBB. Do you have an example of this topology?

DorianVogel, 2015-07-27:
hey,
I've been able to set this kind of stuff last week (as a test before moving to the Rpi):
On ArchLinux host, configured interface as "routing mode" (bottom here: https://github.com/cetic/6lbr/wiki/6LBR-Interface-Configuration)
changed DEV_ETH to loopback interface (lo)
Allowed me to get the websever, and use the coap extension for Firefox (motes are cc2650DK running cc2650-web-demo from Contiki github)

laurentderu, 2015-07-28:
If you lose ssh connectivity, it means that you ask 6lbr to create the bridge and attach eth0 to it. When an interface join a bridge, it loses all its configuration (and so your ssh connection is cut).

Instead you should use a static bridge configured in /etc/network/interfaces, see https://github.com/cetic/6lbr/wiki/6LBR-Interface-Configuration

closed 2016-03-01 by laurentderu