cetic/6lbr #98: 6lbr + bbb stability

msnyder415, 2015-07-24:
This is a duplicate from my post on TI e2e ( https://e2e.ti.com/support/wireless_connectivity/f/158/t/440989 ) 

I am noticing some strange behavior on the BBB when running 6lbr. 

I have a WSN with 15 sensortags, being polled via coap every 10 minutes, by an adjacent raspberry pi.

The coap requests are for temp/humidity/pressure/lux, one at a time. 

Every now and then, the BBB becomes unresponsive. SSHD no longer accepts connections, and I am forced to reboot.

I have been able to verify that it is not a memory leak. CPU usage is high but not pegged. IO looks reasonable. 

I am using the BBB rev C from Element 14. 

BeagleBoard.org Debian Image 2015-03-01

$ uname -a

Linux beaglebone 3.8.13-bone70 #1 SMP Fri Jan 23 02:15:42 UTC 2015 armv7l GNU/Linux

Contiki-6lbr-1.3.2-34-g3715b49 started with IPV6, RPL

I was able to catch this dump from the kernel before losing my SSH connection.

I am not sure how to interpret this data.  

2015-06-24 13:59:03.588780: DEBUG: SCMD: Packet data report for sid:1 st:0 tx:1

2015-06-24 13:59:03.601485: DEBUG: SCMD: Packet data report for sid:2 st:1 tx:1

2015-06-24 13:59:03.610203: DEBUG: SCMD: Packet data report for sid:3 st:1 tx:1

2015-06-24 13:59:03.734984: DEBUG: SCMD: Packet data report for sid:4 st:1 tx:1

2015-06-24 13:59:03.860382: DEBUG: SCMD: Packet data report for sid:5 st:0 tx:1

Message from syslogd@beaglebone at Jul 24 20:59:05 ...

 kernel:[16989.827809] Internal error: Oops: 17 [#1] SMP THUMB2

Message from syslogd@beaglebone at Jul 24 20:59:05 ...

 kernel:[16989.902886] Process swapper/0 (pid: 0, stack limit = 0xc083c240)

Message from syslogd@beaglebone at Jul 24 20:59:05 ...

 kernel:[16989.909249] Stack: (0xc083de6c to 0xc083e000)

Message from syslogd@beaglebone at Jul 24 20:59:05 ...

 kernel:[16989.916755] de60:                            de0d8100 00000001 00000001 bf9f58bb 00000f73

Message from syslogd@beaglebone at Jul 24 20:59:05 ...

 kernel:[16989.926253] de80: 60000193 de0d8100 df16c2c0 08000400 00000020 e087a800 df177a10 00000000

Message from syslogd@beaglebone at Jul 24 20:59:05 ...

 kernel:[16989.935103] dea0: df0e9b40 c0347ae9 c083c000 60000193 c0048639 de0ac6c0 df005f10 de0ac6c0

Message from syslogd@beaglebone at Jul 24 20:59:05 ...

 kernel:[16989.944048] dec0: 00000013 c083df60 413fc082 00000000 00000000 c007422f 80000193 c0845890

Message from syslogd@beaglebone at Jul 24 20:59:05 ...

 kernel:[16989.952889] dee0: 00000000 df005ec0 df00c680 df005ec0 df005f10 de0ac6c0 00000013 c083df60

Message from syslogd@beaglebone at Jul 24 20:59:05 ...

 kernel:[16989.961825] df00: 413fc082 00000000 00000000 c00743a5 df005ec0 df005f10 00000013 c0075cf7

Message from syslogd@beaglebone at Jul 24 20:59:05 ...

 kernel:[16989.970668] df20: c0075ca5 00000013 00000000 c0073e5b c083a728 c000d0df fa200098 fa2000b8

Message from syslogd@beaglebone at Jul 24 20:59:05 ...

 kernel:[16989.979871] df40: fa2000d8 c00085a9 c000d264 60000033 ffffffff c083df94 c083c000 c04d145b

Message from syslogd@beaglebone at Jul 24 20:59:05 ...

 kernel:[16989.989195] df60: ffffffed 00516000 c0845cd8 00000000 c083c000 c08c3a48 c04d9ca0 c0849488

Message from syslogd@beaglebone at Jul 24 20:59:05 ...

 kernel:[16989.998434] df80: c083c000 413fc082 00000000 00000000 00000010 c083dfa8 c000d263 c000d264

Message from syslogd@beaglebone at Jul 24 20:59:05 ...

 kernel:[16990.111678] dfa0: 60000033 ffffffff c000d251 c000d3b7 c0845630 00000000 c0824a10 c0d4bac0

Message from syslogd@beaglebone at Jul 24 20:59:05 ...

 kernel:[16990.137130] dfc0: 00000000 c07fe58b ffffffff ffffffff c07fe20d 00000000 00000000 c0824a10

Message from syslogd@beaglebone at Jul 24 20:59:05 ...

 kernel:[16990.168355] dfe0: 50c5387d c08444f0 c0824a0c c084947c 80004059 8000807f 00000000 00000000

Write failed: Broken pipe

My questions are -

1 - does anyone know how to interpret this dump?
2 - has anyone else had stability issues? 
3 - are people using 6lbr in production reliably?

I tailed /var/log/syslog, and caught it in action this time.

I see a bunch of these

Jul 24 14:46:37 beaglebone kernel: [ 1114.393103] icmpv6_send: no reply to icmp error
Jul 24 14:46:37 beaglebone kernel: [ 1114.645839] icmpv6_send: no reply to icmp error
Jul 24 14:46:51 beaglebone kernel: [ 1128.790995] icmpv6_send: no reply to icmp error
Jul 24 14:46:55 beaglebone kernel: [ 1132.440308] icmpv6_send: no reply to icmp error

followed by this - 

Jul 24 14:47:53 beaglebone kernel: [ 1190.373438] musb_stage0_irq 790: unhandled DISCONNECT transition (a_idle)
Jul 24 14:47:53 beaglebone kernel: [ 1190.373535] Unable to handle kernel paging request at virtual address fffffff8
Jul 24 14:47:53 beaglebone kernel: [ 1190.381205] pgd = c0004000
Jul 24 14:47:53 beaglebone kernel: [ 1190.384086] [fffffff8] *pgd=9fefe821, *pte=00000000, *ppte=00000000
Jul 24 14:47:53 beaglebone kernel: [ 1190.390769] Internal error: Oops: 17 [#1] SMP THUMB2
Jul 24 14:47:53 beaglebone kernel: [ 1190.396032] Modules linked in: bluetooth rfkill g_multi libcomposite omap_rng mt7601Usta(O)
Jul 24 14:47:53 beaglebone kernel: [ 1190.404979] CPU: 0 Tainted: G O (3.8.13-bone70 #1)
Jul 24 14:47:53 beaglebone kernel: [ 1190.411368] PC is at musb_g_rx+0x44/0x78
Jul 24 14:47:53 beaglebone kernel: [ 1190.415540] LR is at musb_interrupt+0x65/0x64c
Jul 24 14:47:53 beaglebone kernel: [ 1190.420258] pc : [<c0344fd4>] lr : [<c0343491>] psr: 800001b3
Jul 24 14:47:53 beaglebone kernel: [ 1190.420258] sp : c083de6c ip : 00000000 fp : 00000000
Jul 24 14:47:53 beaglebone kernel: [ 1190.432411] r10: de0ac8c0 r9 : 00000000 r8 : 0000000b
Jul 24 14:47:53 beaglebone kernel: [ 1190.437951] r7 : 00000019 r6 : 00000001 r5 : 00000000 r4 : e087adb6
Jul 24 14:47:53 beaglebone kernel: [ 1190.444866] r3 : e087adb0 r2 : 00000000 r1 : ffffffcc r0 : de0b2100
Jul 24 14:47:53 beaglebone kernel: [ 1190.451784] Flags: Nzcv IRQs off FIQs on Mode SVC_32 ISA Thumb Segment kernel
Jul 24 14:47:53 beaglebone kernel: [ 1190.459800] Control: 50c5387d Table: 9e7cc019 DAC: 00000015
Jul 24 14:47:53 beaglebone kernel: [ 1190.465885] Process swapper/0 (pid: 0, stack limit = 0xc083c240)
Jul 24 14:47:53 beaglebone kernel: [ 1190.472244] Stack: (0xc083de6c to 0xc083e000)

Message from syslogd@beaglebone at Jul 24 14:47:53 ...
kernel:[ 1190.390769] Internal error: Oops: 17 [#1] SMP THUMB2

Message from syslogd@beaglebone at Jul 24 14:47:53 ...
kernel:[ 1190.465885] Process swapper/0 (pid: 0, stack limit = 0xc083c240)

laurentderu, 2015-07-28:
5copying here the response from 6lbr-dev ml)

Hello,

Unfortunately the BBB has some stability issues regarding USB controller, we have added a workaround to force usb scanning, but you could also have long term stability. Kernel from Robert Nelson seems to improve the stability much though (See http://elinux.org/Beagleboard:Debian_On_BeagleBone_Black and the links for prebuild images)

On RPi (and BBW on Angstrom) we have uptime higher that 1500h. They could have been higher, but we had to stop them to test new version :)

xianlin, 2015-10-02:
@msnyder415 we encountered the same unstable issue with BBB using the same setup.

when running just 1 sensortag with default contiki3.0 example `cc26xx-web-demo.bin`, the BBB will not hang for 2 or 3 days.

when running just 1 or 2 sensortag with default contiki3.0 example `cc26xx-web-demo.bin`, the BBB will not hang for 2 or 3 days.

when running just 1 sensortag with default contiki3.0 example `cc26xx-web-demo.bin`, the BBB will not hang for 2 or 3 days.

when running 3 or above sensortag with default contiki3.0 example `cc26xx-web-demo.bin`, the BBB will hang after about 2 or 3 hours.

We tried many ways to troubleshoot and eventually gave up, now using raspberry pi as border router and it runs for almost one week without failing us.

closed 2016-03-01 by laurentderu