cetic/6lbr #99: Can we discover the resources of 6LBR using Discovey Resource...?

Krishna421TSP, 2015-07-27:
Hi..every one..
I have added hello world and led toggle resources in coap server which is running on 6lbr.

I am able to discover the resources of other nodes from the web server(6lbr) and access them.But i am not able to discover the resources of 6lbr when i do 
coap://[aaaa::212:4b00:42b:3cc3]:5683/.well-known/core it is nor responding at all.
[aaaa::212:4b00:42b:3cc3]  is My 6lbr ip address .

Any suggestion would be greatly appreciated..

Thanks..

laurentderu, 2015-07-28:
I'm not sure if the coap-server is automatically enabled in 1.3.x or not, also, the default coap-server.c does only contain the Device resource.
If you need more resources you need to update coap-server.c like 6lbr-demo/apps/coap/coap-server.c and enable it in the makefile.

In the upcoming new release, the coap-server will be enabled by default and additional resource can be added by plugins or user modules.

Krishna421TSP, 2015-07-30:
HI..lauren..

Thanks for your reply..
I am able to add the resources to coap server like 6lbr demo and enabled and compiled it also.
But i am not able to discover those resources from 6lbr web server since discover(.well-known/core) resource is not working.

Any suggestions on this problem..

laurentderu, 2015-07-30:
Do you see the trace "CoAP server started" in the log ? if so, could you enable debug traces in er-coap-engine.c and send me the traces ?

Krishna421TSP, 2015-07-30:
Hi..
Please find below traces..

handle_incoming_data(): received uip_datalen=21 
receiving UDP datagram from: [bbbb:0000:0000:0000:1d20:951d:17e9:cc4e]:65382
  Length: 21
  Data: 01000000 00000001 00001001 01110001 10111011 00101110 01110111 01100101 01101100 01101100 00101101 01101011 01101110 01101111 01110111 01101110 00000100 01100011 01101111 01110010 01100101 
  Parsed: v 1, t 0, tkl 0, c 1, mid 2417
  URL: .well-known/core
  Payload: 
res: /.well-known/core (#0x2000021c)
pos: s0, o0, b0
res: /hello (#0x200001c4)
pos: s25, o0, b25
BUF 74: </.well-known/core>;ct=40,</hello>;title="Hello world: ?len=0.."
res: DONE

Blockwise: no block option for blockwise resource, using block size 128

handle_incoming_data(): received uip_datalen=21 
receiving UDP datagram from: [bbbb:0000:0000:0000:1d20:951d:17e9:cc4e]:65382
  Length: 21
  Data: 01000000 00000001 00001001 01110001 10111011 00101110 01110111 01100101 01101100 01101100 00101101 01101011 01101110 01101111 01110111 01101110 00000100 01100011 01101111 01110010 01100101 
  Parsed: v 1, t 0, tkl 0, c 1, mid 2417
  URL: .well-known/core
  Payload: 
res: /.well-known/core (#0x2000021c)
pos: s0, o0, b0
res: /hello (#0x200001c4)
pos: s25, o0, b25
BUF 74: </.well-known/core>;ct=40,</hello>;title="Hello world: ?len=0.."
res: DONE
Blockwise: no block option for blockwise resource, using block size 128.

I am keep getting above messages but not able to discover resources..

laurentderu, 2015-07-30:
The block size 128 seems strange, the default value on 6LBR is 64. Which version of 6LBR are you using ? Did you do some modification in the configuration ? Do you only have /hello resource ?

Krishna421TSP, 2015-07-30:
Hi..lauren..
Thanks for your reply...

I have changed the block size to 128.it was giving below error with block size 64.

handle_incoming_data(): received uip_datalen=21 
receiving UDP datagram from: [bbbb:0000:0000:0000:1d20:951d:17e9:cc4e]:65382
Length: 21
Data: 01000000 00000001 00001001 01110001 10111011 00101110 01110111 01100101 01101100 01101100 00101101 01101011 01101110 01101111 01110111 01101110 00000100 01100011 01101111 01110010 01100101 
Parsed: v 1, t 0, tkl 0, c 1, mid 2417
URL: .well-known/core
Payload: 
  
  
res: /.well-known/core (#0x2000021c)
pos: s0, o0, b0
res: /hello (#0x200001c4)
pos: s25, o0, b25

BUF 74: </.well-known/core>;ct=40,</hello>;title="Hello world: ?len=0.."
res: MORE at hello (#0x200001c4)
Blockwise: no block option for blockwise resource, using block size 64
handle_incoming_data(): received uip_datalen=21 
receiving UDP datagram from: [bbbb:0000:0000:0000:1d20:951d:17e9:cc4e]:65382
  Length: 21
  Data: 01000000 00000001 00001001 01110001 10111011 00101110 01110111 01100101 01101100 01101100 00101101 01101011 01101110 01101111 01110111 01101110 00000100 01100011 01101111 01110010 01100101 
  Parsed: v 1, t 0, tkl 0, c 1, mid 2417
  URL: .well-known/core
  Payload: 
res: /.well-known/core (#0x2000021c)
pos: s0, o0, b0
res: /hello (#0x200001c4)
pos: s25, o0, b25
BUF 74: </.well-known/core>;ct=40,</hello>;title="Hello world: ?len=0.."
res: MORE at hello (#0x200001c4)
Blockwise: no block option for blockwise resource, using block size 64
handle_incoming_data(): received uip_datalen=21 
receiving UDP datagram from: [bbbb:0000:0000:0000:1d20:951d:17e9:cc4e]:65382
  Length: 21
  Data: 01000000 00000001 00001001 01110001 10111011 00101110 01110111 01100101 01101100 01101100 00101101 01101011 01101110 01101111 01110111 01101110 00000100 01100011 01101111 01110010 01100101 
  Parsed: v 1, t 0, tkl 0, c 1, mid 2417
  URL: .well-known/core
  Payload: 
res: /.well-known/core (#0x2000021c)
pos: s0, o0, b0
res: /hello (#0x200001c4)
pos: s25, o0, b25
BUF 74: </.well-known/core>;ct=40,</hello>;title="Hello world: ?len=0.."
res: MORE at hello (#0x200001c4)
Blockwise: no block option for blockwise resource, using block size 64

laurentderu, 2015-07-30:
Could you point me the error you had, I don't see it in your log.

Are you able to ping and also reach /hello resource ?

Krishna421TSP, 2015-07-30:
Hi..
If i do ping below error i am getting...and i am not able to reach hello resource..
In web server it just showing Server not responding.............

handle_incoming_data(): received uip_datalen=4 

receiving UDP datagram from: [bbbb:0000:0000:0000:1d20:951d:17e9:cc4e]
r:62638
  Length: 4
  Data: 01000000 00000000 00010110 10100000 
  
rParsed: v 1, t 0, tkl 0, c 0, mid 5792
  
rURL: 
  
rPayload: 

rReceived Ping

rERROR 195: 

handle_incoming_data(): received uip_datalen=4 

receiving UDP datagram from: [bbbb:0000:0000:0000:1d20:951d:17e9:cc4e]
r:62638
  Length: 4
  Data: 01000000 00000000 00010110 10100000 
  
rParsed: v 1, t 0, tkl 0, c 0, mid 5792
  
rURL: 
  
rPayload: 

rReceived Ping

rERROR 195: 

handle_incoming_data(): received uip_datalen=4 

receiving UDP datagram from: [bbbb:0000:0000:0000:1d20:951d:17e9:cc4e]
r:62638
  Length: 4
  Data: 01000000 00000000 00010110 10100000 
  
rParsed: v 1, t 0, tkl 0, c 0, mid 5792
  
rURL: 
  
rPayload: 

rReceived Ping

rERROR 195: 

handle_incoming_data(): received uip_datalen=4 

receiving UDP datagram from: [bbbb:0000:0000:0000:1d20:951d:17e9:cc4e]
r:62638
  Length: 4
  Data: 01000000 00000000 00010110 10100000 
  
rParsed: v 1, t 0, tkl 0, c 0, mid 5792
  
rURL: 
  
rPayload: 

rReceived Ping

rERROR 195:

laurentderu, 2015-07-30:
But you can reach the web server using http, right ? Which version of 6LBR are you using ?

Could you send me a pcap while you try to access /.well-known/core or /hello ?

Krishna421TSP, 2015-07-30:
Yes i am able to reach web server using http.
And 6LBR version 1.3.2..
Please find attched error response..
[image]

laurentderu, 2015-07-30:
I mean by pcap a network traffic capture made using wireshark or tcpdump :)

closed 2016-02-29 by laurentderu