cetic/6lbr #100: Rpi Router mode + ip level routing

DorianVogel, 2015-07-30:
Hello all,
I'm getting quite a bit of troubles to access the webserver running on a Rpi from my laptop.
Here is the topology:
(all ip set up manually using "ip [-6] addr add")
laptop ethernet: enp0s20u1 ipv4: 10.42.0.1/24 ipv6: cccc:1/64
                            ||
                            V
Rpi ethernet: eth0 ipv4: 10.42.0.2/24 ipv6: cccc::10/64
Rpi tap0: ipv6: bbbb::7f44:c563:5759:7e7/64

laptop and rpi both have route to cccc::/64 (automatically added when assigned ip)
laptop route to bbbb::/64 added manually (via cccc::/64)

The goal is to set up the router mode to be able to ssh the Rpi while having access to the wsn network from the laptop.
6lbr.conf:
```
#MODE=ROUTER

RAW_ETH=0
BRIDGE=0
DEV_ETH=eth0
DEV_TAP=tap0

DEV_RADIO=/dev/ttyUSB0

BAUDRATE=115200

DEBUG_LEVEL=12
```
So, from here, I'm able to ping both eth0 and tap0 from the laptop, and enp0s20u1 from Rpi. When trying to get http://[bbbb::100] from the laptop, I can see requests coming on tap0 using ``` tshark -i tap0``` on Rpi
However, tap0 newer replies, and I'm unable to get the web interface.
Using wget http://[bbbb::100] from Rpi confirms that the web-server is running.

Is there something I missed about this router mode ?
Cherrs,
Dorian

laurentderu, 2015-07-30:
Hi,

You need to bridge eh0 and tap0 in order to have access to 6LBR and the WSN from the external world, see https://github.com/cetic/6lbr/wiki/6LBR-Interface-Configuration

DorianVogel, 2015-07-30:
Okay, thanks for your answer,
correct me if I'm wrong, but doesn't this mean to use the "bridge" mode ? my goal is to use the "router" mode (bottom of your link).

laurentderu, 2015-07-30:
If you only need to connect 6LBR and the WSN to a single LAN, you don't need "router" configuration, "bridge" configuration is the best for that case. Actually, it should be named "custom" configuration, as you can link the interfaces either by routing or switching or tunneling, ... So, it's only meant for complex network configuration.

Maybe you are confusing the interface configuration with the runtime mode of 6LBR, SmartBridge and Router, these are completely independent of the interface configuration (although their names are sadly quite similar...)

DorianVogel, 2015-07-31:
Thanks a lot for your tips,
The Rpi is now working in router mode with bridged connection, and I'm able to ssh into it.
6lbr.conf:
```
MODE=ROUTER

RAW_ETH=0
BRIDGE=1
DEV_ETH=eth0
DEV_TAP=tap0
DEV_BRIDGE=br0
DEV_RADIO=/dev/ttyUSB0

CREATE_BRIDGE=MANUAL
ETH_JOIN_BRIDGE=0

BAUDRATE=115200

DEBUG_LEVEL=12
```
using the /etc/network/interfaces example script given in https://github.com/cetic/6lbr/wiki/6LBR-Interface-Configuration, the Rpi creates bridge interface on boot, with the same IP as eth0 (determined by dhcp on my laptop) and attaches eth0 to it. 
6lbr.conf specifies that the bridge is created manually before 6lbr starts up. This allows to ssh to the device even if 6lbr didn't start (e.g. slip-radio device not connected).

laurentderu, 2015-08-05:
Thanks for your feedback, I updated the wiki to make it more clear.

closed 2015-08-05 by laurentderu