cetic/6lbr #101: Start Wrapsix Problem on Beaglebone Black Board

karlrenner214, 2015-08-01:
I am following the procedure at http://processors.wiki.ti.com/index.php/Cc26xx_sw_examples to run 6lbr and wrapsix on a Beaglebone Black board.  After wrapsix is starting it seems to hang with no prompt returned.  Is this correct behavior?

debian@beaglebone:~$ sudo wrapsix &
[5] 10331
debian@beaglebone:~$ [Info] WrapSix 0.2.0 is starting

laurentderu, 2015-08-05:
You should have some log information displaying the interface and ip addresses used (at least with the latest version of wrapsix)

Zjion, 2015-09-17:
I believe it is correct behaviour. I am also running the set-up in the wiki. The steps are as follows in the guide:
screen
press any key
sudo wrapsix &
ctrl-A D to get out of screen

laurentderu, 2015-09-17:
Thanks for the confirmation

closed 2015-09-17 by laurentderu