cetic/6lbr #102: CC2538+ENC28J60 - Console verbosity, 6LBR.conf, slip radio?

troymccann, 2015-08-05:
Sorry for the flood of questions. I feel like the answers should be obvious to me, but after hours of trawling through past issues and mailing lists I still have no idea what I'm doing wrong.

I've got the 6LBR starting and I've definitely got the ENC hooked up to the SmartRF board correctly, but the console output stops at ENC: resetting chip.

```
Contiki 3.x
TI SmartRF06 + cc2538EM
 Net: sicslowpan
 MAC: CSMA
 RDC: nullrdc
 Channel: 25
 PAN-ID: abcd
Rime configured with address 00:12:4b:00:04:13:36:fc
INFO: 6LBR: Starting 6LBR version 1.4.x (Contiki 3.x)
INFO: NVM: Reading 6LBR NVM
INFO: NVM: NVM Magic : 2009
INFO: NVM: NVM Version : 2
INFO: ETH: ENC28J60 init
INFO: ETH: Eth MAC address : 06:00:04:13:36:fc
INFO: ENC: resetting chip
```

Is there a method to increase the verbosity of the console output, so I can see if there's anything else happening? I've found that if I was using 6LBR on Linux I could control the verbosity of the log files using settings in 6LBR.conf, but if I'm just using the CC2538+ENC28J60 combination on their own can I still do this? Also, how do I go about changing NVM config?

I've tried connecting the ENC to a Macbook Pro to sniff packets using Wireshark, but there's not a single packet of data coming from the 6LBR.

Ideally I want to have the CC2538 6LBR connected to a modem/router running IPv6 autoconfig, so that I can access it using other devices on the LAN, and eventually the internet using something like 6to4.

Thanks in advance.

laurentderu, 2015-08-05:
You should have a lot more traces after "reseting chip" (ending with 6LBR started), so the ENC driver is stuck (most probably waiting for an ACK from ENC). That means that there is a wiring problem between the CC2538 and the ENC, check if you are using SPI or GPIO mode and if the pins are properly configured.

laurentderu, 2015-08-05:
For the verbosity, currently you can not change it at runtime on a CC2538, you can modify it in 6lbr-conf-cc2538dk.h by changing LOG6LBR_LEVEL_DEFAULT

troymccann, 2015-08-06:
Oh, great. Yep, I've got extra verbosity. Also, turns out I did have the wrong pin connected to CS on the ENC. For future reference if anyone is unsure, the default ENC CS pin for the CC2538 with the SmartRF06 board is RF.17 aka LCD_CS. I thought it was RF2.10.

Now I'm getting a lot more activity at the console output. But I can't connect to it's web server when either using router or smart bridge mode. In Smart Bridge mode I can't see any packets being sent from it, and in router mode I can see a lot of Neighbour Solicitations, but the source address is just :: . Here's a few entries from Wireshark showing what I think to be the 6LBR trying to talk with the network:
```
193	491.691939000	::	ff02::1:ff00:100	ICMPv6	78	Neighbor Solicitation for bbbb::100
194	491.789503000	::	ff02::1:ff13:36fc	ICMPv6	78	Neighbor Solicitation for aaaa::212:4b00:413:36fc
195	492.164502000	::	ff02::1:ff13:36fc	ICMPv6	78	Neighbor Solicitation for fe80::400:4ff:ff13:36fc
196	492.258242000	::	ff02::1:ff13:36fc	ICMPv6	78	Neighbor Solicitation for fe80::212:4b00:413:36fc
```

On the Mac, I can't add the aaaa:: bbbb::100 route with a 'link-local only' or 'automatically configure' which both use an FE80 address for en1, I can only do it if I manually set the address to something like bbbb::200. That said, even though I've got IPv6 autoconfig running on my modem/router, and the Mac can ping6 my Windows computer, Android phone, etc. when set to 'automatically' configure IPv6, it doesn't show what the IP address or router address is in System Preferences; but when I do use ping6 it looks like a link-local address. I'm not sure if that could indicate an issue with OS X instead.

If I do manually set en1 to bbbb::200 and try to ping bbbb::100, wireshark shows the following, but the ping6 never receives any response. With an 'automatically' configured address, or 'link-local only', ping6 returns a 'no route to host'.
```
491	1648.342599000	bbbb::200	ff02::1:ff00:100	ICMPv6	86	Neighbor Solicitation for bbbb::100 from 40:6c:8f:30:c9:8c
```

This is what I'm getting at the console output, so it looks like the IPv6 addresses are set on the 6LBR.
```
Contiki 3.x
TI SmartRF06 + cc2538EM
 Net: sicslowpan
 MAC: CSMA
 RDC: nullrdc
 Channel: 25
 PAN-ID: abcd
Rime configured with address 00:12:4b:00:04:13:36:fc
INFO: 6LBR: Starting 6LBR version 1.4.x (Contiki 3.x)
INFO: NVM: Reading 6LBR NVM
INFO: NVM: NVM Magic : 2009
INFO: NVM: NVM Version : 2
INFO: ETH: ENC28J60 init
TRACE: 6LE: Low2Eth translate : 00:12:4b:00:04:13:36:fc
TRACE: 6LE: Low2Eth adding prefix
TRACE: 6LE: Low2Eth Lowpan addr : 00:12:4b:00:04:13:36:fc
TRACE: 6LE: Low2Eth Ethernet addr : 06:00:04:13:36:fc
INFO: ETH: Eth MAC address : 06:00:04:13:36:fc
INFO: ENC: resetting chip
DEBUG: ENC: REV 6
INFO: ETH: ENC-28J60 Process started
INFO: LLSEC: Using 'nullsec' llsec driver
INFO: 6LBR: Security layer initialized
INFO: 6LBR: Tentative local IPv6 address fe80::212:4b00:413:36fc
INFO: 6LBR: Tentative global IPv6 address (WSN) aaaa::212:4b00:413:36fc
INFO: 6LBR: Tentative global IPv6 address (ETH) bbbb::100
DEBUG: RIO: Adding route information: length=64, flags=0, route lifetime=1800, dest=aaaa::
INFO: 6LBR: RA Daemon enabled
INFO: 6LBR: Checking addresses duplication
PACKET: PF: bridge_output: Sending packet to Broadcast
PACKET: PF: wireless_output: sending packet
PACKET: PF: eth_output: broadcast
TRACE: 6LE: translateIP: ICMP Message detected
PACKET: PF: eth_output: Sending packet to Ethernet
PACKET: ETH: write: 78
```

Thanks again. Once I have a good understanding of what's going on I'll be sure to contribute to the wiki!

troymccann, 2015-08-06:
I think I've solved a part of the problem. Still making sense of IPv6, I didn't realise each physical port had multiple IPv6 addresses. I changed the settings on my main router (Billion 7800NXL) to use a prefix of bbbb::/64 rather than a randomly generated one, and now each of the devices on my network have a couple of bbbb::/64 address.

That said, I still can't access the web server from anywhere and the wireshark capture is still showing, from what I presume is, the 6LBR sending neighbour solicitations, but with an unspecified source address.

Attempting to ping6 bbbb::100, there's 100% packet loss and wireshark shows the following:
```
5731	448.791970000	bbbb::6aa8:6dff:fe4f:2588	ff02::1:ff00:100	ICMPv6	86	Neighbor Solicitation for bbbb::100 from 68:a8:6d:4f:25:88
```

laurentderu, 2015-08-06:
Hello,

The NS with unspecified source address is normal behavior, this happens at startup, the stack check if the addresses are already in use or not (it's the Duplicate Address Detection mechanism of IPv6).

However, once done, the ipv6 stack should reply to NS coming from your LAN. With the trace level set to the max, do you see any incoming packet on the ENC interface in the log ?

troymccann, 2015-08-06:
Hi Laurent,

Yep, there's plenty of received packets. Here's a ~5 second capture of the console from startup:

```
Contiki 3.x
TI SmartRF06 + cc2538EM
 Net: sicslowpan
 MAC: CSMA
 RDC: nullrdc
 Channel: 25
 PAN-ID: abcd
Rime configured with address 00:12:4b:00:04:13:36:fc
INFO: 6LBR: Starting 6LBR version 1.4.x (Contiki 3.x)
INFO: NVM: Reading 6LBR NVM
INFO: NVM: NVM Magic : 2009
INFO: NVM: NVM Version : 2
INFO: ETH: ENC28J60 init
TRACE: 6LE: Low2Eth translate : 00:12:4b:00:04:13:36:fc
TRACE: 6LE: Low2Eth adding prefix
TRACE: 6LE: Low2Eth Lowpan addr : 00:12:4b:00:04:13:36:fc
TRACE: 6LE: Low2Eth Ethernet addr : 06:00:04:13:36:fc
INFO: ETH: Eth MAC address : 06:00:04:13:36:fc
INFO: ENC: resetting chip
DEBUG: ENC: REV 6
INFO: ETH: ENC-28J60 Process started
INFO: LLSEC: Using 'nullsec' llsec driver
INFO: 6LBR: Security layer initialized
INFO: 6LBR: Tentative local IPv6 address fe80::212:4b00:413:36fc
INFO: 6LBR: Tentative global IPv6 address (WSN) aaaa::212:4b00:413:36fc
INFO: 6LBR: Tentative global IPv6 address (ETH) bbbb::100
DEBUG: RIO: Adding route information: length=64, flags=0, route lifetime=1800, dest=aaaa::
INFO: 6LBR: RA Daemon enabled
INFO: 6LBR: Checking addresses duplication
PACKET: PF: bridge_output: Sending packet to Broadcast
PACKET: PF: eth_output: broadcast
TRACE: 6LE: translateIP: ICMP Message detected
PACKET: PF: eth_output: Sending packet to Ethernet
PACKET: ETH: write: 78
DUMP: ETH: 
         3333ff00 01000600 041336fc 86dd6000 
         00000018 3aff0000 00000000 00000000 
         00000000 0000ff02 00000000 00000000 
         0001ff00 01008700 bcec0000 0000bbbb 
         00000000 00000000 00000000 0100
PACKET: ENC: tx: 78: 33:33:ff:00:01:00
PACKET: ENC: sent_packets 1
PACKET: PF: bridge_output: Sending packet to Broadcast
PACKET: PF: wireless_output: sending packet
PACKET: PF: eth_output: broadcast
TRACE: 6LE: translateIP: ICMP Message detected
PACKET: PF: eth_output: Sending packet to Ethernet
PACKET: ETH: write: 78
DUMP: ETH: 
         3333ff13 36fc0600 041336fc 86dd6000 
         00000018 3aff0000 00000000 00000000 
         00000000 0000ff02 00000000 00000000 
         0001ff13 36fc8700 10cd0000 0000aaaa 
         00000000 00000212 4b000413 36fc
PACKET: ENC: tx: 78: 33:33:ff:13:36:fc
PACKET: ENC: sent_packets 2
PACKET: PF: bridge_output: Sending packet to Broadcast
PACKET: PF: wireless_output: sending packet
PACKET: PF: eth_output: broadcast
TRACE: 6LE: translateIP: ICMP Message detected
PACKET: PF: eth_output: Sending packet to Ethernet
PACKET: ETH: write: 78
DUMP: ETH: 
         3333ff13 36fc0600 041336fc 86dd6000 
         00000018 3aff0000 00000000 00000000 
[... 263 lines cut ...]
DUMP: ETH: 
         3333ff00 01000600 041336fc 86dd
PACKET: PF: eth_input: Dropping packet type=0x0800
PACKET: PF: bridge_output: Sending packet to Broadcast
PACKET: PF: wireless_output: sending packet
PACKET: PF: eth_output: broadcast
TRACE: 6LE: translateIP: ICMP Message detected
PACKET: PF: eth_output: Sending packet to Ethernet
PACKET: ETH: write: 78
DUMP: ETH: 
         3333ff13 36fc0600 041336fc 86dd6000 
         00000018 3aff0000 00000000 00000000 
         00000000 0000ff02 00000000 00000000 
         0001ff13 36fc8700 10cd0000 0000aaaa 
         00000000 00000212 4b000413 36fc
PACKET: ENC: tx: 78: 33:33:ff:13:36:fc
PACKET: ENC: sent_packets 4
PACKET: ENC: EPKTCNT 0x01
PACKET: ENC: rx: 60: 01:80:c2:00:00:00
PACKET: ENC: received_packets 3
PACKET: ETH: read: 14
DUMP: ETH: 
         3333ff13 36fc0600 041336fc 86dd
PACKET: PF: eth_input: Dropping packet type=0x0026
```

troymccann, 2015-08-06:
It's difficult for me to read the console because my terminal program retains limited output, but in the second log in the above post it looks like 6LBR is rebooting again, which is unusual right?

On further inspection, it's doing that quite often, as seen by watching the console output and the 3 LED sequence. Could it be that it's rebooting before it establishes it's IP?

laurentderu, 2015-09-22:
See #95

closed 2015-09-22 by laurentderu