cetic/6lbr #105: AES encryption for 802.15.4 communciations with TI CC2650 SensorTag 2

xianlin, 2015-10-02:
Hi I am trying to enable Contiki 3.0 security features **AES encryption**. While I am exploring how to enable it on the sensortag firmware, I realized my 6lbr slip radio which is a **cc2530 USB Dongle** need to enable the AES as well.

So sensortag and the USB dongle will communicate with encryption as the 802.15.4 protocol supposed to do.

Anyone here from cetic had successfully used AES between a Border Router and the nodes? Using TI CC2650 SensorTag 2.0?

If yes, could you please share how you did it? Thank you.

gillesDD, 2015-10-03:
AES encryption is to be configured from the WEB interface of 6lbr.
YOu don't have to activate encryption in the cc2531 dongle

xianlin, 2015-10-05:
Thank you and the investigation on this issue is moved to https://github.com/contiki-os/contiki/issues/1295.

closed 2015-10-05 by xianlin