cetic/6lbr #109: cc2650 as slip radio running getting ERROR: SLIP Unknown Command

xianlin, 2015-10-22:
I follow the below guide to compile slip-radio.bin for TI sensortag 2.0 and I can access the 6lbr web page at http://[bbbb::100]. 

http://wiki.beyondlogic.org/index.php?title=6LOWPAN_Contiki_6lbr_SLIP_Radio_CC2650_SensorTag 

However the network nodes are not appearing on the 6lowpan network and I have set the channel correctly. 

I have used the latest Contiki code for `slip-radio` and 6lbr version is `1.3.3 (Contiki-6lbr-1.3.2-34-g3715b49)`.

I noticed the 6lbr log file says as below and no more output (6blr LOG_LEVEL=3)

```
ERROR: SLIP: Unknown command
```

laurentderu, 2015-10-22:
Hi,

Sadly the official Contiki slip-radio does not have support for set channel command, and, until the new radio api his supported by all radio, you have to manually add it to the slip-radio firmware.

The cetic repo does not have a support for CC26xx in slip-radio, but it has support for cc2538, so you can easily duplicate the code for the CC26xx

xianlin, 2015-10-22:
I hard coded the channel to 26 in the code and done other changes according to the guide here:

https://github.com/contiki-os/contiki/blob/master/platform/srf06-cc26xx/README.md#cc13xxcc26xx-slip-radio-with-6lbr 

After that although the log file says `SLIP: Unknown command`, my sensortags still can access the IBM mqtt broker via 6lbr host with slip radio (another sensortag 2.0) attached via Devpack.

closed 2016-02-29 by laurentderu

supriya20, 2016-08-02:
Hi Xianlin,

I am trying to do the same as you have done that is CC2650 as Slip Radio and 6lbr running on my BeagleBone Green Wireless. I am getting the same error i.e., "ERROR: SLIP: Unknown command" and with that an other error as "FATAL: TAP: ioctl() : No such device" and I am not able to access the webpage. 

But you have mentioned that you are able to access the webpage. Could you please tell me how you made it work...?

Thanks and Regards,
Supriya

xianlin, 2016-08-03:
@supriya20 , that is long time ago and I don't remember too much. I am not using BBG, I am using Raspberry Pi 2. Will reply to you if I revisit this old project again. Sorry.