cetic/6lbr #110: after updating to jessie on RPi, 6lbr stop working

janeksz, 2015-10-28:
was working on RPi whizzy, after system upgrade to Jessie stop working.

laurentderu, 2015-10-29:
Did you upgrade your RPi or did you use a new flash with Jessie ?

Could you send the 6lbr.log file as well as your 6lbr.conf file and /etc/network/interfaces file ?

janeksz, 2015-10-30:
[image]

[image]

[image]
I have a new Jessie image.

The file /etc/network/interfaces is the same as in previous OS version

6lbr fails on HTTP request /sensors this time

The files are attached as requested (rename before opening)

 

Regards

Jan

2015-09-30 22:09:51.926587: INFO: 6LBR: Starting as RPL ROUTER
2015-09-30 22:09:51.927313: INFO: UDPS: UDP server started
2015-09-30 22:09:51.927533: INFO: 6LBR: CETIC 6LBR Started
2015-09-30 22:09:51.927660: INFO: SLIP: cc2531: Setting RF channel 0x19
2015-09-30 22:10:04.901496: INFO: HTTP: Script : /
CMD output: EUnknown command 2152 (4)
2015-09-30 22:10:30.618481: ERROR: BR-RDC: br-rdc: send failed, slip ack timeout (6)
2015-09-30 22:10:30.820578: INFO: HTTP: Script : /sensors.html
2015-09-30 22:10:30.889564: INFO: HTTP: Script : /6lbr_layout.css
2015-09-30 22:10:35.25777: INFO: HTTP: Script : /sensors.html
2015-09-30 22:10:35.167791: INFO: HTTP: Script : /6lbr_layout.css
2015-09-30 22:10:38.32442: INFO: HTTP: Script : /sensors.html
2015-09-30 22:10:38.48594: INFO: HTTP: Script : /6lbr_layout.css
2015-09-30 22:10:42.172616: INFO: HTTP: Script : /sensors.html
2015-09-30 22:10:42.206697: INFO: HTTP: Script : /6lbr_layout.css
2015-09-30 22:10:50.218307: INFO: HTTP: Script : /sensors.html
2015-09-30 22:10:50.231394: INFO: HTTP: Script : /6lbr_layout.css
2015-09-30 22:10:54.307652: ERROR: BR-RDC: br-rdc: send failed, slip ack timeout (10)
2015-09-30 22:10:56.669615: ERROR: BR-RDC: br-rdc: send failed, slip ack timeout (11)
2015-09-30 22:10:59.175088: ERROR: BR-RDC: br-rdc: send failed, slip ack timeout (12

janeksz, 2015-10-30:
unable to attach files

laurentderu, 2015-11-13:
Without the full log I can not really check what is going on, you can send it to me by mail or using a file sharing tool ?

janeksz, 2015-11-14:
more info here: https://drive.google.com/folderview?id=0BxIxpti4YGOyTEtqdEFGUTR6bWs&usp=sharing

laurentderu, 2015-11-17:
It seems the radio serial port is stuck. Either the FTDI driver provided with jessie does no longer support Econotag or there is another process hijacking the serial device. I will have a closer look at this.

(The two 6lbr process is normal I think, one of them is the watchdog and the other is the launching script)

janeksz, 2015-11-17:
I have tried 2 SLIP radios, specifically CC2531 USB stick and CC2650 sensortag both of them with the same negative result with Jessie, working with Wheezy.

msolters, 2015-12-28:
Did you make sure to add `dwc_otg.speed=1` to `/boot/cmdline.txt` when you flashed your new image?

Source: https://github.com/cetic/6lbr/wiki/RaspberryPi-Software-Configuration#specific-configuration-for-usb-slip-radio-motes

janeksz, 2015-12-29:
Hi Mark,

 

 Surprisingly after following your link I discovered that there is no /etc/inittab in a new distro (???)

After copying it from previous version my RPI 6lbr edge router works now.

Thank you very much.

 

Jan

closed 2015-12-29 by janeksz