cetic/6lbr #112: Border Router Using "Native" Radio (AT86RF233)

msolters, 2015-12-06:
Hi, I have 6LBR setup on a Raspberry Pi.  I have tried and found great success following the documented approaches whereby at least one of my mesh's nodes is attached to the RPi as a SLIP radio.

However, I have now attached [ATMEL's 802.15.4 radio transceiver](http://www.atmel.com/tools/ATREB233-XPRO.aspx?tab=documents) directly to my RPi.  I then create a 802.15.4 `wpan0` interface for it (e.g. as seen in https://github.com/RIOT-OS/RIOT/wiki/How-to-install-6LoWPAN-Linux-Kernel-on-Raspberry-Pi#raspberrypi-with-openlabs-raspberry-pi-802154-radio).  I know the radio is functional because, when set to the correct channel, it picks up RX/TX data from my mesh.

Provided I have a network interface `wpan0` that is an 802.15.4 radio, how do I choose this as my `DEV_RADIO`??  As far as I understand it is not a serial line device but simply an 802.15.4 radio.  Considering that `DEV_ETH=wlan0` or `DEV_ETH=eth0`, etc. it seems it would make sense to allow `DEV_RADIO=wpan0` or equivalent.

msolters, 2015-12-06:
I have found this discussion (http://lists.cetic.be/pipermail/6lbr-dev/2014-October/000356.html) but I'm afraid it was not sufficiently concrete for me to understand how I would actually configure my `wpan0` 802.15.4 interface in the 6LBR settings.

YinxuWang, 2015-12-22:
Hi, msolter.

I guess the thing is not so easy as you think. You need to read the code and understand it well. 

The Dev_Radio was like ttyAMA0/ttyUSB0. It was USB-serial/FTDI USB device. If you are using Raspberry Pi, after you plugged the transceiver dongle into Pi, you can find if the dongle device is detected by tying command lsusb. Then, you can see what the dongle enumerates, maybe ttyAMAx/ttyUSBx or maybe nothing, which means you need install drivers. After adding it like Dev_Radio=ttyAMAx, Pi can communicate with it.

Also, the border router is implemented as using serial port to communicate between the RDC layer on border router and network layer on slip-radio. So if you are using a transceiver, it should at least work as it has a slip-net layer. Possibly, you need to implement it by yourself by referring the original code. 

Hope this helps.

Yinxu

msolters, 2015-12-23:
Sure, the differentiation between the slip framed traffic and the actual traffic is something I do understand.

But it seems to me that it would be faster to be able to (optionally) bypass the SLIP framing completely rather than have to add it to a radio that doesn't need it?

phalloc, 2016-03-14:
hello all,
i believe i am attempting the same setup as @msolters (a raspberry with the openlabs 802.15.4 radio and the 6LBR) but it was not clear to me if this could be done and if you achieved any progress.
has there been any progress or successfull attempts at using this radio with the 6LBR?
if so can you share the necessary steps and guides?

thank you,
Tiago

laurentderu, 2016-03-14:
Hello,

It is possible, yet never tested by us :) A radio driver for the Linux WPAN interface has been integrated recently in Contiki and can be used by 6LBR. But you will need to remove the (few) explicit references to the slip-radio driver in 6LBR code.
I don't think we have the needed hardware, but I can provide some assistance and we can include the result in 6LBR.

phalloc, 2016-03-14:
hello @laurentderu

i'm not very familiar with the contiki-os and 6lbr source but i have the hardware ready.
if the team provides some guidance i can attempt a contribution.

phalloc, 2016-03-15:
how should i approach this issue? where to look and what to adapt ?

SitanshuNanavati, 2016-07-21:
+1 
I have same configuration and would like to contribute to make it work.

SitanshuNanavati, 2016-07-21:
@msolters  @tiagodiogo 
Can you please share the 6lbr.conf /etc/network/interfaces details?  and other commands executed in order to see the web page with http://[bbbb:100].
Thank you.

adutze, 2016-07-25:
I guess the most important changes should be done related to this lines in the 6lbr-conf-native.h 
#undef NETSTACK_CONF_RDC
#define NETSTACK_CONF_RDC     border_router_rdc_driver

#undef NETSTACK_CONF_RADIO
#define NETSTACK_CONF_RADIO   nullradio_driver

You need to port the radio driver to linux as it will be running native, and then change the RDC to something like nullrdc or ContikiMAC. 
At the moment, border_router_rdc_driver is using the SLIP frames to communicate with the radio device. The RDC you choose on 6LBR should also be running on the sensors, and here nullrdc will not help you too much with the energy consumption on the sensors. ContikiMAC uses the rtimer when it is active, but the border router only needs it for strobing the TX messages, not actual duty cycling the radio (the radio should be always ON). I am not so shore if you need to implement the rtimer as a real time interrupt in the linux kernel for ContikiMAC to work. 
The radio RF233 uses just an interrupt pin, perhaps this should also be setup as real time interrupt on the linux side to get better timing.

SitanshuNanavati, 2016-07-27:
Thanks adutze for the reply.

I believe the 802154 driver is available and is running on RPi2.  
In the 6lbr code, I see there are spi.h/c files.  Can I make use of these?
Is there a way to not to package the data with SLIP-USB? and transmit the data as 15.4 packet?

msolters, 2016-07-27:
I had some luck using an AT86RF233 radio tied to the Raspberry Pi zero with
SPI. I was able to get a phy interface in jessie and even tcpdump the
802.15.4 traffic.  Was never able to send data IIRC.

msolters, 2016-07-27:
Now that I read the whole thread here it seems to me that what we really need to get cooking here is for

msolters, 2016-07-27:
Now that I read the whole thread here it seems to me that what we really need to get cooking is for @laurentderu to point us towards those references and tweaks necessary to use the wpan interface driver!

SitanshuNanavati, 2016-07-27:
I am going thru the code in samples given in 6lbr/examples/ipv6/.  I am not yet expert in this arena but was if this code could be modified to support AT86RF233 or connect the wpan0 interface on the other side tap/tun and provide 802.15.4 packets.  Please ignore if it doesn't make sense.  Its just a thought.

SitanshuNanavati, 2016-07-27:
BTW, the 6lbr/examples/ipv6/native-border-router needs following changes in code.  It doesn't run on RPi2 without them.

# vi border-router.c, line 161: remove ; (contiki 2.7 bug)
# vi slip-config.c, line 76: change -1 to 255 (RPi kernel??)
(source: http://thingschat.blogspot.in/2014/06/border-router-with-raspberry-pi-for-lln.html)

adutze, 2016-07-27:
Bridging wpan0 to 6LBR TAP interface doesn't make sens. It only make sens if you would like to use the kernel module for 6loWPAN network and 6LBR RPL root to manage a mesh network of devices (at least for me). 
I thought you want to use 6LBR as a 6loWPAN interface service. In this case, you need to port the rf233-arch.c  radio driver to 6LBR running native on linux in order to avoid SLIP packages. This implies the SPI communication and GPIO interrupt pin the radio is using.
For more advanced power management applications, like ContikiMAC or TISCH, you have some real time constraints. I don't know if linux kernel can meet them.

SitanshuNanavati, 2016-07-28:
Please ignore my thoughts for TAP.  I was just writing down my thoughts.

My requirement is to communicate between Ubuntu host to 6lowpan device.  RPi2 would act as a middleman between Ubuntu and 6lowpan device. Something like:
Ubuntu(host)<--eth0-->RPi2(with AT86RF233)<--Radio-->6lowpan Device.

adutze, 2016-07-28:
I would suggest to start changing this defines I mentioned previously
#undef NETSTACK_CONF_RDC
#define NETSTACK_CONF_RDC border_router_rdc_driver

#undef NETSTACK_CONF_RADIO
#define NETSTACK_CONF_RADIO nullradio_driver

into

#undef NETSTACK_CONF_RDC
#define NETSTACK_CONF_RDC nullrdc_driver

#undef NETSTACK_CONF_RADIO
#define NETSTACK_CONF_RADIO rf230_driver

And then port the halbb.c low level radio driver to RPI linux kernel. It should help a lot if there is already sourcecode for the native linux driver implementation.

Then start with RAW_ETHERNET basic configuration of the 6LBR config file.

SitanshuNanavati, 2016-07-28:
Thanks!
Let me get started with macro changes and ensure the code /6lbr/example/ipv6/native-border-router compiles.

fbalseirolago, 2017-04-07:
Hello,

Has this issue been resolved? I have the same set up (ATMEL's 802.15.4 radio transceiver  and a RPI2) and I was looking for some guidance in making 6lbr work with this module.

Thanks!

SitanshuNanavati, 2017-04-12:
I wonder why but it worked with specific version of Kernel version pasted below.

root@raspberrypi:/home/pi# uname -a
Linux raspberrypi 4.1.19+ #853 Wed Mar 9 18:04:48 GMT 2016 armv6l GNU/Linux

ffischer91, 2017-04-28:
Does anyone know if if these small changes work on an RPI3 with Kernel version:
Linux raspberrypi 4.7.10-v7+ #1 SMP Fri Apr 21 17:14:10 CEST 2017 armv7l GNU/Linux ?

fbalseirolago, 2017-05-29:
@SitanshuNanavati What do you mean when you say it worked? Did you manage to implement 6lbr with the AT86RF233 module?

jayarammatta, 2017-08-02:
Hi,
      I attached ATMEL's 802.15.4 radio transceiver directly to my Rpi 3 and created a wpan0 (lowpan0) interface for it (e.g. as seen in https://github.com/RIOT-OS/RIOT/wiki/How-to-install-6LoWPAN-Linux-Kernel-on-Raspberry-Pi#raspberrypi-with-openlabs-raspberry-pi-802154-radio).
    Then I able to perform the Local PING6, Global PING6, SSH, Wpan-tools applications.
   Now I want to test it with Intel celeron based Custom Board.please provide necessary suggestions.

florian-popup, 2017-10-31:
This thread is quite old but I have the same configuration and issue.

Just a through
If you can't connect the 6lbr to your interface, bring slip from your interface ;) 

I ll work on that and keep you up to date on this as it might help some people

leojrfs, 2018-03-10:
@florian-popup did it work for you in the end?

SitanshuNanavati, 2018-03-12:
No, it did not work for me.

florian-popup, 2018-03-12:
I ended up working with RIOT-OS and wakaama for the 6lowpan network and comissioning

Matheus-Garbelini, 2018-04-28:
Any new on this? I'm tring to use riot-os with lowpan0 interface nativelly on linux.