cetic/6lbr #116: 6lbr not updated with latest contiki commits

Netbulae, 2016-01-11:
*This repository is continuously updated with new commits in Contiki. The master branch will never be more than a few commits behind the master Contiki.*

The latest commit in 6lbr is now 2 months old while contiki has been updated a lot in this time.

Anything we can do to help?

arurke, 2016-02-21:
I have the same concerns as @Netbulae at the moment. 9 months since last release, 3 months since last commit. Is the project being discontinued? Anything we can do to help? @laurentderu

adutze, 2016-02-23:
Are there any pull requests from Contiki that 6LBR team is waiting for to be merged before updating? Is there some showstopper in the current Contiki source code that could brake 6lbr implementation?

laurentderu, 2016-02-25:
There has been a lot of activity in Contiki the last months (TCSH, RPL update, NDP tuning, as well as several architecture changes) and the previous changes (LLSEC, NAT64, mDNS, ...) are not yet fully stable in 6LBR.
At the same time we have big features still to be validated and integrated (OpenWRT support, CoRE Interface, 6LoWPAN-ND, ...)

So, in the next days or weeks, the develop branch will be frozen and released as version 1.4.0 (Based on Contiki 3.0) and a complete rebase will be done to synch with latest Contiki. And for the latter, help will be needed to validate 6LBR on all the supported platforms and configurations. We wanted 6LBR to be as flexible and as reliable as possible, but we are now paying the price as the validation time and maintenance explode.

adutze, 2016-02-26:
Thank you for the update, feels great to have this small roadmap in mind and thank you for the great work.

I guess there are several people testing 6LBR in their own projects around here, including me. I personally work mostly on the develop branch with the latest features and I am validating it almost continuously using different configurations. From this point of view, having new commits more often would release some workload from your side perhaps. Also cherry-picking some easy merges from latest Contiki, like Contiki-MAC fixes, sicslowpan.c fixes, etc could be added and tested by the community. It is a bit confusing how things are working around here, but there are people willing to help

closed 2017-06-15 by laurentderu