cetic/6lbr #118: How to access 6lbr web Interface.

juanaevv, 2016-01-23:
I set raspberry pi to 6lbr router mote.
And I tried to access 6lbr web Interface using Unix computer.
Unix computer has bbbb::101/64 and Raspberry pi has bbbb::100/64
Then, I checked computer can connect Raspberry pi using 'ping bbbb::100' command.
But, I can't access 6lbr web Interface using web browser in Unix computer.
I already set 'sysctl -w net.ipv6.conf.eth0.accept_ra=1' and 'sysctl -w net.ipv6.conf.eth0.accept_ra_rt_info_max_plen 64'

Then what should I do?
Are there more settings to run Web Interface in Raspberry pi?

serdarkose, 2016-01-29:
I also wonder the answer of the problem. I have the same issue as well.

shantanoo-desai, 2016-02-01:
try using `http://[bbbb::100]`. In most scenarios the square brackets need to be used when accessing the IPv6 addresses.

serdarkose, 2016-02-01:
I use brackets, but the web interface doesn't work. I see such logs on 6lbr.log
2016-01-01 14:32:05.521452: INFO: HTTP: Script : /
2016-01-01 14:32:05.756979: INFO: HTTP: Script : /
2016-01-01 16:41:39.686469: INFO: HTTP: Script : /
2016-01-01 16:41:39.794429: INFO: HTTP: Script : /
2016-01-01 16:41:44.810494: INFO: HTTP: Script : /
2016-01-01 16:42:14.827385: INFO: HTTP: Script : /
2016-01-01 16:43:14.846408: INFO: HTTP: Script : /
2016-01-01 16:48:14.925215: INFO: HTTP: Script : /
2016-01-01 16:58:15.35901: INFO: HTTP: Script : /
2016-01-01 17:14:26.25738: INFO: HTTP: Script : /

but nothing is shown on the web interface

shantanoo-desai, 2016-02-01:
does this help?
```
sudo ip -6 addr add bbbb::101/64 dev eth0
sudo route -A inet6 add aaaa::/64 gw bbbb::100
```
I tried some basic Example with 6lbr it is right [here](https://github.com/shantanoo-desai/coap-wsn) see if you can get any help from the Extensive README something might come up.

serdarkose, 2016-02-01:
thanks for the suggestion, but it didn't help

shantanoo-desai, 2016-02-01:
I think it would be a brute-force solution but try using a brand new version of 6lbr. If you have done that and still facing this problem then I think only @laurentderu  or @sdawans  can help you with it.

serdarkose, 2016-02-02:
I use  cetic-6lbr-1.3.3_armhf.deb, what version of 6lbr are you using?

shantanoo-desai, 2016-02-02:
I was using the same. Last try would be to reinstall it again ! 
Sorry couldn't be much help to you.

adutze, 2016-02-02:
Are you using Raw Ethernet like specified in the wiki here:
https://github.com/cetic/6lbr/wiki/6LBR-Interface-Configuration

?

laurentderu, 2016-02-29:
(I'm slowly purging the backlog)

the line : 

<pre>
2016-01-01 17:14:26.25738: INFO: HTTP: Script : /
</pre>

means that 6LBR got the http request and tries to send back the page. Either you have a routing issue, traffic flowing out of 6LBR goes to the wrong interface, or you have the multicast snooping issue (see /etc/6lbr/interfaces.dhcp.example)

closed 2016-04-18 by laurentderu