cetic/6lbr #120: How to use 6lbr on an IPv4 network

serdarkose, 2016-02-01:
Hi,

Here is my set-up.

- CC26xx as a sensor module: has aaaa::xx address

CC2538 as a slip-radio: has aaaa::xx address

RaspberrPi is the edge router: wlan0 interface has a public IPv4 addr,and use "router" mode of 6lbr
[image]

I am very confused how to access CC26xx nodes (aaaa::xx) from the server side?

Referring  processors.wiki.ti.com/.../Cc26xx_sw_examples , I installed Wrapsix on the RPI. What end address do I need to write when I send a UDP packet from the server/computer? Do I need to use Wrapsix or a different tunneling solution?

Second question: How can I use 6lbr under an Access Point that assigns private IPv4 addr (and doesn't support IPv6)? In that case, how can I access to the sensors?

laurentderu, 2016-02-29:
WrapSix allows only communication from the IPv6 network to a IPv4 network, not the other way. The NAT64 functionality integrated in the develop branch of 6LBR allows static configuration of NAT64 mapping and so allows IPv4 to IPv6 communication.

For you second question, you can also use the integrated NAT64 to access 6LBR and the nodes using IPv4.

Note: the documentation for NAT64 is yet lacking...

laurentderu, 2016-03-31:
NAT64 is now fully functional and the documentation is available here https://github.com/cetic/6lbr/wiki/NAT64

closed 2016-03-31 by laurentderu