cetic/6lbr #122: Starting 6lbr at boot on BeagleBone Black

emreerisgen, 2016-02-09:
Hi,

We've started a project which uses TI 1310 MCUs and BeagleBone Black to create a WSN. We are following this tutorial: 
[http://processors.wiki.ti.com/index.php/Cc26xx_sw_examples](url)

Everything is work as expected if we manually start the 6lbr as service, but when we try to add 6lbr to start at boot some things fail. 
We can see nodes on 6lbr UI screen. ([bbbb::100]) But we cant enter their UI screens by following their URL which is something like [aaaa:212:4b11:800:1234].
And nodes cant establish MQTT connection to the router.

What should we do to add 6lbr on startup on BeagleBone Black which runs Debian 7.8.
Any help will be much appreciated.

emreerisgen, 2016-02-09:
I forgot to add our config file.

MODE=ROUTER

RAW_ETH=1
BRIDGE=0
DEV_BRIDGE=br0
DEV_TAP=tap0
DEV_ETH=eth0
RAW_ETH_FCS=0

DEV_RADIO=/dev/ttyO1 
BAUDRATE=115200

LOG_LEVEL=3

laurentderu, 2016-02-29:
Could you check if the 6LBR process is properly started at boot ? It might be the init.d script is not properly installed, or the interface configuration is removed by an existing network manager

emreerisgen, 2016-03-31:
I thought we solved the problem by doing these:

- created a script which enables the UART then start the 6lbr service.
- run this script at boot via rc.local.

Everything is working as expected, MQTT, Coap, Web Server etc.

Today we plugged in a usb internet dongle on our modem and restarted the beaglebone and after reboot the problem occurred again. We thought its a coincidence so we plugged out usb dongle and restart the bb again and the problem was gone.

And the strange thing about this problem is this, if we restart the 6lbr service (which is not working correctly after boot) manually the problem is gone and everything is working correctly. 

Any idea? What are we doing wrong?

Our system is this:

- An ipv6 supporting modem
- A beaglebone black (runs 6lbr service and connected to modem via ethernet cable)
- A TI1310 (working as a slip radio and connected to BB by UART1)
- 2 TI1310 nodes
- A PC  (to send messages to nodes via MQTT)

laurentderu, 2016-03-31:
The driver of the USB controller of the BeagleBone is a bit buggy, so it may be that your slip-radio is no longer detected by the controller and so not usable by 6LBR (though I don't see at first sight the relationship between the USB controller and the UART).

Could you check if the UART device /dev/ttyO1 is still visible when you have you dongle plugged in, and if so if 6LBR is able to open it ?

In /usr/lib/6lbr there is a script to force a reset of the USB controller, maybe it will unlock your dongle.

emreerisgen, 2016-03-31:
Thank you for you answer Laurent. 

Maybe i couldnt tell the problem. Usb dongle is not connected to the BB, its connected to the modem. I use usb internet dongle to give internet to all devices which are connected to the modem.

I tried the USB reset script, no luck nothing changed, and yes /dev/ttyO1 is still visible.

By the way i can see the nodes at [bbbb::100] but i cant reach their web ui, or i cant send coap messages to them. But sometimes they disappear on [bbbb:100] and come back few seconds later. (This never happens if usb dongle is not connected to the modem)

It is a bit strange problem for me so i couldn't think anything what causes this problem.

closed 2016-05-11 by emreerisgen