cetic/6lbr #123: Getting (DUP!) for ping6 (2001:db8:1:ffff::8.8.8.8) from WSN radio

chennakeshavaSoC, 2016-02-11:
[image]
I am trying to create a test setup using CC2538DKs (as end nodes), CC2531 dongle, Beagle Bone Black , in IPv4 environment. I have tried different combinations to make it work but no success. Currently trying the setup as shown. Configured the 6LBR "Eth default router" as PC1's IP6 address (bbbb::xxxx), added iptables to pass the data between nat64 and eth0.
I am able to communicate between WSN nodes and PC's (intranet). But when I am trying to ping6 google (2001....8.8.8.8) from WSN node I am seeing the reply ripples in the wireshark running on PC.
And I am seeing the two IPv6 ping requests (one from "Unicast to PC", one from "Sent from US"). 

If I do same ping6 (2001:....8.8.8.8) from PC2, I don't have any issue, I am just getting one to one reply.

Please suggest me how to resolve the issue.
It would be great help if nat64 merged into BBB.

chennakeshavaSoC, 2016-02-19:
I have moved the tayga into BBB, and with RAW_ETH0=0, my WSN ping6 to google (2001:...8.8.8.8) is working fine.
Here I see two IPv6 ping requests in BBB, one is from tap0 (unicast), other one is sent by us resulting in two ping4 packets  (one from nat64, one from eth0) out of eth. I am not sure how to stop the other (sent by us) packet.

laurentderu, 2016-03-01:
Could you clarify what you mean by "two IPv6 ping requests, one is sent by us" ? Could it be that tap0 interface is not bridged with eth0 and the ping packet is routed between the two interfaces and Tayga listen to both interfaces ?

closed 2016-04-18 by laurentderu