cetic/6lbr #124: 6lbr didn't assign an ipv6 address for the network interface

rafaeldelucena, 2016-02-12:
Hi,

When I run the example the 6LBR application can't assign an ipv6 address with bbbb:: prefix for eth0 using a desktop Linux PC, so I cant access the webserver [bbbb::1000] on the browser.

My border router consists of a SmartRF06EB+CC13xxEM burned with slip-radio and instead of a BeagleBone, I am using an Linux desktop with Debian 8 (jessie), I have also change the RF channel to 26 and the frequency to 915Mhz for the slip-radio and the cc26xx-web-demo example.

My configurations and log from the applications are shown bellow:

6lbr configuration file (https://gist.github.com/anonymous/a61c10c689ce82feb792)
6lbr output (https://gist.github.com/rafaeldelucena/2c644d189400c34457e2)

I forgot something?

Thanks

akshat48, 2016-02-15:
Hi rafaeldelucena,

The issue that you are facing may be solved by looking at the following post:

https://github.com/cetic/6lbr/issues/109

I solved a similar issue on RPi2  by following the above

Thanks,

laurentderu, 2016-02-29:
The 'ERROR: SLIP: Unknown command' line usually means that the slip-radio does not support the set-channel command, so you're stuck with the default channel hardcoded in the firmware.

About your interface configuration, see : https://github.com/cetic/6lbr/wiki/6LBR-Connection

closed 2016-04-18 by laurentderu