cetic/6lbr #127: Problem connecting 6lbr through OpenVPN

Rasfumi, 2016-02-16:
My Setup is a Raspberry Pi (debian jessie) running:
* 6lbr in Router mode (static IPv6 bbbb::100) connected to LAN network through bridge interface (br0 - tap0)
* radvd running on the same Raspberry distributing IP addresses to other LAN hosts (br0 interface, IPv6 static bbbb::101)
* openvpn server using tun0 interface with IP 10.8.0.6 and IPv6 cccc::1, 

openVPN client gets cccc::1000

my Problem is that i cannot reach 6lbr through the tunnel
i can ping 6lbr from any host in the LAN
i can ping Raspberrys bbbb::101 through the tunnel
i can ping any other IPv6 host through the tunnel
i can ping any IPv4 host through the tunnel

it seems Raspberry doesnt know how to forward packets between tun0 and tap0 interfaces.
all ip forwarding is enabled

any ideas how i can get this to work?

laurentderu, 2016-02-29:
I think enabling forwarding is not enough, you have to have explicit routes towards your WSN subnet (I guess aaaa::/64).
Another possibility would be to add tun0 to the bridge br0, but that would require modifying the subnet configuration.

Rasfumi, 2016-03-08:
Thanks for your reply, but i am still not able to get it working.
i have tried to add tun0 to br0 with the command: sudo brctl addif br0 tun0
but i get: "can't add tun0 to bridge br0: Invalid argument "
i found out that its not possible to add tun interfaces to a bridge, only tap interfaces can be bridged...
adding a route also doesnt work, at least they way i have tried it: " sudo route add -6 aaaa::/64 gw bbbb::100"
this is my routing table:
Destination                    Next Hop                   Flag Met Ref Use If
::1/128                        ::                         U    256 0     0 lo
aaaa::/64                      bbbb::100                  UG   1   0     0 br0
bbbb::/64                      ::                         U    256 0     1 br0
cccc::/64                      ::                         U    256 0     0 tun0
fe80::/64                      ::                         U    256 0     0 br0
::/0                           ::                         !n   -1  1   117 lo
::1/128                        ::                         Un   0   1     2 lo
bbbb::/128                     ::                         Un   0   1     0 lo
bbbb::101/128                  ::                         Un   0   1     2 lo
cccc::/128                     ::                         Un   0   1     0 lo
cccc::1/128                    ::                         Un   0   1     0 lo
fe80::/128                     ::                         Un   0   1     0 lo
fe80::ba27:ebff:fea3:bbf9/128  ::                         Un   0   1     9 lo
ff00::/8                       ::                         U    256 1     0 br0
ff00::/8                       ::                         U    256 0     0 tun0
ff00::/8                       ::                         U    256 0     0 tap0
::/0                           ::                         !n   -1  1   117 lo

I will try running 6LBR on a second Raspberry to see if this works...

closed 2016-04-18 by laurentderu