cetic/6lbr #128: Setting Peer Router for Ti sensortag web-demo example

weijian19391, 2016-02-17:
Hi 

I followed the following tutorial: http://processors.wiki.ti.com/index.php/Cc26xx_sw_examples ,and am stucked at the part about setting a peer router. It states that we enter PC or IPv6 address of access point in the peer router section, may i know if the IPv6 address of Access point must be prefixed by bbbb? because i do not see such IPv6 address from my access point web page

weijian19391, 2016-02-20:
I've tried to add the ip address that is stated from my router's webserver, with a fe80 prefix, but from the network page of the 6lbr web server, under the neighbour section, it list my the mac address of the fe80 ip address as "0:0:0:0:0:0:0:0 INCOMPLETE". Is there something i did wrong?

weijian19391, 2016-03-08:
Hi! Does anyone has any idea how should i approach this? What Address of my Access point should i select?

adutze, 2016-03-09:
Hej.

I think the "peer router" in the tutorial is also the PC with adress:
bbbb::3e07:54ff:fe74:4885		
or
bbbb::d518:4dc:99bd:c48e

In case you want to connect the sensor-nodes to the internet, you
should have a "peer router" that has global access to IPv6 network.

At least this is my understanding. If you don't want that, just leave
it "::" default.

This is my understanding

Adrian

##SELECTION_END##
-- 

________________________
Embedded Systems Developer
UPWIS AB, Salagatan 18b, Uppsala
Ph: +46(0)184967090

weijian19391, 2016-03-10:
@adutze which means to fully utilize the capabilities of the 6lbr, i need to keep another PC constantly on to act as the peer router?

adutze, 2016-03-10:
The tutorial states that an "Access Point" is needed also:
Needed Hardware
This is the hardware you need to get started:
Access Point:
- The Access Point need to support IPv6. 
.......

I would say you could build a "peer router" in the BBB using Huricane
Electric tunnel broker (IPv6 in IPv4 tunnelling) to get global IPv6
connectivity. Things will mix up a bit. 
I suggest, if you do that, use 6LBR as an SMART-BRIDGE and you will
have all PCs, BBB ##SELECTION_END##and sensors in the same network
globally connected using the HE tunnel.
-- 

________________________
Embedded Systems Developer
UPWIS AB, Salagatan 18b, Uppsala
Ph: +46(0)184967090

closed 2016-04-18 by laurentderu