cetic/6lbr #129: On Ubuntu;I can't visit the [bbbb::100]website through another pc

limotkl, 2016-02-26:
Now I can run 6lbr as a RPL ROUTHER on unbuntu and I can see the [bbbb::100]webset on the pc which runs 6lbr but i can't see the  bbbb::100 through another pc(my laptop) , I think it's something about IPv6 setting, the two computers are in the same subnet. 
Can you tell me how can I visit the [bbbb::100] through another pc(with windows os)? Thank you very much!

laurentderu, 2016-02-29:
If the other pc has a bbbb::something address and you can't contact 6LBR, either your tap0 interface is not bridged with bR0 or you most probably have the multicast snooping issue, see  /etc/6lbr/interfaces.static.example

limotkl, 2016-02-29:
Problem solved. I failed to bridge tap0 with br0. After I amended  /etc/network/interfaces it works.
Does it work in the same way with /etc/6lbr/interfaces.static.example?

laurentderu, 2016-02-29:
Its the same way for static and dhcp, the only difference is the address configuration of the host. I will clarify the documentation.

limotkl, 2016-02-29:
Thank you very much !

closed 2016-04-18 by laurentderu