cetic/6lbr #131: TAP0 interface not being removed at watchdog failure

arurke, 2016-02-29:
Hi,
I spend too much time in an ifup script which cause the watchdog to fail. When the 6lbr resets, the tap0 interface is still present which causes 6lbr to go into an endless loop with "FATAL: TAP: tun_alloc() : Device or resource busy". Please see the log at level 10:
https://gist.github.com/arurke/e4387686daf7cb774dc0

I do nothing to tap0 in the ifup script except
ethtool --offload tap0 gso off
ethtool --offload tap0 gro off

My 6lbr.conf:
MODE=SMART-BRIDGE
RAW_ETH=0
BRIDGE=0
DEV_BRIDGE=br0
DEV_TAP=tap0
DEV_ETH=eth0
RAW_ETH_FCS=0
DEV_RADIO=/dev/ttyO1

Am I assuming correctly that it is the responsibility of 6lbr to remove this interface? I.e. I should not add handling for it in the ifdown scripts?

laurentderu, 2016-02-29:
A dynamically created tap interface is destroyed automatically when the endpoint is closed. Also, the existence of tap0 should not be a problem, 6LBR can open an existing tap interface (which can be useful if you're using the tap interface elsewhere and don't want the tap id to change). However, if another process still has that interface locked, 6LBR can not claim it and so it fails.

So maybe ethtool somehow claim (or keep a lock on) the tap interface, or the watchdog script failed to kill the previous 6lbr process, but I don't think so as the log shows the correct shutdown log.

arurke, 2016-03-01:
So you think this is specific for my setup, not a general issue? Afaik I do nothing with tap0 at all. The last line in my 60dev ifup script configures Jool to use it, but this line is never executed since the code before (infinite loop of trying to establish 3G modem connection) does not complete. In addition Jool is removed in the ifdown. I will try to replace the 3G code with a 120sec sleep to see if this can impact.

laurentderu, 2016-03-01:
The tap interface created by 6LBR is by default transient, so it should disappear when it gets killed. That means something is keeping it open. Could you check if there are no remaining 6lbr processes running ?

If it has been made persistent by one of the ifup scripts, you can try to force the deletion of the interface by using either 'tunctl -d tap0' or 'ip link delete tap0'

arurke, 2016-03-05:
Hi, I exchanged the 3G modem setup with a very long sleep (PS! the watchdog interval is a lot more than the stated 60 seconds, closer to 2-3 min). My 60dev looks like this:

```
#!/bin/bash
. $CETIC_6LBR_CONF
. $1/6lbr-functions
config_default
MODE_6LBR=$2
DEV=$3
OS=`uname`
ip -6 address add fd51:542a:930e:0000::2/64 dev tap0
sysctl -w net.ipv4.conf.all.forwarding=1
sysctl -w net.ipv6.conf.all.forwarding=1
ethtool --offload tap0 gso off
ethtool --offload tap0 gro off
sysctl -w net.ipv4.conf.all.log_martians=1
sleep 250`
```

I still get the same issue. Regarding your question on remaining processes, it might be this is the issue:

After watchdog timeout and in continuous restart of 6lbr:
```
admin@beaglebone:~$ ps -aux | grep 6lbr
root       465  0.1  0.4   2868  2104 ?        S    14:57   0:00 /bin/bash /usr/lib/6lbr/6lbr /etc/6lbr/6lbr.conf
root       523  0.0  0.1   1444   864 ?        S    14:57   0:00 sh -c /usr/lib/6lbr/6lbr-ifup tap tap0 2>&1
root       526  0.0  0.4   2848  2084 ?        S    14:57   0:00 /bin/bash /usr/lib/6lbr/6lbr-ifup tap tap0
root       565  0.0  0.4   2852  2088 ?        S    14:57   0:00 /bin/bash /etc/6lbr/ifup.d/60dev /usr/lib/6lbr tap tap0
```

After a normal start of 6lbr:
```
admin@beaglebone:~$ ps -aux | grep 6lbr
root       457  0.0  0.4   2868  2088 ?        S    14:51   0:00 /bin/bash /usr/lib/6lbr/6lbr /etc/6lbr/6lbr.conf
root       493  2.2  0.3   2164  1588 ?        S    14:51   0:05 /usr/lib/6lbr/bin/cetic_6lbr_smart_bridge -c /etc/6lbr/nvm.dat -s /dev/ttyO1 -t tap0 -R -B 460800 -U /usr/lib/6lbr/6lbr-ifup -D /usr/lib/6lbr/6lbr-ifdown -w /usr/lib/6lbr/www -L 10 -W /var/log/6lbr.timestamp -P 60 -C /var/log/6lbr.ip -m /usr/lib/6lbr/plugins
root       494  0.5  0.3   2868  1724 ?        S    14:51   0:01 /bin/bash /usr/lib/6lbr/6lbr /etc/6lbr/6lbr.conf

```

At first glance there is at least one extra process than in the normal startup. Was this what you were expecting?

Deleting the interface in ifdown will solve it, however:
Two side-notes: 
The ifdown script is called after the watchdog restart, but it is not called in the later continuous reboots:
```
2016-02-05 14:00:20.892753: FATAL: TAP: tun_alloc() : Device or resource busy
********************************************************************************
Sat Mar 5 14:00:21 UTC 2016 : Starting 6LBR
```

And the same happens when doing "service 6lbr restart": The log says it runs ifdown but I cannot see my echos from the actual script.
```
2016-02-05 14:29:58.293771: FATAL: TAP: signal 15
2016-02-05 14:29:58.297462: INFO: TAP: Running 6lbr-ifdown script '/usr/lib/6lbr/6lbr-ifdown'
********************************************************************************
Sat Mar 5 14:29:58 UTC 2016 : Starting 6LBR
```

arurke, 2017-03-13:
I have moved away from this setup so I will not investigate it further. Also I believe this was a very special use-case, and no one else has reported anything similar - so I'll just close the issue.

closed 2017-03-13 by arurke