cetic/6lbr #132: How to use 6lbr on coniki-os to send data over cloud.

tarungarg15, 2016-03-05:
Currently I am able to use 6lbr stack to send data only to intranet. But I want to send data over the internet. Could you please suggest how can I do ?.

laurentderu, 2016-03-11:
If you have IP6 connectivity, you can configure the peer router to your own router and the sensor can send their data directly.
If you don't have IPv6 connectivity, you can either use the integrated NAT64 functionality available on the develop branch, or use a service like gogoc or wrapsix

tarungarg15, 2016-03-12:
Can I get same working example from Contiki ?.

laurentderu, 2016-03-14:
6LBR is based on Contiki and should support any nodes running contiki, provided they use the same radio configuration (channel, pan id, rdc, ...)

tarungarg15, 2016-03-17:
Thanks a lot I will check

weijian19391, 2016-04-06:
@laurentderu Can i ask how do you configure the peer router to my own router?

laurentderu, 2016-04-06:
This is on the Network configuration page, see : https://github.com/cetic/6lbr/wiki/LAN

weijian19391, 2016-04-06:
@laurentderu Pardon me for being new to networking, it states that "Address of the Peer Router (or Default Router, or Gateway) of the LAN Network. ". May i know if this address needs to be one that is prefixed with bbbb?

laurentderu, 2016-04-06:
The peer router address is not the address of your 6LBR, it's the link-local IPv6 address of your Internet Gateway or LAN router.

Note that if you want to have internet connectivity you have to change aaaa:: and bbbb:: prefixes are they will be rejected by your ISP, they are for demo only.

weijian19391, 2016-04-06:
Oh i see. Are there any restrictions on what the prefixes should be?

laurentderu, 2016-04-06:
Actually you must use the IPv6 prefix your ISP or your network admin gave you, or your packet will not be routed.

weijian19391, 2016-04-06:
oh. Okie! thank you!

closed 2016-04-18 by laurentderu

dana555a, 2018-04-03:
hello there im trying to do the same hing but my network has Raspberry pi running 6lbr and cc1310 with slip radio as gateway. whith these settings how can i connect to cloud ?