cetic/6lbr #134: New NVM Configuration / 6LBR no longer starts

msolters, 2016-03-11:
Hi @laurentderu I see you have been changing some things about the NVM and configuration storage of 6LBR in develop.  I wonder if you can speak to/document the system you are implementing?

I tried simply renaming my old `6lbr.conf` to `nvm.conf`, but when I booted 6LBR it got stuck in a `SLIP: write: 2` loop where it seemed it could not get the MAC address from the slip radio.  Now, restarting/stopping/starting and rebooting the host has no effect -- it seems 6LBR no longer does *anything* when I try to run it.  The `/var/log/6lbr.log` just shows the last errors I get, `Fetching MAC address`.

laurentderu, 2016-03-11:
Hi,

I'm pushing the last features for the upcoming releases and things might get a bit unstable.
We're introducing a new configuration file for the native platform, this file is text based and will allow to have much more configuration parameters and it will be easier to change it compared to the binary nvm.

So now you will have the following configuration files :
* 6lbr.conf : contains the startup parameters of 6lbr, without it 6lbr can not start
* nvm.dat : contains the runtime configuration
* nvm.conf : the new runtime configuration

In my previous push, there was a bug, nvm.conf was mandatory, or it should be optional, I fixed that in today's push.

I hope to write the documentation about all the new features next week and make a release candidate Friday :)

hauseja3, 2016-03-18:
@laurentderu any update for releasing?

janeksz, 2016-03-21:
@laurentderu can you kindly advice briefly how to setup NAT64? 
I have

NAT 64 Enabled : True
NAT 64 DHCP Enabled : False
NAT 64 static port mapping Enabled : False
NAT 64 RFC-6052 prefix : False
NAT 64 address : 172.16.0.2
NAT 64 netmask : 255.255.255.0
NAT 64 gateway : 172.16.0.1
It doesn't work - I don't see [bbbb::100]. What should I put as 'NAT 64 address' and 'NAT64 gateway' ?

laurentderu, 2016-03-21:
@hauseja3 I hope tomorrow

@janeksz I fixed a last bug in NAT64 that prevented to access the BR when activated. The NAT64 address and gateway are that IPv4 address of the Border Router and the address of your IPv4 gateway/Router on your network.

closed 2016-04-18 by laurentderu

cuongdodinh, 2018-01-05:
Dear @msolters 
How can I create file 6lbr.conf in /etc/6lbr and modify it?

Thanks,