cetic/6lbr #135: CC2650 SensorTag cannot join 6lbr running on CC2538DK+enc28j60

chenek, 2016-03-14:
I pull the latest 6lbr and build it for CC2538DK+enc28j60. I can access to 6lbr web page but my CC2650 SensorTag cannot join 6lbr running on CC2538DK+enc28j60. I have see 6lbr uses channel 25 and pan ID: abcd. The followings are logs from my CC2538DK. Do I do anything wrong?

Contiki-6lbr-1.3.2-34-g3715b49
TI SmartRF06 + cc2538EM
 Net: sicslowpan
 MAC: CSMA
 RDC: nullrdc
 Channel: 25
 PAN-ID: abcd
Rime configured with address 00:12:4b:00:00:e5:00:3e
INFO: 6LBR: Starting 6LBR version 1.3.3 (Contiki-6lbr-1.3.2-34-g3715b49)
INFO: NVM: Reading 6LBR NVM
INFO: NVM: NVM Magic : 2009
INFO: NVM: NVM Version : 1
INFO: ECON: Eth MAC address : 06:00:00:e5:00:3e
INFO: ETH: ENC-28J60 Process started
INFO: ETH: ENC28J60 init
INFO: ENC: resetting chip
INFO: 6LBR: Tentative local IPv6 address fe80::212:4b00:e5:3e
INFO: 6LBR: Starting as 6LR
INFO: 6LBR: CETIC 6LBR Started
INFO: HTTP: Script : /
INFO: HTTP: Script : /favicon.ico
INFO: HTTP: Script : /
INFO: HTTP: Script : /sensors.html
INFO: HTTP: Script : /sensors.html

laurentderu, 2016-03-14:
You have built 6LBR in '6LR' mode, meaning it's not a Border Router but rather an intermediate RPL router : 
<pre>
INFO: 6LBR: Starting as 6LR
</pre>

You should clean your building directory and build a Router target, like cetic_6lbr_router.bin

chenek, 2016-03-14:
I build cetic_6lbr_router.bin and download it to my CC2538DK. However, CC2650 SensorTag still doesn't join. Do I miss anything else?

Contiki-6lbr-1.3.2-34-g3715b49
TI SmartRF06 + cc2538EM
 Net: sicslowpan
 MAC: CSMA
 RDC: nullrdc
 Channel: 25
 PAN-ID: abcd
Rime configured with address 00:12:4b:00:00:e5:00:3e
INFO: 6LBR: Starting 6LBR version 1.3.3 (Contiki-6lbr-1.3.2-34-g3715b49)
INFO: NVM: Reading 6LBR NVM
INFO: NVM: NVM Magic : ffff
INFO: NVM: NVM Version : ffff
ERROR: NVM: Invalid NVM magic number or unsupported NVM version, reseting it...
WARN: NVM: Migrate NVM version 0 towards 1
INFO: NVM: Flashing 6LBR NVM
INFO: ECON: Eth MAC address : 06:00:00:e5:00:3e
INFO: ETH: ENC-28J60 Process started
INFO: ETH: ENC28J60 init
INFO: ENC: resetting chip
INFO: 6LBR: Tentative local IPv6 address fe80::212:4b00:e5:3e
INFO: 6LBR: Tentative global IPv6 address (WSN) aaaa::212:4b00:e5:3e
INFO: 6LBR: Tentative global IPv6 address (ETH) bbbb::100
INFO: 6LBR: RA Daemon enabled
INFO: NVM: Flashing 6LBR NVM
INFO: 6LBR: Configured as DODAG Root
INFO: 6LBR: Starting as RPL ROUTER
INFO: 6LBR: CETIC 6LBR Started
INFO: ENC: resetting chip
INFO: HTTP: Script : /
INFO: HTTP: Script : /favicon.ico
INFO: HTTP: Script : /rpl.html
INFO: HTTP: Script : /config.html
INFO: HTTP: Script : /config?channel=25&wsn_pre=aaaa::&wsn_pre_len=64&wsn_auto=1&wsn_addr=aaaa::100&eth_pre=bbbb::&eth_pre_len=64&eth_auto=0&eth_addr=bbbb::100&eth_dft=fe80::290a:6044:3f2c:35b4&ra_daemon=1&ra_lifetime=0&ra_max_interval=600&ra_min_interval=200&ra_min_delay=3&ra_pio=1&ra_prefix_o=1&ra_prefix_a=1&ra_prefix_vtime=86400&ra_prefix_ptime=14400&ra_rio_en=1&ra_rio_lifetime=1800&rpl_instance_id=30&rpl_preference=0&rpl_dio_intdoubl=8&rpl_dio_intmin=12&rpl_dio_redundancy=10&rpl_min_hoprankinc=256&rpl_default_lifetime=30&rpl_lifetime_unit=256&rewrite=1
INFO: WEB: Got param: 'channel' = '25'
INFO: WEB: Got param: 'wsn_pre' = 'aaaa::'
INFO: WEB: Got param: 'wsn_pre_len' = '64'
INFO: WEB: Got param: 'wsn_auto' = '1'
INFO: WEB: Got param: 'wsn_addr' = 'aaaa::100'
INFO: WEB: Got param: 'eth_pre' = 'bbbb::'
INFO: WEB: Got param: 'eth_pre_len' = '64'
INFO: WEB: Got param: 'eth_auto' = '0'
INFO: WEB: Got param: 'eth_addr' = 'bbbb::100'
INFO: WEB: Got param: 'eth_dft' = 'fe80::290a:6044:3f2c:35b4'
INFO: WEB: Got param: 'ra_daemon' = '1'
INFO: WEB: Got param: 'ra_lifetime' = '0'
INFO: WEB: Got param: 'ra_max_interval' = '600'
INFO: WEB: Got param: 'ra_min_interval' = '200'
INFO: WEB: Got param: 'ra_min_delay' = '3'
INFO: WEB: Got param: 'ra_pio' = '1'
INFO: WEB: Got param: 'ra_prefix_o' = '1'
INFO: WEB: Got param: 'ra_prefix_a' = '1'
INFO: WEB: Got param: 'ra_prefix_vtime' = '86400'
INFO: WEB: Got param: 'ra_prefix_ptime' = '14400'
INFO: WEB: Got param: 'ra_rio_en' = '1'
INFO: WEB: Got param: 'ra_rio_lifetime' = '1800'
INFO: WEB: Got param: 'rpl_instance_id' = '30'
INFO: WEB: Got param: 'rpl_preference' = '0'
INFO: WEB: Got param: 'rpl_dio_intdoubl' = '8'
INFO: WEB: Got param: 'rpl_dio_intmin' = '12'
INFO: WEB: Got param: 'rpl_dio_redundancy' = '10'
INFO: WEB: Got param: 'rpl_min_hoprankinc' = '256'
INFO: WEB: Got param: 'rpl_default_lifetime' = '30'
INFO: WEB: Got param: 'rpl_lifetime_unit' = '256'
INFO: WEB: Got param: 'rewrite' = '1'
INFO: NVM: Flashing 6LBR NVM
ERROR: NVM: verify NVM failed, retry
INFO: NVM: Flashing 6LBR NVM
INFO: 6LBR: Rebooting...
Contiki-6lbr-1.3.2-34-g3715b49
TI SmartRF06 + cc2538EM
 Net: sicslowpan
 MAC: CSMA
 RDC: nullrdc
 Channel: 25
 PAN-ID: abcd
Rime configured with address 00:12:4b:00:00:e5:00:3e
INFO: 6LBR: Starting 6LBR version 1.3.3 (Contiki-6lbr-1.3.2-34-g3715b49)
INFO: NVM: Reading 6LBR NVM
INFO: NVM: NVM Magic : 2009
INFO: NVM: NVM Version : 1
INFO: ECON: Eth MAC address : 06:00:00:e5:00:3e
INFO: ETH: ENC-28J60 Process started
INFO: ETH: ENC28J60 init
INFO: ENC: resetting chip
INFO: 6LBR: Tentative local IPv6 address fe80::212:4b00:e5:3e
INFO: 6LBR: Tentative global IPv6 address (WSN) aaaa::212:4b00:e5:3e
INFO: 6LBR: Tentative global IPv6 address (ETH) bbbb::100
INFO: 6LBR: RA Daemon enabled
INFO: NVM: Flashing 6LBR NVM
INFO: 6LBR: Configured as DODAG Root
INFO: 6LBR: Starting as RPL ROUTER
INFO: 6LBR: CETIC 6LBR Started
ERROR: PF: Trying to send empty packet
INFO: HTTP: Script : /index.html
INFO: HTTP: Script : /config.html
INFO: HTTP: Script : /sensors.html
INFO: HTTP: Script : /
INFO: HTTP: Script : /sensors.html
INFO: HTTP: Script : /network.html

laurentderu, 2016-03-14:
The radio configuration of your sensor tag must match the 6LBR one. Check in the channel and PANID is  the same as the sensortag (The default value has changed in Contiki only recently, so you may still have TI panid in your tag)
Also, check the RDC configuration

chenek, 2016-03-14:
I am sure channel and pan ID are good since I have run 6lbr on BBB and this SensorTag can join it. What do you mean check the RDC configuration?

Thanks for your help!

laurentderu, 2016-03-16:
The RDC, Radio Duty Cycling, is a layer controlling the radio activity. You can use, nullRDC, which keep the radio continuously on, or an active RDC, like contikimac, which activate the radio only at specific time. Usually they are not compatible.
You are using nullRDC for the 6LBR but I think the Sensortag is using ContikiMAC

chenek, 2016-03-17:
Can you specify where I can change nullRDC to ContikiMAC?

chenek, 2016-03-17:
I try to change my CC2650STK to nullRDC and it works now. Thanks for your help!

closed 2016-03-17 by chenek

naivewong, 2017-06-17:
Can you tell me how to change it to nullRDC? I have the same problem. Thank you very much~

chenek, 2017-06-17:
@naivewong Add the following lines in project-conf.h of cc26xx-web-demo example to use nullRDC to test again.

#undef NETSTACK_CONF_RDC
#define NETSTACK_CONF_RDC nullrdc_driver