cetic/6lbr #136: er-rest-example build failing

phalloc, 2016-03-15:
after switching from contiki-os to 6lbr in order to attempt to use DTLS i tried to compile the basic er-rest-example for platform targets z1 and remote.

in branch develop the build is failing
in branch master the build is terminating with several warnings

laurentderu, 2016-03-16:
The er-coap library in 6lbr has been modified to support DTLS, but we did not report all the modifications to er-rest-example. I will have a look at it.
The example in 6lbr-demo should build without issues.

laurentderu, 2016-03-16:
There was problem with the API, I fixed it and pushed the fix, it should build properly now.

tiagodiogo, 2016-03-17:
hello @laurentderu 
thank you for your quick fix, it is working now :+1: 
however following your suggestion i started using the 6lbr-demos and i believe i have found some problems, i've opened issue #139 with some experimental results

thank you,
Tiago

closed 2016-03-17 by tiagodiogo