cetic/6lbr #138: How to make CC2650STK data send through 6lbr running on CC2538DK+enc28j60 to IBM MQTT quick start service?

chenek, 2016-03-17:
I am running 6lbr on CC2538DK+enc28j60 and make CC2650STK joins. I can see CC2650STK trying to send MQTT packets but it shows nothing on IBM MQTT quick start service. I think I need to do some NAT64 settings to make it work. I used to run 6lbr on BBB and enable wrapsix to make my CC2650STK can publish data on IBM MQTT quick start service. But, how can I do it with CC2538DK+enc28j60?

laurentderu, 2016-03-17:
The latest code in the develop branch has a NAT64 service integrated, you should activate it in the Configuration page and activate the RFC6102 prefix option.
It's not fully tested yet but I believe it's stable enough.

chenek, 2016-03-17:
Can you specify where Configuration page is and how to activate the RFC6102 prefix option?

chenek, 2016-03-18:
I get it now. You mean to build and run 6lbr develop version on CC2538DK. Then, access to Config page of 6lbr web page to activate the RFC6102 prefix option. I am building it now and will report back.

chenek, 2016-03-18:
After git the latest develop branch and build it, I cannot access to 6lbr web page on CC2538DK. The following is log which looks fine.

Contiki-develop-20150825-206-g9a51a6b
TI SmartRF06 + cc2538EM
 Net: sicslowpan
 MAC: CSMA
 RDC: nullrdc
 Channel: 25
 PAN-ID: abcd
Rime configured with address 00:12:4b:00:00:e5:00:3e
INFO: 6LBR: Starting 6LBR version 1.4.x (Contiki-develop-20150825-206-g9a51a6b)
INFO: NVM: Reading 6LBR NVM
INFO: NVM: NVM Magic : ffff
INFO: NVM: NVM Version : ffff
ERROR: NVM: Invalid NVM magic number or unsupported NVM version, reseting it...
WARN: NVM: Migrate NVM version 0 towards 1
WARN: NVM: Migrate NVM version 1 towards 5
INFO: NVM: Flashing 6LBR NVM
ERROR: NVM: verify NVM failed, retry
INFO: NVM: Flashing 6LBR NVM
INFO: ETH: ENC28J60 init
INFO: ETH: Eth MAC address : 06:00:00:e5:00:3e
INFO: ENC: resetting chip
INFO: ETH: ENC-28J60 Process started
INFO: LLSEC: Using 'nullsec' llsec driver
INFO: 6LBR: Security layer initialized
INFO: 6LBR: Tentative local IPv6 address fe80::212:4b00:e5:3e
INFO: 6LBR: Tentative global IPv6 address (WSN) aaaa::212:4b00:e5:3e
INFO: 6LBR: Tentative global IPv6 address (ETH) bbbb::100
INFO: 6LBR: RA Daemon enabled
INFO: 6LBR: Checking addresses duplication
INFO: NVM: Flashing 6LBR NVM
ERROR: NVM: verify NVM failed, retry
INFO: NVM: Flashing 6LBR NVM
INFO: 6LBR: Configured as DODAG Root aaaa::212:4b00:e5:3e
INFO: 6LBR: Starting as RPL ROUTER
INFO: NODECFG: Node Config init
INFO: HTTP: Starting webserver on port 80
INFO: UDPS: UDP server started
INFO: DNS: DNS proxy started
INFO: 6LBR: CETIC 6LBR Started

Any help?

laurentderu, 2016-03-21:
The log seems ok,check if your pc is properly configured to access the BR (address, route, ..) see https://github.com/cetic/6lbr/wiki/6LBR-Connection If that's correct, could you send me a pcap capture done while the 6lbr is starting up and when you try to ping it ?

chenek, 2016-03-22:
I can see the following config on my 6lbr web page. I only see RFC 6052 prefix and don't see RFC6102 prefix option. Can you specify where I can find it?

[image]

chenek, 2016-03-22:
After I enable everything in IP64, it works now.

[image]

closed 2016-03-22 by chenek