cetic/6lbr #139: 6lbr-demos build failing

tiagodiogo, 2016-03-17:
hello,
after trying the 6lbr-demos in order to have coap clients with LLSEC and DTLS i have found some problems compiling the examples for some target platforms with different arguments.
i have not made an extensive test but these are some of the results with errors for target platforms z1, remote and wismote with the command line arguments: none, WITH_TINYDTLS=1, WITH_DTLS_COAP=1 WITH_LLSEC=1 WITH_COAPSERVER=1

- target platform = z1
`cc1: error: unrecognized command line option ‘-mmemory-model=medium’`

- target platform = wismote
 WITH_TINYDTLS= 1 ` fatal error: dtls.h: No such file or directory`

thank you,
Tiago

tiagodiogo, 2016-03-17:
removing the command line option for platform Z1 from the Makefile lets the build continue,but i do now know the importance of this argument for proper runtime.

using WITH_COAPSERVER=1 there is an error with an LWM2M_DEVICE_POWER_VOLTAGE_RESOURCE_RES_ID that is being defined although the WITH_LW2M2 command line argument is not being used.
comenting lines 248 and 284 of examples/6lbr-demo/./platform/contiki/contiki-resources.c allows compilation termination but it is not a fix

laurentderu, 2016-03-18:
I pushed the fix (along with a few other fixes related to that part of the code)

tiagodiogo, 2016-03-18:
i pulled the latest commit but the errors still remain.

using platform z1 : `cc1: error: unrecognized command line option ‘-mmemory-model=medium`

using platform remote and argument WITH_DTLS_COAP=1 : `/apps/er-coap/er-coap-dtls.c:7:18: fatal error: dtls.h: No such file or directory`

using platform remote and argument WITH_TINYDTLS=1 : `/apps/tinydtls/dtls.c:20:22: fatal error: tinydtls.h: No such file or directory`

laurentderu, 2016-03-18:
It seems your msp430 compiler does not support 20 bits memory models, you can deactivate it by setting  MSP430_20BITS=1 on the command line.

About your second problem, you must set WITH_TINYDTLS in conjunction with WITH_DTLS_COAP or TinyDTLS itself is not build (The idea is that maybe in the future we will have another DTLS implementation)

The third problem is a documentation problem, TInyDTLS must be manually configured before used, in apps/tinydtls you have to run reconf.sh to create the necessary configuration files.

tiagodiogo, 2016-03-18:
thank you for you thorough explanation! it is working now
if you believe that it can be helpfull for future users i can update the COAP-DTLS example with the information you provided.
let me know

laurentderu, 2016-03-21:
Any kind of help is always gladly accepted, especially for the documentation :) it is severely lacking.

tiagodiogo, 2016-03-24:
just asked because you might wanted to keep the documentation with "team only" privileges.
i've added your explanations to the DTLS Coap Server example

closed 2016-03-24 by tiagodiogo