cetic/6lbr #140: Cannot access to web page of 6lbr running on CC2538DK+enc28j60 using latest develop commit 9a51a6b83e70050a22c5460b8d724b6cf9c46928

chenek, 2016-03-18:
I git latest develop commit 9a51a6b83e70050a22c5460b8d724b6cf9c46928 and build 6lbr for CC2538DK+enc28j60 but I cannot access to web page. It seems there is bug here.

laurentderu, 2016-03-18:
I just tested on Openmote platform (CC2538DK + ENC28J60) and the router mode works fine. could you check if your pc got an IPv6 address from the Border Router (by default a bbbb::/64).
If you don't have one, you need to configure your pc : https://github.com/cetic/6lbr/wiki/6LBR-Connection
If you have one, could you send me the log output of the Border Router and a pcap capture when you try to access the BR ?

chenek, 2016-03-18:
The IPv6 address of my PC
 IPv6 位址. . . . . . . . . . . . .: bbbb::290a:6044:3f2c:35b4
 臨時 IPv6 位址. . . . . . . . . . : bbbb::593:b362:b80f:af9
 連結-本機 IPv6 位址 . . . . . . . : fe80::290a:6044:3f2c:35b4%11
 IPv4 位址 . . . . . . . . . . . . : 10.15.1.96
 子網路遮罩 . . . . . . . . . . . .: 255.255.255.0
 預設閘道 . . . . . . . . . . . . .: 10.15.1.254

The log output from my CC2538DK are

Contiki-develop-20150825-206-g9a51a6b
TI SmartRF06 + cc2538EM
Net: sicslowpan
MAC: CSMA
RDC: nullrdc
Channel: 25
PAN-ID: abcd
Rime configured with address 00:12:4b:00:00:e5:00:3e
INFO: 6LBR: Starting 6LBR version 1.4.x (Contiki-develop-20150825-206-g9a51a6b)
INFO: NVM: Reading 6LBR NVM
INFO: NVM: NVM Magic : ffff
INFO: NVM: NVM Version : ffff
ERROR: NVM: Invalid NVM magic number or unsupported NVM version, reseting it...
WARN: NVM: Migrate NVM version 0 towards 1
WARN: NVM: Migrate NVM version 1 towards 5
INFO: NVM: Flashing 6LBR NVM
ERROR: NVM: verify NVM failed, retry
INFO: NVM: Flashing 6LBR NVM
INFO: ETH: ENC28J60 init
INFO: ETH: Eth MAC address : 06:00:00:e5:00:3e
INFO: ENC: resetting chip
INFO: ETH: ENC-28J60 Process started
INFO: LLSEC: Using 'nullsec' llsec driver
INFO: 6LBR: Security layer initialized
INFO: 6LBR: Tentative local IPv6 address fe80::212:4b00:e5:3e
INFO: 6LBR: Tentative global IPv6 address (WSN) aaaa::212:4b00:e5:3e
INFO: 6LBR: Tentative global IPv6 address (ETH) bbbb::100
INFO: 6LBR: RA Daemon enabled
INFO: 6LBR: Checking addresses duplication
INFO: NVM: Flashing 6LBR NVM
ERROR: NVM: verify NVM failed, retry
INFO: NVM: Flashing 6LBR NVM
INFO: 6LBR: Configured as DODAG Root aaaa::212:4b00:e5:3e
INFO: 6LBR: Starting as RPL ROUTER
INFO: NODECFG: Node Config init
INFO: HTTP: Starting webserver on port 80
INFO: UDPS: UDP server started
INFO: DNS: DNS proxy started
INFO: 6LBR: CETIC 6LBR Started

laurentderu, 2016-03-18:
The log and ip addresses looks ok, what is the error reported by your browser when you try to connect to http://[bbbb::100]/ ? Could you make a capture when you try to access it and/or ping it ?

chenek, 2016-03-18:
I can ping it (Sorry about some of strings are in Chinese).
Ping bbbb::100 (使用 32 位元組的資料):
回覆自 bbbb::100: 時間=9ms
回覆自 bbbb::100: 時間=5ms
回覆自 bbbb::100: 時間=5ms
回覆自 bbbb::100: 時間=5ms

bbbb::100 的 Ping 統計資料:
    封包: 已傳送 = 4，已收到 = 4, 已遺失 = 0 (0% 遺失)，
大約的來回時間 (毫秒):
    最小值 = 5ms，最大值 = 9ms，平均 = 6ms

But the web page give  connection time out error.
ERR_CONNECTION_TIMED_OUT

laurentderu, 2016-03-18:
It's really strange, could you capture the traffic using wireshark or tcpdump while you try to access it and send me the file ?

chenek, 2016-03-18:
By the way, I can ping [bbbb::100] and [aaaa::212:4b00:e5:3e] successfully. But when I ping to [fe80::212:4b00:e5:3e], it returns timeout.

chenek, 2016-03-18:
I attache the capture traffic for you.
[CC2538DK.zip](https://github.com/cetic/6lbr/files/179561/CC2538DK.zip)

Thanks for your help.

laurentderu, 2016-03-21:
The issue is quite strange, the pcap file shows that the Border Router receives the TCP packets correctly and send back the reply, but it seems that reply is not received by your PC.
Which browser are you using ? Do you have a local firewall running ?

chenek, 2016-03-22:
I try to access [bbbb::100] from both Firefox and Chrome. Both of them fail. But if I don't use 6lbr develop branch, it works fine. So, I don't think this is about firewall.

chenek, 2016-03-22:
After I use another notebook to access [bbbb::100] web page, it works. I think there might be something wrong with my desktop. Anyway, Thanks for your help!

laurentderu, 2016-04-27:
This problem actually occurs on Windows (at least 7) because the Windows stack reject IP packet with extra padding. Linux and Mac OS ignore these padding bytes.

I pushed a fix on the develop branch that should fix the problem.

closed 2017-12-14 by laurentderu