cetic/6lbr #141: 6lbr can't route to sensor network

serdarkose, 2016-03-18:
I have been using 6lbr for couple months without any problem, but somehow it stopped routing udp packets to the sensor network (aaaa::/64) . I am able to receive UDP packets from the contiki nodes.

I monitor the network through Wireshark, but I don't see any packet traffic when I send a UDP packet or try to ping any aaaa:: node.

I would appreciate your comments.

Here are some debug outputs:

`root@raspberrypi:/# /usr/lib/6lbr/bin/nvm_tool --print /etc/6lbr/nvm.dat
Reading nvm file '/etc/6lbr/nvm.dat'
Channel : 25

WSN network prefix : aaaa::
WSN network prefix length : 64
WSN IP address : aaaa::100
WSN accept RA : True
WSN IP address autoconf : True

Eth network prefix : bbbb::
Eth network prefix length : 64
Eth IP address : bbbb::100
Eth default router : ::
Eth IP address autoconf : False

Local address rewrite : True
Smart Multi BR : False

RA daemon : True
RA router lifetime : 0
RA maximum interval : 600
RA minimum interval : 200
RA minimum delay : 3
RA PIO enabled : True
RA prefix valid lifetime : 86400
RA prefix preferred lifetime : 14400
RA RIO enabled : True
RA RIO lifetime : 1800

RPL instance ID : 30
RPL Preference : 0
RPL version ID :  : 251
RPL DIO interval doubling : 8
RPL DIO minimum interval : 12
RPL DIO redundancy : 10
RPL default lifetime : 30
RPL minimum rank increment : 256
RPL lifetime unit : 256

Webserver configuration page disabled : False
`

```
root@raspberrypi:/# cat /etc/6lbr/6lbr.conf
MODE=ROUTER
#MODE=SMART-BRIDGE
#MODE=RPL-RELAY
#MODE=FULL-TRANSPARENT-BRIDGE
#MODE=NDP-ROUTER
#MODE=6LR
#MODE=RPL-ROOT

RAW_ETH=1
BRIDGE=0
DEV_BRIDGE=br0
DEV_TAP=tap0
DEV_ETH=wlan0
#RAW_ETH_FCS=1
RAW_ETH_FCS=0

DEV_RADIO=/dev/ttyACM0 #2.4GHz USB Dongle will enumerate like ACM0
#DEV_RADIO=/dev/ttyUSB1 #SmartRF06EB+CC13xxEM will enumerate like USB1
BAUDRATE=115200

LOG_LEVEL=3
root@raspberry
```

`root@raspberrypi:/var/log# /etc/init.d/6lbr start
Starting 6lbr (via systemctl): 6lbr.service.
root@raspberrypi:/var/log# cat 6lbr.err
********************************************************************************
Fri 18 Mar 20:19:43 CET 2016 : Starting 6LBR
root@raspberrypi:/var/log# cat 6lbr.log
********************************************************************************
Fri 18 Mar 20:19:43 CET 2016 : Starting 6LBR
/usr/lib/6lbr/bin/cetic_6lbr_router -c /etc/6lbr/nvm.dat -s /dev/ttyACM0 -t wlan0 -r -B 115200 -U /usr/lib/6lbr/6lbr-ifup -D /usr/lib/6lbr/6lbr-ifdown -w /usr/lib/6lbr/www -L 3  -W /var/log/6lbr.timestamp -P 60 -C /var/log/6lbr.ip  -m /usr/lib/6lbr/plugins  
Contiki-6lbr-1.3.3 started with IPV6, RPL
Rime started with address 1.2.3.4.5.6.7.8
MAC CSMA RDC br-rdc NETWORK sicslowpan
Log level : 30
Log services : ffffffff
2016-02-18 20:19:43.84130: INFO: ETH: 6LBR watchdog started (interval: 60)
2016-02-18 20:19:43.84896: INFO: 6LBR: Starting 6LBR version 1.3.3 (Contiki-6lbr-1.3.3)
2016-02-18 20:19:43.84921: INFO: NVM: Opening nvm file '/etc/6lbr/nvm.dat'
2016-02-18 20:19:43.84984: INFO: NVM: NVM Magic : 2009
2016-02-18 20:19:43.84998: INFO: NVM: NVM Version : 1
2016-02-18 20:19:43.85013: INFO: SCMD: Started br-cmd process
2016-02-18 20:19:43.85038: INFO: NODECFG: No node_config.conf file specified
2016-02-18 20:19:43.85059: INFO: ETH: RAW/TAP init
2016-02-18 20:19:43.85569: INFO: SLIP: SLIP started on /dev/ttyACM0
2016-02-18 20:19:43.134190: INFO: TAP: opened device /dev/wlan0
2016-02-18 20:19:43.134404: INFO: TAP: Running 6lbr-ifup script '/usr/lib/6lbr/6lbr-ifup'
wlan0     Link encap:Ethernet  HWaddr 84:c9:b2:d6:34:a0  
          inet addr:85.235.3.135  Bcast:85.235.3.255  Mask:255.255.255.0
          inet6 addr: bbbb::100/64 Scope:Global
          inet6 addr: fe80::68e8:993:22c6:b4ca/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:168007 errors:0 dropped:0 overruns:0 frame:0
          TX packets:1255 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:113725940 (108.4 MiB)  TX bytes:192324 (187.8 KiB)

2016-02-18 20:19:43.278863: INFO: TAP: Eth MAC address : 84:c9:b2:d6:34:a0
2016-02-18 20:19:43.279014: INFO: BR-RDC: Reset SLIP Radio
2016-02-18 20:19:43.279071: INFO: BR-RDC: Fetching MAC address
2016-02-18 20:19:43.310594: ERROR: SLIP: Unknown command
2016-02-18 20:19:43.341459: INFO: BR-RDC: Got MAC: 00:12:4b:00:04:0e:ff:3a
2016-02-18 20:19:44.279453: INFO: 6LBR: Tentative local IPv6 address fe80::212:4b00:40e:ff3a
2016-02-18 20:19:44.279823: INFO: 6LBR: Tentative global IPv6 address (WSN) aaaa::212:4b00:40e:ff3a
2016-02-18 20:19:44.280039: INFO: 6LBR: Tentative global IPv6 address (ETH) bbbb::100
2016-02-18 20:19:44.280168: INFO: 6LBR: RA Daemon enabled
2016-02-18 20:19:44.280243: INFO: NVM: Opening nvm file '/etc/6lbr/nvm.dat'
2016-02-18 20:19:44.280878: INFO: 6LBR: Configured as DODAG Root
2016-02-18 20:19:44.280976: INFO: 6LBR: Starting as RPL ROUTER
2016-02-18 20:19:44.281748: INFO: UDPS: UDP server started
2016-02-18 20:19:44.281980: INFO: 6LBR: CETIC 6LBR Started
2016-02-18 20:19:44.282126: ERROR: SLIP: Unknown command
root@raspberrypi:/var/log# cat 6lbr.ip 
bbbb::100
root@raspberrypi:/var/log# cat 6lbr.err
********************************************************************************
Fri 18 Mar 20:19:43 CET 2016 : Starting 6LBR`

`root@raspberrypi:/# uname -a
Linux raspberrypi 4.1.19-v7+ #853 SMP Wed Mar 9 18:09:16 GMT 2016 armv7l GNU/Linux
root@raspberrypi:/# ifconfig 
eth0      Link encap:Ethernet  HWaddr b8:27:eb:46:0b:5e  
          inet6 addr: fe80::6cc4:1239:89c9:cc62/64 Scope:Link
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

tun0      Link encap:UNSPEC  HWaddr 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  
          inet addr:10.8.0.6  P-t-P:10.8.0.5  Mask:255.255.255.255
          UP POINTOPOINT RUNNING NOARP MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:272 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:100 
          RX bytes:0 (0.0 B)  TX bytes:22160 (21.6 KiB)

wlan0     Link encap:Ethernet  HWaddr 84:c9:b2:d6:34:a0  
          inet addr: x.x.x.x Bcast:x.x.x.255  Mask:255.255.255.0
          inet6 addr: bbbb::100/64 Scope:Global
          inet6 addr: fe80::68e8:993:22c6:b4ca/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:174992 errors:0 dropped:0 overruns:0 frame:0
          TX packets:1311 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:115017330 (109.6 MiB)  TX bytes:200299 (195.6 KiB)

root@raspberrypi:/# ping6 aaaa::212:4b00:40e:ff3a
connect: Network is unreachable
`

laurentderu, 2016-03-21:
Hello,

The error :
<pre>
root@raspberrypi:/# ping6 aaaa::212:4b00:40e:ff3a
connect: Network is unreachable
</pre>

indicates that the RPI does not have a route configured for aaaa::/64. This is probably because the RPI reject the route auto configuration from 6LBR. You could configure that using the command found here : https://github.com/cetic/6lbr/wiki/6LBR-Connection

rohitsivakumar, 2016-04-26:
I faced similar problem as mentioned by Laurentderu and I followed the link suggested in the above thread. Now I don't receive the error but nor does the ping packet get through. So there is 100% ping loss.

```
pi@bananapi /etc/6lbr $ ping6 aaaa::212:4b00:7b5:9104
PING aaaa::212:4b00:7b5:9104(aaaa::212:4b00:7b5:9104) 56 data bytes
--- aaaa::212:4b00:7b5:9104 ping statistics ---
33 packets transmitted, 0 received, 100% packet loss, time 32007ms
```

Here is my 6lbr.conf
```
MODE=ROUTER
#MODE=SMART-BRIDGE
#MODE=RPL-RELAY
#MODE=FULL-TRANSPARENT-BRIDGE
#MODE=NDP-ROUTER
#MODE=6LR
#MODE=RPL-ROOT

RAW_ETH=0
BRIDGE=1
CREATE_BRIDGE=1
DEV_BRIDGE=br0
DEV_TAP=tap0
DEV_ETH=eth0
RAW_ETH_FCS=0

DEV_RADIO=/dev/ttyACM0
BAUDRATE=115200

LOG_LEVEL=3 #INFO and above only
```

My network interfaces configuration:
```
auto lo

iface lo inet loopback

iface eth0 inet static
address 0.0.0.0

auto br0
iface br0 inet dhcp
        bridge_ports eth0
        bridge_stp off
        up echo 0 > /sys/devices/virtual/net/br0/bridge/multicast_snooping
        post-up ip link set br0 address 'ip link show eth0 | grep ether | awk '{print $2}''

allow-hotplug wlan0
iface wlan0 inet dhcp
wpa-conf /etc/wpa_supplicant/wpa_supplicant.conf

iface default inet dhcp
```
```

pi@bananapi /var/log $ brctl show
bridge name     bridge id               STP enabled     interfaces
br0             8000.020a0b0c0d0e       no              eth0
                                                        tap0
```

but after doing manual entry of config as suggested in https://github.com/cetic/6lbr/wiki/6LBR-Connection, I am unable to ping bbbb::100 (earlier it was pinging but wsn sensors were not getting pinged)
```
pi@bananapi /var/log $ ping bbbb::100
ping: unknown host bbbb::100

pi@bananapi /var/log $ cat 6lbr.ip
bbbb::100
```

What am I missing?

laurentderu, 2016-04-26:
My feeling looking at all your traces is that you have a network manager process overriding the configuration set by /etc/network/interfaces and 6LBR. Could you check if you have an active wild, NetworkManager or ConnMan ?

Also you you give the output of ifconfig and netstat -nr6, and the status of accept_ra of the bridge interface ?

rohitsivakumar, 2016-04-27:
These are the status of my currently running services (jessie - bananapi).
There is gnome running; I don't have ConnMan or active Wild. Don't think jessie has a networkManager but there is something (not known by what name in Jessie variant - but seems there must be some network manager utility). Hence sharing the services currently running on my board running 6LBR. Can this help ?
```
 service --status-all
 [ + ]  6lbr
 [ ? ]  alsa-utils
 [ - ]  bootlogs
 [ ? ]  bootmisc.sh
 [ ? ]  checkfs.sh
 [ ? ]  checkroot-bootclean.sh
 [ - ]  checkroot.sh
 [ - ]  console-setup
 [ + ]  cron
 [ + ]  dbus
 [ ? ]  dphys-swapfile
 [ ? ]  fake-hwclock
 [ - ]  hostname.sh
 [ ? ]  hwclock.sh
 [ + ]  ifplugd
 [ - ]  kbd
 [ - ]  keyboard-setup
 [ ? ]  killprocs
 [ ? ]  kmod
 [ + ]  lightdm
 [ - ]  motd
 [ ? ]  mountall-bootclean.sh
 [ ? ]  mountall.sh
 [ ? ]  mountdevsubfs.sh
 [ ? ]  mountkernfs.sh
 [ ? ]  mountnfs-bootclean.sh
 [ ? ]  mountnfs.sh
 [ ? ]  mtab.sh
 [ ? ]  networking
 [ - ]  nfs-common
 [ + ]  ntp
 [ ? ]  plymouth
 [ ? ]  plymouth-log
 [ - ]  procps
 [ ? ]  rc.local
 [ - ]  rmnologin
 [ - ]  rpcbind
 [ - ]  rsync
 [ + ]  rsyslog
 [ ? ]  sendsigs
 [ + ]  ssh
 [ - ]  sudo
 [ + ]  triggerhappy
 [ + ]  udev
 [ ? ]  udev-mtab
 [ ? ]  umountfs
 [ ? ]  umountnfs.sh
 [ ? ]  umountroot
 [ - ]  urandom
 [ - ]  x11-common
```

**output of ifconfig:**

```
pi@bananapi /etc $ ifconfig
br0       Link encap:Ethernet  HWaddr 02:0a:0b:0c:0d:0e
          inet addr:192.168.1.185  Bcast:192.168.1.255  Mask:255.255.255.0
          inet6 addr: fe80::4e:6ff:fec2:d995/64 Scope:Link
          inet6 addr: bbbb::a:bff:fe0c:d0e/64 Scope:Global
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:175902 errors:0 dropped:0 overruns:0 frame:0
          TX packets:12064 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:38528619 (36.7 MiB)  TX bytes:1071044 (1.0 MiB)

eth0      Link encap:Ethernet  HWaddr 02:4e:06:c2:d9:95
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:257061 errors:0 dropped:1958 overruns:0 frame:0
          TX packets:86420 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:60892149 (58.0 MiB)  TX bytes:17699007 (16.8 MiB)
          Interrupt:117 Base address:0xc000

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:16436  Metric:1
          RX packets:354 errors:0 dropped:0 overruns:0 frame:0
          TX packets:354 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:130890 (127.8 KiB)  TX bytes:130890 (127.8 KiB)

p2p0      Link encap:Ethernet  HWaddr 9a:3b:16:c7:d4:ba
          inet6 addr: fe80::983b:16ff:fec7:d4ba/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

tap0      Link encap:Ethernet  HWaddr 02:0a:0b:0c:0d:0e
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:1050 errors:0 dropped:0 overruns:0 frame:0
          TX packets:156459 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:500
          RX bytes:225556 (220.2 KiB)  TX bytes:19528287 (18.6 MiB)

tunl0     Link encap:IPIP Tunnel  HWaddr
          UP RUNNING NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

wlan0     Link encap:Ethernet  HWaddr 98:3b:16:c7:d4:ba
          inet addr:192.168.1.161  Bcast:192.168.1.255  Mask:255.255.255.0
          inet6 addr: bbbb::9a3b:16ff:fec7:d4ba/64 Scope:Global
          inet6 addr: fe80::9a3b:16ff:fec7:d4ba/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:87254 errors:0 dropped:0 overruns:0 frame:0
          TX packets:56947 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:8899274 (8.4 MiB)  TX bytes:10613109 (10.1 MiB)
```

**output of netstat -nr6:**
```
pi@bananapi /etc $ netstat -nr6
Kernel IPv6 routing table
Destination                    Next Hop                   Flag Met Ref Use If
bbbb::/64                      ::                         UAe  256 0     0 wlan0
bbbb::/64                      ::                         UAe  256 0     0 br0
fe80::/64                      ::                         U    256 0     0 wlan0
fe80::/64                      ::                         U    256 0     0 p2p0
fe80::/64                      ::                         U    256 0     0 br0
::/0                           ::                         !n   -1  1 16349 lo
::1/128                        ::                         Un   0   1    15 lo
bbbb::a:bff:fe0c:d0e/128       ::                         Un   0   1     0 lo
bbbb::9a3b:16ff:fec7:d4ba/128  ::                         Un   0   1   839 lo
fe80::4e:6ff:fec2:d995/128     ::                         Un   0   1     0 lo
fe80::983b:16ff:fec7:d4ba/128  ::                         Un   0   1     0 lo
fe80::9a3b:16ff:fec7:d4ba/128  ::                         Un   0   1     0 lo
ff00::/8                       ::                         U    256 0     0 wlan0
ff00::/8                       ::                         U    256 0     0 p2p0
ff00::/8                       ::                         U    256 0     0 br0
::/0                           ::                         !n   -1  1 16349 lo
pi@bananapi /etc $
```

rohitsivakumar, 2016-05-02:
Hi, did you get a chance to look at the logs. I am kind of clueless and stuck at this point in going further in setting up  contiki network which allows me to ping the WSN sensors from ipv4/ipv6 network.

laurentderu, 2016-05-03:
Your netstat output is a bit weird, it seems that both br0 and wlan0 are configured with a bbbb::/64 address and worse the route towards aaaa:/64 goes through wlan0 instead of br0

Maybe you have another RADVD running on your network using the same configuration as 6LBR and it ends up with both interfaces configured with the same prefix or somehow the RA messages from 6LBR "leaked" onto wlan0

you should try and remove the configuration on wlan0 and the aaaa::/64 route and recreate it manually to see if it works better.

rohitsivakumar, 2016-05-03:
Hi Laurent, Could you please guide me a bit here as I am completely lost.
Can you help me with commands to manually remove (from where - linux host?
from the bananaPI hw that is running 6lbr +contiki?)

rohitsivakumar, 2016-05-03:
I removed the wlan0 interface to check this out but this time the ping packets are lost:
I added the route from aaaa:: to bbbb:: on the bananaPI hardware on which I am running from 6LBR. So from this same hardware when I try to ping the wsn node with address aaaa::212:4b00:7b5:9104 the ping packets are lost.

```
pi@bananapi ~ $ sudo route -A inet6 add aaaa::/64 gw bbbb::100
pi@bananapi ~ $ netstat -nr6
Kernel IPv6 routing table
Destination                    Next Hop                   Flag Met Ref Use If
aaaa::/64                      bbbb::100                  UG   1   0     0 br0
bbbb::/64                      ::                         UAe  256 0     1 br0
fe80::/64                      ::                         U    256 0     0 br0
::/0                           ::                         !n   -1  1    17 lo
::1/128                        ::                         Un   0   1     9 lo
bbbb::a:bff:fe0c:d0e/128       ::                         Un   0   1     4 lo
fe80::649f:ff:fefc:769d/128    ::                         Un   0   1     0 lo
ff00::/8                       ::                         U    256 0     0 br0
::/0                           ::                         !n   -1  1    17 lo
pi@bananapi ~ $ ping6 aaaa::212:4b00:7b5:9104
PING aaaa::212:4b00:7b5:9104(aaaa::212:4b00:7b5:9104) 56 data bytes
```

rohitsivakumar, 2016-05-03:
**Status shown on the 6lbr webpage under the section "NETWORK"**

```
Addresses

fe80::212:4b00:768:ce5d P A
bbbb::100 P M
aaaa::212:4b00:768:ce5d P A
fe80::7ff:ff68:ce5d P A
Prefixes

bbbb:: A
fe80:: 
Neighbors

[del] fe80::9a3b:16ff:fec7:d4ba 98:3b:16:ff:ff:c7:d4:ba STALE
[del] fe80::212:4b00:768:ce5d 0:0:0:0:0:0:0:0 INCOMPLETE
[del] fe80::212:4b00:7b5:9104 0:12:4b:0:7:b5:91:4 REACHABLE
[del] fe80::546:8fa8:e2d3:12f1 14:2d:27:ff:ff:e9:7e:e7 STALE
[del] fe80::a6d1:8cff:feed:d650 a4:d1:8c:ff:ff:ed:d6:50 STALE
[del] fe80::1a59:36ff:fe09:d823 18:59:36:ff:ff:9:d8:23 STALE
[del] fe80::649f:ff:fefc:769d 2:a:b:ff:ff:c:d:e STALE
[del] bbbb::e8ff:972:28f6:f840 0:26:c7:ff:ff:65:6b:8a REACHABLE
[del] fe80::16f6:5aff:fe66:1689 14:f6:5a:ff:ff:66:16:89 STALE
[del] fe80::4288:5ff:feb0:fb9b 40:88:5:ff:ff:b0:fb:9b STALE
[del] fe80::1191:69e2:18a0:fb3c 78:dd:8:ff:ff:f0:fc:1c STALE
[del] fe80::92c1:15ff:fe12:fe77 90:c1:15:ff:ff:12:fe:77 STALE
[del] fe80::c2ee:fbff:fe54:e610 c0:ee:fb:ff:ff:54:e6:10 STALE
[del] fe80::208:22ff:fee6:45fc 0:8:22:ff:ff:e6:45:fc STALE
[del] fe80::ba57:d8ff:fe00:fa22 b8:57:d8:ff:ff:0:fa:22 STALE
[del] fe80::f68b:32ff:feca:5b77 f4:8b:32:ff:ff:ca:5b:77 STALE
[del] fe80::1630:c6ff:fe1a:1fdc 14:30:c6:ff:ff:1a:1f:dc STALE
[del] fe80::e692:fbff:fee8:e153 e4:92:fb:ff:ff:e8:e1:53 STALE
[del] fe80::1eb7:2cff:fe5a:7ab1 1c:b7:2c:ff:ff:5a:7a:b1 STALE
[del] fe80::f68b:32ff:feca:5cc7 f4:8b:32:ff:ff:ca:5c:c7 STALE
[del] fe80::c69a:2ff:fe7c:1eb3 c4:9a:2:ff:ff:7c:1e:b3 STALE
Routes

[del] aaaa::212:4b00:7b5:9104/128 via fe80::212:4b00:7b5:9104 7664 s
Default Routers

Route info

aaaa::/64 (0) 1800s
```

rohitsivakumar, 2016-05-10:
I have been trying to get the demo as mentioned on TI's WIKI to experience contiki but in vain and I am stuck on this 6LBR thing. Neither can I ping the WSN sensors from my windows host nor am I able to open the browser. Can you please help what is going messy here!

I changed my mode from router to smart bridge with the following  confs.
```
**6LBR:**
MODE=SMART-BRIDGE
RAW_ETH=1
BRIDGE=0 # was 1
DEV_BRIDGE=br0
DEV_TAP=tap0
DEV_ETH0=eth0
RAW_ETH_FCS=0
DEV_RADIO=/dev/ttyACM0
BAUDRATE=115200
LOG_LEVEL=3 #INFO and above only
```

**My Radvd conf is:**
```
pi@bananapi /etc $ cat radvd.conf
interface tap0
{
   AdvSendAdvert on;
   AdvManagedFlag off;     #stateless autoconfiguration
   AdvOtherConfigFlag on;  #clients get extra parameters via DHCPv6
   MaxRtrAdvInterval 10;   #resend RA @ random times, max 10sec delay
   prefix 2001:db8:2::/64  #announce prefix to clients
   {
       AdvOnLink on;
       AdvAutonomous on;
       AdvRouterAddr on;
   };
   RDNSS 2001:db8:2::2
   {
   };
};
```
**6LBR IP:** ::31f8:4ae3:212:4b00:768:ce5d

My Windows Laptop from where I trying to open the 6LowPAN browser to see the connected sensors on WSN and to ping their Ips from Windows host:
```
Wireless LAN adapter Wireless Network:
   Connection-specific DNS Suffix  . :
   Description . . . . . . . . . . . : Intel(R) WiFi Link 1000 BGN
   DHCP Enabled. . . . . . . . . . . : Yes
   Autoconfiguration Enabled . . . . : Yes
   IPv6 Address. . . . . . . . . . . : ::31f8:4ae3:84:74f0:9fb5:b63e(Preferred)
   Temporary IPv6 Address. . . . . . : ::31f8:4ae3:619a:ca77:a769:f897(Preferred)
   Link-local IPv6 Address . . . . . : fe80::84:74f0:9fb5:b63e%38(Preferred)
   IPv4 Address. . . . . . . . . . . : 192.168.1.98(Preferred)
   Subnet Mask . . . . . . . . . . . : 255.255.255.0
   Lease Obtained. . . . . . . . . . : 10 May 2016 09:25:58
   Lease Expires . . . . . . . . . . : 11 May 2016 11:47:48
   Default Gateway . . . . . . . . . : fe80::ae9e:17ff:fe94:2328%38
                                        192.168.1.1
   DHCP Server . . . . . . . . . . . : 192.168.1.1
```

**6Lbr.log:**

```
********************************************************************************
Tue May 10 07:55:32 UTC 2016 : Starting 6LBR
/usr/lib/6lbr/bin/cetic_6lbr_smart_bridge -c /etc/6lbr/nvm.dat -s /dev/ttyACM0 -t eth0 -r -B 115200 -U /usr/lib/6lbr/6lbr-ifup -D /usr/lib/6lbr/6lbr-ifdown -w /usr/lib/6lbr/www -L 3  -W /var/log/6lbr.timestamp -P 60 -C /var/log/6lbr.ip  -m /usr/lib/6lbr/plugins
Contiki-6lbr-1.3.3 started with IPV6, RPL
Rime started with address 1.2.3.4.5.6.7.8
MAC CSMA RDC br-rdc NETWORK sicslowpan
Log level : 30
Log services : ffffffff
2016-04-10 7:55:32.414980: INFO: ETH: 6LBR watchdog started (interval: 60)
[... 43 lines cut ...]
2016-04-10 7:56:25.865256: INFO: HTTP: Script : /
2016-04-10 7:56:28.240381: INFO: 6LBR: CETIC_BRIDGE : set_prefix
2016-04-10 7:56:28.562577: INFO: HTTP: Script : /6lbr_layout.css
2016-04-10 7:56:29.168980: INFO: HTTP: Script : /favicon.ico
2016-04-10 7:56:35.740385: INFO: HTTP: Script : /network.html
2016-04-10 7:56:35.784823: INFO: HTTP: Script : /6lbr_layout.css
2016-04-10 7:56:36.241295: INFO: 6LBR: CETIC_BRIDGE : set_prefix
2016-04-10 7:56:39.454717: INFO: HTTP: Script : /favicon.ico
2016-04-10 7:56:41.992360: INFO: HTTP: Script : /sensors.html
2016-04-10 7:56:42.36591: INFO: HTTP: Script : /6lbr_layout.css
2016-04-10 7:56:42.476963: INFO: HTTP: Script : /favicon.ico
2016-04-10 7:56:45.242338: INFO: 6LBR: CETIC_BRIDGE : set_prefix
2016-04-10 7:56:52.440753: INFO: 6LBR: CETIC_BRIDGE : set_prefix
2016-04-10 7:56:59.441782: INFO: 6LBR: CETIC_BRIDGE : set_prefix
2016-04-10 7:57:07.442775: INFO: 6LBR: CETIC_BRIDGE : set_prefix
2016-04-10 7:57:14.985416: INFO: 6LBR: CETIC_BRIDGE : set_prefix
2016-04-10 7:57:23.990820: INFO: 6LBR: CETIC_BRIDGE : set_prefix
2016-04-10 7:57:30.987177: INFO: 6LBR: CETIC_BRIDGE : set_prefix
2016-04-10 7:57:37.893753: INFO: 6LBR: CETIC_BRIDGE : set_prefix
2016-04-10 7:57:44.701313: INFO: 6LBR: CETIC_BRIDGE : set_prefix
2016-04-10 7:57:52.703013: INFO: 6LBR: CETIC_BRIDGE : set_prefix
2016-04-10 7:58:01.79196: INFO: 6LBR: CETIC_BRIDGE : set_prefix
2016-04-10 7:58:09.691444: INFO: 6LBR: CETIC_BRIDGE : set_prefix

```

laurentderu, 2016-05-10:
As you have enabled Raw Ethernet configuration, you don't need any RADVD on the RPi (but you will not be able to access 6LBR from the RPi).
According to the log, 6LBR is correctly started and configured, it has received the network configuration from your LAN router, so everything should be OK, so the problem is maybe on the Windows host. Could you send me a pcap file when you try to ping and open a browser towards 6LBR ?

rohitsivakumar, 2016-05-12:
I made some progress... but got stuck enroute but now at a different point.
I have configured my 6LBR in smartbridge mode. I am able to ping the nodes. In addition, since I had issues in getting jool, I went ahead to use the latest copy of the 6lbr from the 'develop' branch to use the nat64. However when I click on each individual node's web access to look at the sensor details or to configure the setup of mqtt config so I could see the message on IBM cloud, the node ping stops and it says request timed out. Also during this phase, when i click on the web (hyperlink) on the 6LBR Sensors list table (column-3) of that individual node, the ping stops for this node and webpage of this specific node shows error:  "This site can’t be reached"  with message "[2001:db8:2:0:212:4b00:7b5:9104] took too long to respond." . Why is it happening? The only way to recover from this state is to reset the node. This happens for all the nodes I connect.

**Here is my 6lbr conf:**
```
MODE=SMART-BRIDGE
RAW_ETH=0
BRIDGE=1
DEV_BRIDGE=br0
DEV_TAP=tap0
DEV_ETH0=eth0
RAW_ETH_FCS=0
DEV_RADIO=/dev/ttyACM0
BAUDRATE=115200
LOG_LEVEL=3 #INFO and above only
```

**here is my radvd conf:**
```
pi@bananapi /var/log $ cat /etc/radvd.conf
interface br0
{
   AdvSendAdvert on;
   AdvManagedFlag off;     #stateless autoconfiguration
   AdvOtherConfigFlag on;  #clients get extra parameters via DHCPv6
   MaxRtrAdvInterval 10;   #resend RA @ random times, max 10sec delay
   prefix 2001:db8:2::/64  #announce prefix to clients
   {
       AdvOnLink on;
       AdvAutonomous on;
       AdvRouterAddr on;
   };
   RDNSS 2001:db8:2::2
   {
   };
};
```

Here is my nvm config on 6LBR webserver:

```
**Info TAB**

Hostname : bananapi
Version : 1.4.x (Contiki-develop-20160421-5-gb7db545)
Mode : SMART BRIGDE
Uptime : 0h 8m 6s

WSN

MAC: CSMA
RDC: br-rdc (0 Hz)
Security: nullsec
HW address : 0:12:4b:0:7:68:ce:5d
Address : 2001:db8:2::212:4b00:768:ce5d
Local address : fe80::212:4b00:768:ce5d

Ethernet

HW address : 2:0:7:68:ce:5d
IP64 Address : 192.168.1.248
```

**Sensors Tab (on 6LBR web browser)**
SensorsNode treePRRParent switchHop count
Sensors

```
Sensors list

Node	Type	Web	Coap	Parent	Up PRR	Down PRR	Last seen	Status
2001:db8:2::212:4b00:a56:2f00	TI	web	coap	fe80::212:4b00:7b5:9104	100.0%	100.0%	0	OK
2001:db8:2::212:4b00:7b5:9104	TI	web	coap	fe80::212:4b00:768:ce5d	100.0%	14.3%	14	OK

```

**Ipconfig of my windows host from where I am accessing this 6LBR web-browser and pinging:**
```
Wireless LAN adapter Wireless Network:
   Connection-specific DNS Suffix  . :
   IPv6 Address. . . . . . . . . . . : 2001:db8:2:0:84:74f0:9fb5:b63e
   IPv6 Address. . . . . . . . . . . : bbbb::84:74f0:9fb5:b63e
   Temporary IPv6 Address. . . . . . : 2001:db8:2:0:e867:322d:d578:d7a3
   Temporary IPv6 Address. . . . . . : bbbb::e867:322d:d578:d7a3
   Link-local IPv6 Address . . . . . : fe80::84:74f0:9fb5:b63e%38
   IPv4 Address. . . . . . . . . . . : 192.168.1.98
   Subnet Mask . . . . . . . . . . . : 255.255.255.0
   Default Gateway . . . . . . . . . : fe80::4e:6ff:fec2:d995%38
                                                     192.168.1.1
```

**I am able to ping both the nodes (sensortags -cc26xx):**
```
C:\Users\Rohit>ping -6 2001:db8:2::212:4b00:7b5:9104

Pinging 2001:db8:2:0:212:4b00:7b5:9104 with 32 bytes of data:
Reply from 2001:db8:2:0:212:4b00:7b5:9104: time=414ms
Reply from 2001:db8:2:0:212:4b00:7b5:9104: time=248ms
Reply from 2001:db8:2:0:212:4b00:7b5:9104: time=771ms
Reply from 2001:db8:2:0:212:4b00:7b5:9104: time=748ms

Ping statistics for 2001:db8:2:0:212:4b00:7b5:9104:
    Packets: Sent = 4, Received = 4, Lost = 0 (0% loss),
Approximate round trip times in milli-seconds:
    Minimum = 248ms, Maximum = 771ms, Average = 545ms

C:\Users\Rohit>ping -6 2001:db8:2::212:4b00:a56:2f00

Pinging 2001:db8:2:0:212:4b00:a56:2f00 with 32 bytes of data:
Reply from 2001:db8:2:0:212:4b00:a56:2f00: time=731ms
Reply from 2001:db8:2:0:212:4b00:a56:2f00: time=478ms
Reply from 2001:db8:2:0:212:4b00:a56:2f00: time=87ms
Reply from 2001:db8:2:0:212:4b00:a56:2f00: time=87ms

Ping statistics for 2001:db8:2:0:212:4b00:a56:2f00:
    Packets: Sent = 4, Received = 4, Lost = 0 (0% loss),
Approximate round trip times in milli-seconds:
    Minimum = 87ms, Maximum = 731ms, Average = 345ms
```

**As soon as I click on the web hyper-link of the individual node on the 6LBR Sensors list table (column-3), the ping stops and it says**
```
C:\Users\Rohit>ping -6 -t 2001:db8:2::212:4b00:a56:2f00

Pinging 2001:db8:2:0:212:4b00:a56:2f00 with 32 bytes of data:
Reply from 2001:db8:2:0:212:4b00:a56:2f00: time=526ms
Reply from 2001:db8:2:0:212:4b00:a56:2f00: time=2153ms
Reply from 2001:db8:2:0:212:4b00:a56:2f00: time=1976ms
Request timed out.
Request timed out.
Request timed out.
Request timed out.
Request timed out.
```

laurentderu, 2016-05-12:
This is an unknown problem so far. Are you trying to access your node web server using IPv4 or IPv6 ? Does the problem occurs also if you use the other protocol ? On your node, what is the web server you are using, the one from web-demo or a custom made ?
Could you also send me a pcap taken when you try to open the web server of the node ?

rohitsivakumar, 2016-05-24:
Sending across the pcapng file and the error snap when opening the webserver of the node. I am using the web-demo one.
[image]
[image]

rohitsivakumar, 2016-05-24:
I was unable to load the file in zip format so renamed a .pcap file to .jpg. Please download and rename it with file extension .pcap to view.
[image]

laurentderu, 2016-05-24:
The connection seems ok, but getting more and more slow, it looks like a radio congestion due to too many packet retransmitted. Could you monitor the statistics page and look at the CSMA and Radio counters when you try to ping or access a node ?

closed 2017-12-14 by laurentderu