cetic/6lbr #142: Unable to talk to motes with NAT64

janeksz, 2016-03-23:
Hi,
I'm using 6lbr on RPi as edge router for Contiki based network (cc2650 motes) experiments as described here http://processors.wiki.ti.com/index.php/Cc26xx_sw_examples with the exception of wrapsix, (which never worked for me). In my home setup I use Microsoft  6to4 driver on Windows 7 desktop and everything works well. At university however I am not allowed to use it by  our IT people.
Taking advantage of newly released version of 6LBR I am trying to setup it's NAT64 and the result so far is that I can access 6LBR website, but not able to communicate with motes (using Firefox CoAP axtension)
If anyone was able to make it work, can I get some help how to do it?
Jan
my setup is:
root@raspberrypi:~# /usr/lib/6lbr/bin/nvm_tool --print /etc/6lbr/nvm.dat
Reading nvm file '/etc/6lbr/nvm.dat'
Channel : 25
PAN ID : abcd
WSN network prefix : aaaa::
WSN network prefix length : 64
WSN IP address : aaaa::100
WSN accept RA : True
WSN IP address autoconf : True
WSN 6LoWPAN context 0 : aaaa::
DNS server : ::
Eth network prefix : bbbb::
Eth network prefix length : 64
Eth IP address : bbbb::100
Eth default router : ::
Eth IP address autoconf : False
RA daemon : True
RA router lifetime : 0
RA maximum interval : 600
RA minimum interval : 200
RA minimum delay : 3
RA PIO enabled : True
RA prefix valid lifetime : 86400
RA prefix preferred lifetime : 14400
RA RIO enabled : True
RA RIO lifetime : 1800
RPL instance ID : 30
RPL DODAG ID manual : False
RPL DODAG ID : ::
RPL DODAG ID global : True
RPL Preference : 0
RPL version ID :  : 15
RPL DIO interval doubling : 8
RPL DIO minimum interval : 12
RPL DIO redundancy : 10
RPL default lifetime : 30
RPL maximum rank increase : 1792
RPL minimum rank increment : 256
RPL lifetime unit : 1
MAC layer : 1
Security layer : 0
Security level : 6
Security key : 0:1:2:3:4:5:6:7:8:9:a:b:c:d:e:f
Noncoresec anti-replay disabled : False
Noncoresec anti-replay workaround : False
Filter unknown nodes : False
NAT 64 Enabled : True
NAT 64 DHCP Enabled : False
NAT 64 static port mapping Enabled : True
NAT 64 RFC-6052 prefix : True
NAT 64 address : 192.168.1.55
NAT 64 netmask : 255.255.255.0
NAT 64 gateway : 192.168.1.1
Local address rewrite : True
Smart Multi BR : False
Webserver configuration page disabled : False
Webserver port : 80
root@raspberrypi:~#
[image]

janeksz, 2016-03-23:
do I need to change any of this (to use NAT64)? 
...

Eth network prefix : bbbb::
Eth network prefix length : 64
Eth IP address : bbbb::100
Eth default router : ::
Eth IP address autoconf : False
...
If yes, how

laurentderu, 2016-03-23:
To be able to communicate from a PC to a mote using NAT64/IPv4 you must preconfigure the port mapping, this can be done using the node configuration file on a Linux 6LBR.

In your 6lbr.conf file you need to add :

    NODE_CONFIG=/etc/6lbr/node_config.conf

The node_config.conf file contains the node definition, for example :

<pre>
[00:12:4b:00:06:a0:09:87]
name=Sensortag-1
coap=10087
</pre>

The node will be then accessible using port 10087

If you don't specify the port, 6LBR will assign a port number that you can retrieve from Configuration -> Sensors page.

Btw, looking at your snapshot, you are not trying to access the node using NAT64 but instead native IPv6... And it seems your PC  lacks the route to access the aaaa::/64 subnet.

janeksz, 2016-03-23:
Thanks Laurent,

That makes me a step further – close, but no cigar.
And I am still in a dark, when you say: … you are not trying to access the node using NAT64 but instead native IPv6…
What specifically do I need to change to correct it? Can you kindly point it out?
And if I am using CoAP port 5683 – default for Copper Firefox extension should I put coap=5683 or will still work with a different value specified in a node_config.conf?

Jan

laurentderu, 2016-03-23:
In the screenshot, you are using an IPv6 address, coap://[aaaa::...]/ NAT64 is a mechanism that allows to reach an IPv6 host using an IPv4 address and port combination. So you are mixing two concepts here.
Using NAT64, you will try access something more like coap://192.168.1.55:10087/ instead.
The port you configure in the node_config.conf file (and in the coap url) is not the coap port of your node, it stays to 5683, it's instead the port used to map the node IPv6 address (aaaa::....) and port (5683) onto one IPv4 port

janeksz, 2016-03-23:
Of course, silly me, I did right click>open in a new tab on the mote coap link without thinking. (maybe the webserver should provide the correct link, based on the settings). Anyhow thanks  for enlightening me, much appreciate it. You see, there are people like me missing on a basic understanding and I thank you for your patience and time. Looking forward to see the documentation and hope to have things going smoothly.

laurentderu, 2016-03-23:
Having NAT64 links also in the sensors page is a good idea (especially when you have dynamically allocated ports), I'll see how to integrate that without cluttering the page too much.

hauseja3, 2016-03-25:
@janeksz @laurentderu 
Hi guyz, 
could you please give me some advice, what code from examples should I use for nodes to create PoC or working (pinging) concept?

laurentderu, 2016-03-31:
The links are now NAT64 urls when the web server is accessed from an IPv4 host.

closed 2016-03-31 by laurentderu